pschatzmann/arduino-audio-tools #621: mp3 audio file duration in variable .

aqildad-create, 2023-02-12:
example  @file streams-sd_mp3-i2s.ino is working great . But iam unable to get file duration in  variable ? 
Is their any way out of this ? 

 THE LOG HAS INFORMATION BUT I NEED TO HAVE RETURN AS VARIABLE FOR FURTHER PROCESSING .

pschatzmann, 2023-02-12:
Please open issues only if you have found a bug.  

Feel free to extend the functionality which will force you to think about what you are really asking for!

closed 2023-02-12 by pschatzmann

aqildad-create, 2023-02-13:
thanks for the library and quick response too. 
can you please suggest how can i get the audio file duration ?

pschatzmann, 2023-02-13:
I suggest that you use google to figure out how to calculate the duration of a mp3 file! 
This will be your starting point that leads you to the relevant information and their source: If I remember right you need to use the bit rate that you can get from the decoder and the file size that you can get from the file