pschatzmann/arduino-audio-tools #622: streams-audiokit-sd-audiokit rec.raw has 0 Bytes

lechk82, 2023-02-12:
I have difficulties with the example streams-audiokit-sd-audio. A file rec.raw is created on the sdcard, but it has 0 bytes after recording.
The streams-audiokit-sd-audiokit.ino and player-sd-audiokit.ino examples work as expected without any problems. I assume that the hardware is working correctly. The Board is selected using `#define AUDIOKIT_BOARD 5`. I'm using an 16GB sdcard formatted as FAT32. Serial Output:

```
rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13836
load:0x40080400,len:3608
entry 0x400805f0
Initialization done.
[W] AudioKit.h : 54 - sd_active = true
[W] AudioKit.h : 611 - Mode Button ignored because of conflict: 13 
Press Key 1 to record
Recording...

```

i would appreciate any advice.

pschatzmann, 2023-02-12:
I guess that file.availableForWrite() returns 0. You can try to deactivate this check with  copier.setCheckAvailableForWrite(false);

lechk82, 2023-02-12:
Thank you for your quick reply and this great library. That was indeed the problem.

pschatzmann, 2023-02-12:
OK, I will plan to change the default setting in copier...

closed 2023-02-13 by pschatzmann