pschatzmann/arduino-audio-tools #627: No sound from streams-generator-a2dp.ino

korkbaum, 2023-02-14:
Dear Phil,

your work is really awesome.

I do, however, have an issue to get something running that I thought should be easy.
In the end, I wish to send sound from an I2S source to an A2DP sink (Headset). 

On my way there, I tried example "bt_music_sender.ino" first and it worked instantly. That means my set-up of compiler/board and BT device is generally working.

Example "streams-i2s-a2dp.ino" did not work so I checked "streams-generator-a2dp.ino" as a simpler starting point. I though that would basically do the same as the one that worked (from the ESP32-A2DP library), but it didn't. 

The only thing I changed in the examples was the BT device name and the BT connection part works flawlessly with both examples, only there is no sound in the latter one.

Am I missing a substantial point here to get this going?

Thanks for any hints.

BR
Klaus

pschatzmann, 2023-02-15:
In order to analyse what is happening you should change the log level both for the audiotools by calling AudioLogger::instance().begin(Serial, AudioLogger::Info); and for Bluetooth by setting the Core Debug level in the Tools menu (e.g. to verbose).

In any case I recommend to prefer to use the A2DP examples from the basic-api-examples menu.

pschatzmann, 2023-03-22:
I was redesigning the A2DPStream class to use the new concurrency classes.
I also provide some more examples that can be used for testing:
See https://github.com/pschatzmann/arduino-audio-tools/commit/47765b9feef378a6bf98145241e80d2a178285d6

closed 2023-03-22 by pschatzmann