pschatzmann/arduino-audio-tools #636: player.setIndex();   abnormal

gary7530, 2023-02-23:
There are two songs in the SD card, I use `player.setIndex(0);` with success, but use `player.setIndex(1);` it is wrong.

Use `player.setIndex(0);`
```
[I] AudioPlayer.h : 468 - void audio_tools::AudioPlayer::writeEnd()
[I] AudioSourceSDFAT.h : 101 - selectStream SDFAT: 0
[I] AudioSourceSDFAT.h : 120 - -> selectStream: /128kbps.mp3
[I] AudioEncoded.h : 322 - virtual void audio_tools::EncodedAudioPrint::end()
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] AudioPlayer.h : 180 - reset codec
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
```

use `player.setIndex(1);`
```
[I] AudioPlayer.h : 468 - void audio_tools::AudioPlayer::writeEnd()
[I] Fade.h : 104 - faded out 1152 frames to volume 0.000000
[I] Fade.h : 104 - faded out 1152 frames to volume 0.000000
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioSourceSDFAT.h : 101 - selectStream SDFAT: 1
[E] AudioSourceSDFAT.h : 111 - Filename is null
[I] AudioEncoded.h : 322 - virtual void audio_tools::EncodedAudioPrint::end()
[I] Fade.h : 109 - fade in 1152 frames from volume 0.000000
[I] MetaDataID3.h : 570 - virtual void audio_tools::MetaDataID3::end()
[I] AudioPlayer.h : 180 - reset codec
```

pschatzmann, 2023-02-23:
I can't reproduce your issue. For me it is working perfectly.
Make sure that you have more than 1 matching file!

gary7530, 2023-02-23:
```
#include "AudioTools.h"
#include "AudioLibs/AudioSourceSDFAT.h"
#include "AudioCodecs/CodecMP3Helix.h"

#define SPI_MISO 17
#define SPI_CLK 5
#define SPI_MOSI 18
#define SD_CS 19

const char* startFilePath = "/";
const char* ext = "mp3";
AudioSourceSDFAT source(startFilePath, ext, SD_CS);
AnalogAudioStream out;
MP3DecoderHelix decoder;
AudioPlayer player(source, out, decoder);

int i0 = 0;
unsigned long time_now = 0, period = 3000;

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  SPI.begin(SPI_CLK, SPI_MISO, SPI_MOSI, SD_CS);

  // setup output
  auto cfg = out.defaultConfig();
  out.begin(cfg);

  // setup player
  player.begin();
}

void loop() {
  player.copy();

  if (millis() > time_now + period)
  {
    time_now = millis();
    player.setIndex(i0);
    Serial.printf("%d\t%s\r\n", source.index(), source.toStr());
    i0++;
    if (i0 > source.size())
      i0 = 0;
  }
}
```

Songs [01.zip](https://github.com/pschatzmann/arduino-audio-tools/files/10816678/01.zip)

pschatzmann, 2023-02-23:
I am confused: what are you trying to do?
Just leave away all your logic (in the loop) and after finishing the first file the player will move automatically to the next

gary7530, 2023-02-24:
I originally wrote it in `button();`, for your convenience, I will rewrite it to `loop()` to trigger every 3 seconds.

gary7530, 2023-02-24:
```
    switch (buttton)
    {
    case 1: // Mode

        break;
    case 2: // Up
        if (SongNumber >= TotalSongs)
            SongNumber = 0;
        else
            SongNumber++;

        player.setIndex(SongNumber);
        Serial.printf("%d\t%s\r\n", source.index(), source.toStr());
        break;
    case 3: // Down
        if (SongNumber == 0)
            SongNumber = TotalSongs;
        else
            SongNumber--;

        player.setIndex(SongNumber);
        Serial.printf("%d\t%s\r\n", source.index(), source.toStr());
        break;
    case 4: // Delay
        Serial.printf("%d\t%s\r\n", source.index(), source.toStr());
        break;
    case 5: // Set

        break;
    case 6: // Play
        if (player.isActive())
        {
            player.stop();
            player.setSilenceOnInactive(true);
        }
        else
        {
            player.play();
        }
        break;
    default:
        break;
    }
    buttton = 0;
```

pschatzmann, 2023-02-24:
No, I do not have the time to debug your code. Can you confirm that if you use my standard sketch, not all two songs are playing after each other ?
And if you look at the examples they have the setIndex in the begin() commented out. Did you try to use it from there with setIndex(1) ?

gary7530, 2023-02-24:
I use [player-sdfat-analog.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-player/player-sdfat-analog/player-sdfat-analog.ino)

The songs all play normally,The problem I encounter now is the `setIndex()` abnormal,There is only one song in the SD card, and an error will occur when using setIndex(0).

pschatzmann, 2023-02-24:
I don't consider this as a high priority issue. After all, I created the player to play many files. In my case I have around 1000 in multiple directories.  If you just have one, you would use a simpler sketch that plays just that file like this one: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-sd_mp3-i2s/streams-sd_mp3-i2s.ino

gary7530, 2023-02-24:
I seem to have found the reason. When I use `source.size();` and then use `setIndex()`, an error will occur.
Without using size(), setIndex() is normal

gary7530, 2023-02-25:
#644

pschatzmann, 2023-02-25:
Thanks a lot for your PR. It has been merged!

closed 2023-02-25 by pschatzmann