pschatzmann/arduino-audio-tools #640: Mozzi examples not compiling

mkoppanen, 2023-02-24:
I am trying to get the mozzi integration running in order to create a pink noise generator for my toddler. At the moment running into a few issues with https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/sandbox/streams-mozzi-a2dp

If I take the sketch as is it fails to compile with:

````
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino:11:1: error: 'MozziStream' does not name a type; did you mean 'Stream'?
 MozziStream in;        // Stream generated with mozzi
 ^~~~~~~~~~~
 Stream
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino:13:24: error: 'in' was not declared in this scope
 StreamCopy copier(out, in); // copy in to out
                        ^~
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino:13:24: note: suggested alternative: 'yn'
 StreamCopy copier(out, in); // copy in to out
                        ^~
                        yn
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino: In function 'void setup()':
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino:40:14: error: 'in' was not declared in this scope
   auto cfg = in.defaultConfig();
              ^~
/Users/mikko/Documents/Arduino/bt-player/bt-player.ino:40:14: note: suggested alternative: 'yn'
   auto cfg = in.defaultConfig();
              ^~
              yn

exit status 1

Compilation error: 'MozziStream' does not name a type; did you mean 'Stream'?
````

I assume the above is for missing include for AudioMozzi.h

Additionally I cannot for the life of me find Mozzi.h referenced here: https://github.com/pschatzmann/arduino-audio-tools/blob/main/src/AudioLibs/AudioMozzi.h#L7

i don't have much experience (none) with Mozzi yet but the main header they have seems to be called MozziGuts.h (https://github.com/sensorium/Mozzi/blob/master/MozziGuts.h)

Am on the wrong track here or is the integration currently broken?

pschatzmann, 2023-02-24:
If you want to use this, you will need to install my extended version of Mozzi: https://github.com/pschatzmann/Mozzi

pschatzmann, 2023-02-24:
I just added the White Noise Generator.
The test sketch looks as follows:
```
#include "AudioTools.h"
uint16_t sample_rate=32000;
uint8_t channels = 2;                                   // The stream will have 2 channels 
PinkNoiseGenerator<int16_t> noise(32000);               // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound(noise);             // Stream generated from sine wave
I2SStream out; 
StreamCopy copier(out, sound);                          // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.sample_rate = sample_rate; 
  config.channels = channels;
  config.bits_per_sample = 16;
  out.begin(config);

  // Setup Pink noise
  sound.begin(config);
  Serial.println("started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}
```

mkoppanen, 2023-02-24:
@pschatzmann Amazing! Thank you very much for quick replies. I'll give it a spin. 

I got a couple of different boards ESP32-WROOM-32D, ESP32-WROOM-32D, ESP32-CAM and a few m5stack atom lites. At the moment the board I am testing with is ESP32-WROOM-32D.

closed 2023-02-25 by pschatzmann