pschatzmann/arduino-audio-tools #648: ESP8266 Compile Errors: NOT_ENOUGH_MEMORY_MSG with streams-url_mp3_helix-i2s

ksaye, 2023-02-26:
Great audio-tools and it appears that they are not all the examples are compatible with ESP8266 (just guessing).
If I should just move on to ESP32, please just say so.  I do see in the readme: An ESP32 was used to test this sketch.
With a ESP8266 and trying the 'streams-url_mp3_helix-i2s' sample I get the errors below.
Setup:
* NodeMCU .9
* AUDIOTOOLS_VERSION: 0.8.0
* Arduino: 2.0.3
* ESP Board Library: 3.1

       In file included from c:\users\ksaye\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\assert.h:10,
                           from c:\users\ksaye\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\sys\reent.h:503,
                           from c:\users\ksaye\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\stdlib.h:18,
                           from c:\users\ksaye\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\cstdlib:75,
                           from c:\users\ksaye\appdata\local\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.1.0-gcc10.3-e5f9fec\xtensa-lx106-elf\include\c++\10.3.0\stdlib.h:36,
                           from C:\Users\ksaye\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.1.1\cores\esp8266/Arduino.h:27,
                           from C:\Users\ksaye\AppData\Local\Temp\arduino-sketch-965E40EDAFF0F409D089B323F1D662DA\sketch\streams-url_mp3_helix-i2s.ino.cpp:1:
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In constructor 'audio_tools::StreamCopyT<T>::StreamCopyT(Print&, audio_tools::AudioStream&, int)':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:29:22: error: initializer fails to determine size of '__pstr__'
             29 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h:148:83: note: in expansion of macro 'LOG_OUT_PGMEM'
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:29:17: note: in expansion of macro 'LOGE'
             29 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In constructor 'audio_tools::StreamCopyT<T>::StreamCopyT(Print&, Stream&, int)':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:39:22: error: initializer fails to determine size of '__pstr__'
             39 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h:148:83: note: in expansion of macro 'LOG_OUT_PGMEM'
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:39:17: note: in expansion of macro 'LOGE'
             39 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In constructor 'audio_tools::StreamCopyT<T>::StreamCopyT(int)':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:48:22: error: initializer fails to determine size of '__pstr__'
             48 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h:148:83: note: in expansion of macro 'LOG_OUT_PGMEM'
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:48:17: note: in expansion of macro 'LOGE'
             48 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In instantiation of 'audio_tools::StreamCopyT<T>::StreamCopyT(int) [with T = unsigned char]':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:322:90:   required from here
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:48:22: error: array must be initialized with a brace-enclosed initializer
             48 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h:148:83: note: in expansion of macro 'LOG_OUT_PGMEM'
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:48:17: note: in expansion of macro 'LOGE'
             48 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In instantiation of 'audio_tools::StreamCopyT<T>::StreamCopyT(Print&, audio_tools::AudioStream&, int) [with T = unsigned char]':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:326:131:   required from here
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:29:22: error: array must be initialized with a brace-enclosed initializer
             29 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioLogger.h:148:83: note: in expansion of macro 'LOG_OUT_PGMEM'
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:29:17: note: in expansion of macro 'LOGE'
             29 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h: In instantiation of 'audio_tools::StreamCopyT<T>::StreamCopyT(Print&, Stream&, int) [with T = unsigned char]':
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:330:126:   required from here
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:39:22: error: array must be initialized with a brace-enclosed initializer
             39 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
[... cut ...]
            148 | #define LOGE(fmt, ...) if (AudioLogger::instance().level()<=AudioLogger::Error) { LOG_OUT_PGMEM(AudioLogger::Error, fmt, ##__VA_ARGS__);}
                |                                                                                   ^~~~~~~~~~~~~
          c:\Users\ksaye\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioCopy.h:39:17: note: in expansion of macro 'LOGE'
             39 |                 LOGE(NOT_ENOUGH_MEMORY_MSG, buffer_size);
                |                 ^~~~
          
          exit status 1
          
          Compilation error: exit status 1

pschatzmann, 2023-02-26:
I just committed a correction, so the example is compiling agin. 
However please note that I have never tested this on the ESP8266...

closed 2023-02-26 by pschatzmann

ksaye, 2023-02-26:
Thanks for the quick change.

It now compiles, but does not run correctly.  I will switch to an ESP32:

           User exception (panic/abort/assert)
      --------------- CUT HERE FOR EXCEPTION DECODER ---------------
      
      Unhandled C++ exception: OOM
      
      >>>stack>>>
      
      ctx: cont
      sp: 3ffffe50 end: 3fffffd0 offset: 0000
      3ffffe50:  00000002 3ffffe7d 0000000a 4020fa4c  
      3ffffe60:  000000fe 00000000 00000000 00000000  
      3ffffe70:  00000000 00000000 00000000 3fff65dc  
      3ffffe80:  3fffefd0 000000fe 3fff2f10 00000001  
      3ffffe90:  00000000 00000000 3fff491c 3fff2648  
      3ffffea0:  3fffdad0 40209193 00002800 40211472  
      3ffffeb0:  3fffdad0 40209193 00000020 40211491  
      3ffffec0:  3fffdad0 00000000 00002800 4021031c  
      3ffffed0:  4020f684 00000000 3fff364c 40209193  
      3ffffee0:  3fffdad0 00000067 3fff2594 3fff2648  
      3ffffef0:  3fffdad0 00000000 3fff3624 40209516  
      3fffff00:  3fffdad0 3fff1eb4 3fff1eb4 4020a7b4  
      3fffff10:  3fffdad0 00000000 3fff1eb4 4020dc5d  
      3fffff20:  40214648 0000ac44 00000002 00000010  
      3fffff30:  00000001 feefef01 00000000 ffffffff  
      3fffff40:  ffffffff ffffffff feefeffe 00000000  
      3fffff50:  40214648 0000ac44 00000002 00000010  
      3fffff60:  00000001 feefef01 00000000 ffffffff  
      3fffff70:  ffffffff ffffffff feefeffe 00000000  
      3fffff80:  40214648 0000ac44 00000002 00000010  
      3fffff90:  00000001 feefef01 00000000 ffffffff  
      3fffffa0:  ffffffff ffffffff feefeffe 00000000  
      3fffffb0:  feefeffe feefeffe 3fff261c 40211028  
      3fffffc0:  feefeffe feefeffe 3fffdab0 40100ea1  
      <<<stack<<<
      
      last failed alloc call: 40209193(10240)
      
      --------------- CUT HERE FOR EXCEPTION DECODER ---------------
      
      last failed alloc caller: 0x40209193
      
       ets Jan  8 2013,rst cause:2, boot mode:(3,0)
      
      load 0x4010f000, len 3424, room 16 
      tail 0
      chksum 0x2e
      load 0x3fff20b8, len 40, room 8 
      tail 0
      chksum 0x2b
      csum 0x2b
      v00058540
      ~ld

pschatzmann, 2023-02-27:
I had a quick look what the problem is: there is just not enough RAM for doing both on an ESP8266: Wifi and mp3 decoding at the same time. Each functionality alone is working fine!

ksaye, 2023-02-27:
Thanks for the confirmation.