pschatzmann/arduino-audio-tools #649: Error: NPE on streams-i2s-webserver-wav

ksaye, 2023-02-27:
Using the following script on a Heltech LoRa v2 (ESP32), I get the error when I open the web browser.

      starting I2S...
      [I] I2SConfig.h : 104 - rx/tx mode: RX_MODE
      [I] I2SConfig.h : 105 - port_no: 0
      [I] I2SConfig.h : 106 - is_master: Master
      [I] I2SConfig.h : 107 - sample rate: 22050
      [I] I2SConfig.h : 108 - bits per sample: 16
      [I] I2SConfig.h : 109 - number of channels: 2
      [I] I2SConfig.h : 110 - i2s_format: I2S_STD_FORMAT
      [I] I2SConfig.h : 112 - auto_clear:0
      [I] I2SConfig.h : 119 - buffer_count:10
      [I] I2SConfig.h : 120 - buffer_size:512
      [I] I2SConfig.h : 124 - pin_bck: 12
      [I] I2SConfig.h : 126 - pin_ws: 13
      [I] I2SConfig.h : 128 - pin_data: 36
      I2S started
      [I] CodecWAV.h : 452 - sample_rate: 22050
      [I] CodecWAV.h : 453 - channels: 2
      [I] CodecWAV.h : 457 - is_streamed! because length is 2147418112
      .......
      IP address: 192.168.15.147
      [I] AudioServer.h : 216 - New Client.
      [I] AudioServer.h : 367 - sendReply - Returning encoded stream...
      [I] CodecWAV.h : 452 - sample_rate: 22050
      [I] CodecWAV.h : 453 - channels: 2
      [I] CodecWAV.h : 457 - is_streamed! because length is 4294967295
      [I] AudioCopy.h : 71 - buffer_size=1024
      [E] AudioEncoded.h : 341 - NPE
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 2 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 3 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 4 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 5 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 6 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 7 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 8 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 9 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 10 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 11 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 12 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 13 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 14 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 15 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 16 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 17 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 18 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 19 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 20 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [I] AudioCopy.h : 287 - try write - 21 (open 1024 bytes) 
      [E] AudioEncoded.h : 341 - NPE
      [E] AudioCopy.h : 281 - write to target has failed! (1024 bytes)
      [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
      [I] AudioServer.h : 122 - stop client...
      [I] AudioServer.h : 216 - New Client.
      [I] AudioServer.h : 367 - sendReply - Returning encoded stream...
      [I] CodecWAV.h : 452 - sample_rate: 22050
      [I] CodecWAV.h : 453 - channels: 2
      [I] CodecWAV.h : 457 - is_streamed! because length is 4294967295
      [I] AudioCopy.h : 71 - buffer_size=1024
      [E] AudioEncoded.h : 341 - NPE

I have the INMP441 to ESP wired as follows:

INMP441 | ESP
-- | --
VDD | 3.3 v
GND | GND
SD | GPIO_NUM_36
L / R | GND
WS | GPIO_NUM_13
SCK | GPIO_NUM_12

Script:

```
/**
 * @file streams-i2s-webserver_wav.ino
 *
 *  This sketch reads sound data from I2S. The result is provided as WAV stream which can be listened to in a Web Browser
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("ssid","password");  // the same a above
/*
| INMP441 |  ESP32
| --------| ---------------
| VDD     |  3.3
| GND     |  GND
| SD      |  IN (GPIO32)
| L/R     |  GND
| WS      |  WS (GPIO15)
| SCK     |  BCK (GPIO14)
*/

I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Error);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 22050;
  config.channels = 2;
  config.bits_per_sample = 16;
  config.pin_ws = GPIO_NUM_13;       //WS
  config.pin_bck = GPIO_NUM_12;      //SCK
  config.pin_data = GPIO_NUM_36;     //SD

  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();  
}
```

pschatzmann, 2023-02-27:
Did you download the library today ? I might have comitted a change that was causing this...

pschatzmann, 2023-02-27:
The correction has been committed: it is working again...

closed 2023-02-27 by pschatzmann

ksaye, 2023-02-27:
Thank you and as you already know, it works fine.