pschatzmann/arduino-audio-tools #656: Example: streams-i2s-tf.ino is locked to Board Type ARDUINO_ARDUINO_NANO33BLE

ksaye, 2023-03-01:
the example: https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-i2s-tf/streams-i2s-tf.ino does not compile because https://github.com/pschatzmann/tflite-micro-arduino-examples/blob/main/src/peripherals/peripherals.h seems to be locked to ARDUINO_ARDUINO_NANO33BLE

Perhaps a small update to: https://github.com/pschatzmann/arduino-audio-tools/wiki/TensorFlow-Lite---MicroSpeech and/or the Readme could help

pschatzmann, 2023-03-01:
What make you think so. I was testing this on an ESP32...

ksaye, 2023-03-01:
I could be wrong, but here is the error I see:

      In file included from C:\Users\kevinsay\OneDrive - Microsoft\Documents\Arduino\libraries\tflite-micro-arduino-examples-main\src/TensorFlowLite.h:22:0,
                       from C:\Users\kevinsay\OneDrive - Microsoft\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/TfLiteAudioStream.h:6,
                       from C:\Users\kevinsay\AppData\Local\Temp\.arduinoIDE-unsaved202321-40508-6dddy3.pokvh\streams-i2s-tf\streams-i2s-tf.ino:13:
      C:\Users\kevinsay\OneDrive - Microsoft\Documents\Arduino\libraries\tflite-micro-arduino-examples-main\src/peripherals/peripherals.h:61:2: error: #error "unsupported board"
       #error "unsupported board"
        ^
      In file included from C:\Users\kevinsay\AppData\Local\Temp\.arduinoIDE-unsaved202321-40508-6dddy3.pokvh\streams-i2s-tf\streams-i2s-tf.ino:13:0:
      C:\Users\kevinsay\OneDrive - Microsoft\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/TfLiteAudioStream.h:16:56: fatal error: tensorflow/lite/micro/micro_error_reporter.h: No such file or directory
      compilation terminated.
      
      exit status 1
      
      Compilation error: exit status 1

[image]

pschatzmann, 2023-03-01:
I think you have some incompatiable version installed as well which is used instead of mine.
If you double check in the code that you sent in the link there is a condition to allow the ESP32

ksaye, 2023-03-01:
Thanks, let me clean up and try again.  :)

ksaye, 2023-03-01:
@pschatzmann , after cleaning out the directories and re importing, it now compiles.

I did get a warning:

    Library Arduino_TensorFlowLite has been declared precompiled:
    Precompiled library in "C:\Users\kevinsay\OneDrive - Microsoft\Documents\Arduino\libraries\Arduino_TensorFlowLite\src\esp32" not found

While it did compile, what does this mean?

[image]

pschatzmann, 2023-03-01:
Arduino has the functionality to use precompiled code. This has not been provided for the ESP32, so it is compiling from the source code

ksaye, 2023-03-01:
As always, thanks for the help.

closed 2023-03-01 by ksaye