pschatzmann/arduino-audio-tools #660: Compilation error with AudioESP8266

BeijingUncle, 2023-03-03:
Arduino/libraries/arduino-audio-tools/src/AudioLibs/AudioESP8266.h:17:51: error: expected class-name before ',' token
 class AudioOutputWithCallback : public AudioOutput, public BufferedStream {

pschatzmann, 2023-03-03:
How can I reproduce this ?

BeijingUncle, 2023-03-03:
I'm trying to have a quick test with esp8266Audio lib. I am following the article on your blog: https://www.pschatzmann.ch/home/2021/04/29/stream-sd-mp3-files-to-a2dp-bluetooth-using-the-esp8266audio-library/.

pschatzmann, 2023-03-03:
A lot has changed since I published this article. That the reason why I also removed this example from Github.
Please use the examples from the basic-api directory!
Though you might be able to make this work again, you are petty much on your own.

closed 2023-03-03 by pschatzmann

pschatzmann, 2023-03-03:
I committed some corrections, so it should at least compile again.

BeijingUncle, 2023-03-03:
It works! Thanks for your fast and responsible reply!

#include <SPI.h>
#include <SD.h>
#include "AudioFileSourceSD.h"
#include "AudioGeneratorMP3.h"
#include "AudioTools.h"
#include "AudioLibs/AudioESP8266.h"
#include "BluetoothA2DPSource.h"

using namespace audio_tools;  

const int sd_ss_pin = 5;
const char* fileName = "/test.mp3";
BluetoothA2DPSource a2dp_source;
AudioFileSourceSD *file;
AudioGeneratorMP3 *mp3;
AudioOutputWithCallback *out;

// callback used by A2DP to provide the sound data
int32_t get_sound_data(Frame* data, int32_t len) {
  return out == nullptr ? 0 : out->read(data, len);
}

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  audioLogger = &Serial;

  // Setup Audio
  file = new AudioFileSourceSD(); 
  mp3 = new AudioGeneratorMP3();
  out = new AudioOutputWithCallback(512, 5);
  
  // Open MP3 file and play it
  SD.begin(sd_ss_pin);
  if (file->open(fileName)) {
    
    // start the bluetooth
    Serial.println("starting A2DP...");
    a2dp_source.start("", get_sound_data);  

    // start to play the file
    mp3->begin(file, out);
    Serial.printf("Playback of '%s' begins...\n", fileName);
  } else {
    Serial.println("Can't find .mp3 file");
  }
}

// Arduino loop - repeated processing 
void loop() {
  if (mp3->isRunning()) {
    if (!mp3->loop()) mp3->stop();
  } else {
    Serial.println("MP3 done");
    delay(10000);
  }
}

pschatzmann, 2023-03-03:
If I remember right the EncodedAudioStream can now be used as an Audio Source as well. So it should be much easier using this class and in the callback to just do a readBytes() on the encoded stream.

BeijingUncle, 2023-03-03:
I'll try it, thanks again!