pschatzmann/arduino-audio-tools #667: MultiOutput causing LoadProhibited error

colpiche, 2023-03-05:
Hi pschatzmann,

I'm trying to implement MultiOutput for splitting streams but I get a `Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.`

Currently the code that is working is doing `AudioSourceSDFAT` -> `MP3DecoderHelix` -> `AudioPlayer` -> `AnalogAudioStream`. 

What I would like to achieve is :
                                                                                                                     | `AnalogAudioStream`
`AudioSourceSDFAT` -> `MP3DecoderHelix` -> `AudioPlayer` -> `MultiOutput` |
                                                                                                                     | `AudioRealFFT` -> `LEDOutput`
                                                                                                                     
                                                                                                                     
I get the error as soon that the AudioPlayer's stream goes to the MultiOuput and the ESP32 keeps rebooting ever and ever.

Here's my code (currently only one output, just for testing purpose) :
``` cpp
#include "AudioTools.h"
#include "AudioLibs/AudioSourceSDFAT.h"
#include "AudioCodecs/CodecMP3Helix.h"

int channels = 2; // The stream has 2 channels 

const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSDFAT mp3_file(startFilePath, ext);
MP3DecoderHelix decoder;
AnalogAudioStream out1dac;
MultiOutput out(out1dac);
AudioPlayer player(mp3_file, out, decoder);

void printMetaData(MetaDataType type, const char* str, int len){
  Serial.print("==> ");
  Serial.print(toStr(type));
  Serial.print(": ");
  Serial.println(str);
}

void setup() {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Info);

    // setup DAC output
    auto dac_cfg = out1dac.defaultConfig();
    out1dac.begin(dac_cfg);

    // setup player
    // player.setMetadataCallback(printMetaData);
    player.begin();
}

void loop() {
    player.copy();
}
```

And here's the output of the serial monitor :
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13192
load:0x40080400,len:3028
entry 0x400805e4
[    18][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[I] AnalogAudioBase.h : 57 - I2S_MODE_DAC_BUILT_IN
[I] AnalogAudio.h : 57 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
[I] AudioTypes.h : 63 - sample_rate: 44100
[I] AudioTypes.h : 64 - channels: 2
[I] AudioTypes.h : 65 - bits_per_sample: 16
[I] AnalogAudioBase.h : 72 - analog left output pin: 25
[I] AnalogAudioBase.h : 73 - analog right output pin: 26
[I] AnalogAudioESP32.h : 106 - TX_MODE
[I] AudioPlayer.h : 397 - virtual void audio_tools::AudioPlayer::setMetadataCallback(void (*)(audio_tools::MetaDataType, const char*, int), audio_tools::ID3TypeSelection)
[I] AudioPlayer.h : 343 - setVolume(1.000000)
[I] VolumeStream.h : 183 - setVolume: 1.000000
[I] VolumeStream.h : 183 - setVolume: 1.000000
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceSDFAT.h : 101 - selectStream SDFAT: 0
[I] AudioSourceSDFAT.h : 120 - -> selectStream: /Robot-rock_44k.mp3
[I] AudioCopy.h : 71 - buffer_size=1024
[I] MetaDataID3.h : 508 - callback TALB
==> Album: Human after all
[I] MetaDataID3.h : 508 - callback TPE1
==> Artist: Daft Punk
[I] MetaDataID3.h : 508 - callback TIT2
==> Title: Robot rock
[I] MetaDataID3.h : 508 - callback TCON
==> Genre: Disco
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d5916  PS      : 0x00060c30  A0      : 0x800d6c9f  A1      : 0x3ffb2230  
A2      : 0x3ffc2938  A3      : 0x3ffc2c50  A4      : 0x00000400  A5      : 0x00000400  
A6      : 0x00000001  A7      : 0xff000000  A8      : 0x800df9ae  A9      : 0x3ffb2210  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x000000c1  A13     : 0x00002b40  
A14     : 0x00000400  A15     : 0x00000400  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40084431  LEND    : 0x40084439  LCOUNT  : 0x00000027  

Backtrace: 0x400d5913:0x3ffb2230 0x400d6c9c:0x3ffb2250 0x400d6cd6:0x3ffb2270 0x400e03dd:0x3ffb2290

ELF file SHA256: 6022743848d906cb

Rebooting...
```

Do you have any clue why's MultiOuput is causing this error ? Or maybe another way to achieve my purpose ?

Thanks a lot for your library by the way, it's an awesome job !

pschatzmann, 2023-03-05:
Just analye your stack trace: I bet you just run out of memory!

closed 2023-03-05 by pschatzmann

colpiche, 2023-03-05:
Sorry but I'm totally unable to interpret the log, how can I know I'm running out of memory ? Is this sketch so resource-intensive ?

pschatzmann, 2023-03-05:
You can't live without the Exception Decoder: 
https://github.com/me-no-dev/EspExceptionDecoder
There is just no other way to fingure out what's going on exception using a Debugger

pschatzmann, 2023-03-05:
https://www.pschatzmann.ch/home/2023/01/25/analyzing-esp32-exceptions/

colpiche, 2023-03-05:
Thanks, it helps. I get this from the decoder :
[image]

Should I understand the error is caused by the moveToNextFileOnTimeout() method ?

pschatzmann, 2023-03-05:
Look what's on line 465. There you are running into a null pointer exception because some objects have not been defined.
I can't look it up for you because I don't know what version you are running and there were some recent changes there which might or might not correct the problem.

colpiche, 2023-03-05:
Line 465 is the moveToNextFileOnTimeout() method definition. I was using version 0.9.3, updated to 0.9.4 with no success and then reverted back to the 0.9.3. Now it's working, I don't get any exception and my sketch is working. I don't understand anything.

pschatzmann, 2023-03-05:
Usually I would recommend to use main because this contains the latest error corrections.

colpiche, 2023-03-05:
Yes I was working on main branch. I've seen you've made changes 2 hours ago, maybe it helped, the non working version was installed yesterday.