pschatzmann/arduino-audio-tools #671: Only white noise when implementing streams-i2s-a2dp example

bkis, 2023-03-07:
The problem I have is very unlikely to originate from your library, so I'm just asking for a little help here. For me, it's very difficult to identify whats going wrong (new to MC development) but I bet you'll have an idea right away.

I followed your [example for streaming from I2s input to Bluetooth output](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-i2s-a2dp) (and then to a BT headphone) very closely. The hardware I used is the same INMP441 microphone module and [this](https://www.azdelivery.de/en/collections/d1-mini/products/esp32-d1-mini) ESP32 board. 

To my surprise, connecting my headphones worked right away. But I only get something close to white noise with periodic interruptions and no signal from the microphone: https://eksss.de/temp/inmp441-to-esp32.m4a

I commented out the `AudioLogger` line because it produced so much output I doubted the esp could handle it. It changed the interval of the interruptions, but that's about it.

Do you have any idea what the problem could be here? Is the microphone broken? Did I connect the devices in the wrong way (I doubt it). Thank you for your help and thank you for this library!

pschatzmann, 2023-03-07:
https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working

I doubt that there is an issue with my library, so I am closing this.

closed 2023-03-07 by pschatzmann

bkis, 2023-03-07:
Yeah, fair enough. I told you it's a request for help and not an issue with the library. GitHub doesn't have a contact function (anymore) so I misused the issue system. Anyway, sorry for that.

pschatzmann, 2023-03-07:
There is a Discussion tab for this!