pschatzmann/arduino-audio-tools #672: Mixer has (buffering) problems with PCM files with a samplerate > 16000

onyx99, 2023-03-07:
I tried to play PCM, WAV, RAW files trough the inputmixer but when the samplerate is > 16000 it starts to make a alot of noise, 
i also increased the DEFAULT_BUFFER_SIZE to 4096 in AudioConfig.h but without any luck
when i don't use the mixer and play the url trough the kit it plays fine upto 48000 with 2 channels, 

can i try something else? or is the ESP32 to weak to use the audio-tools mixer

below is the code that i used:

#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"

InputMixer<int16_t> mixer;
AudioKitStream kit;

#define sampleRate 16000

URLStream url("ssid","pwd", DEFAULT_BUFFER_SIZE);

StreamCopy copier(kit, mixer);
//StreamCopy copier(kit, url);

void setup() {

  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // setup output
  auto cfg = kit.defaultConfig(TX_MODE);
  cfg.channels = 2;
  cfg.sample_rate = sampleRate;
  kit.begin(cfg);
  // max volume
  kit.setVolume(1.0);

  mixer.add(url);

  if(cfg.sample_rate==16000){
    url.begin("http://192.168.2.1/audio-16000.pcm");
  } else {
    url.begin("http://192.168.2.1/audio-32000.pcm");
  }

 //url.begin("https://pschatzmann.github.io/Resources/audio/audio-8000.raw");

  mixer.begin(cfg);

}

void loop() {  
  copier.copy();
}

pschatzmann, 2023-03-07:
Are you sure that the noise is not coming from the logger with  AudioLogger::Info?

onyx99, 2023-03-07:
I included the logger afterwards, to look if noticed something going on.

pschatzmann, 2023-03-08:
I could trace down the issue on the fact that the URLStream did return -1 as the size when there was no data instead of 0
I have committed some corrections.

pschatzmann, 2023-03-08:
I was testing with AudioLogger::Warning. 
You are right with AudioLogger::Info it is still returning 0 or 1 many times from the URLStream. 

I commented out the available() method so it takes the default implementation and this seems to do the trick...

onyx99, 2023-03-08:
Streamcopy looks fine now, thanks! and i can play now upto with a samplerate of 22050 without any glitches, at 32000 it (still) sounds like it cannot catch up for some reason, may be a lack of esp32 resources, i don't know, i used a local webserver with 1mbit UTP connection to stream the PCM files.

pschatzmann, 2023-03-08:
I added a delay in the readBytes of the InputMixer. 
You can try to comment this out or decrease the delay amount

onyx99, 2023-03-08:
I remember earlephilhower also has a mixer in his sources, maybe i should try if that plays normal at 32000 and 44100 samplerate to check if it is not a lack of the ESP32 itself.

pschatzmann, 2023-03-08:
> 

The issue is not with the InputMixer but rather with the URLStream() and how it is reaporting the availableBytes() and it seems that it does not do a blocking readBytes() even though a setTimeout() has been set

onyx99, 2023-03-08:
> I added a delay in the readBytes of the InputMixer. You can try to comment this out or decrease the delay amount

I tried, but unfortunatly that made not any noticable difference.

> The issue is not with the InputMixer but rather with the URLStream() and how it is reaporting the availableBytes() and it seems that it does not do a blocking readBytes() even though a setTimeout() has been set

Yes i also suspect, so i looked to it but my knowledge (in this library) does not go very deep.

pschatzmann, 2023-03-08:
Did you try to set the log level to Warning ? Info might have some negative impacts...

onyx99, 2023-03-08:
> Did you try to set the log level to Warning ? Info might have some negative impacts...

No not yet, i hope to look to that this evening, but i disabled the log and it did not resolve the problem, currently it just sounds like it can't catch up at higher samplerates, it plays fine upto 22050 2 channels currently.

pschatzmann, 2023-03-08:
Just keep in mind the limitations of the ESP32: https://www.pschatzmann.ch/home/2023/02/28/arduino-audiotools-esp32-vs-rp2040-pico-w/

148’400 byte / second gives a max of 37100 of stereo 2 byte samples....

onyx99, 2023-03-08:
In theory the highest possible samplerate in stereo with ESP32 that can be played without dropping frames or experiencing buffer underruns would be 44100 Hz Stereo,

It's more depending on other factors such as the complexity of the code used to play the WAV file, and the available processing power.

pschatzmann, 2023-03-08:
It depends on the storage medium: As you can see in my posted link, the TCP/IP speed is limited!

onyx99, 2023-03-08:
I will test it with an SD drive tonight, also i will do (make) a speedtest how long it takes to download the full PCM file with the ESP32.

onyx99, 2023-03-08:
Btw i also tried to decode it as an mp3 but that was not possible at all with the mixer, i got the decoded sound with alot of gaps, but when i played it trough the kit it played fine at 44100 stereo.

onyx99, 2023-03-08:
I have just tested with a Sandisc Extreme Pro 23GB SD micro card,

Because of not using Wifi connection now there were a bit more resources left, with a samplerate of 32000 in PCM format plays fine now, 44100 still jitters alot when playing it trough the mixer, when i play it directly from a file to the kit it plays fine at 44100 from SD

Below is the sketch that i wrote to test with.

/*
 Mixer SD Audiokit Example
 */

#include <SPI.h>
#include <SD.h>
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
InputMixer<int16_t> mixer;

const int chipSelect=PIN_AUDIO_KIT_SD_CARD_CS;
AudioKitStream i2s; // final output of decoded stream
File audioFile;

StreamCopy copier(i2s, mixer);

//#define sampleRate 32000
#define sampleRate 44100

void setup(){
  //Serial.begin(115200);
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);  
  
  // setup audiokit before SD!
  auto config = i2s.defaultConfig(TX_MODE);
  config.channels = 2;
  config.sample_rate = sampleRate;
  i2s.begin(config);

  // setup file
  SD.begin(chipSelect);
  
  char str[64];
  sprintf(str, "/audio/audio-%d.pcm", sampleRate);
  audioFile = SD.open(str);

  mixer.add(audioFile);
  mixer.begin(config);
  
}

void loop(){
  if (!copier.copy()) {
    stop();
  }
}

pschatzmann, 2023-03-09:
I changed the InputMixer implementation to work with  vectors of samples instead of individual samples.
The file examples is working w/o any issues now

onyx99, 2023-03-09:
> I changed the InputMixer implementation to work with vectors of samples instead of individual samples. The file example is working w/o any issues now with 44100 and 48000 stereo

Great! in a few hours (have an appointement now) i have time to test and if that indeed works the issue could be closed.

onyx99, 2023-03-09:
I just tested and confirm it indeed works perfect now, thank you very much! 

solved, so i will close the issue.

closed 2023-03-09 by onyx99