pschatzmann/arduino-audio-tools #676: Issue in record and play analog micrphone ..

aqildad-create, 2023-03-10:
I want to record approximately 20sec audio from analog microphone.  
my issue is  am getting no audio out , while if I  play the recorded audio in computer its only distortion and surpassingly the recorded duration is only 3 sec  .
recorded file size is approximately 485420

I know problem is in my code . I don't know were the problem is 

 

//#include <TFT_eSPI.h>
#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
//#include "Audio.h"
//Audio audio;
//#define SD_CS     5
//#define SD_MOSI   33
//#define SD_MISO   1
//#define SD_SCK    9
#define I2S_DOUT      25
#define I2S_BCLK      27
#define I2S_LRC       26

#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18
const uint16_t sample_rate = 44100;

//SPIClass SPISD(HSPI); // Creates SPI object using the HSPI interface

bool recording = false;
unsigned long runtime;

I2SStream i2s; // Access I2S as a stream

File audioFile; //empty audio file object

// Arduino Setup
void setup(void) {  
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode

  Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
       
  //SPI.begin(SD_SCK, SD_MISO, SD_MOSI); // Begin SPI used for SD card
 if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
  //if(!SD.begin(SD_CS, SPISD)) {
    Serial.println("SPI FAILED");
  }

  // configure I2S stream
  Serial.println("starting I2S...");

//copier.setCheckAvailableForWrite(false);

 // auto cfg = i2s.defaultConfig(RX_MODE);
//  cfg.i2s_format = I2S_STD_FORMAT; // or try with I2S_LSB_FORMAT
//  cfg.bits_per_sample = 32;
//  cfg.channels = 2;
 // cfg.sample_rate = 16000;
 // cfg.pin_ws = 14;
 // cfg.pin_bck = 12;
 // cfg.pin_data = 38;
 // i2s.begin(cfg);
AnalogAudioStream in; 
EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream
[... 160 lines cut ...]
12:42:36.039 -> [D] I2SESP32.h : 42 - bool audio_tools::I2SBase::begin(audio_tools::I2SConfig)
12:42:36.039 -> [D] I2SESP32.h : 138 - bool audio_tools::I2SBase::begin(audio_tools::I2SConfig, int, int)
12:42:36.074 -> [I] I2SConfig.h : 95 - rx/tx mode: TX
12:42:36.074 -> [I] I2SConfig.h : 96 - port_no: 0
12:42:36.074 -> [I] I2SConfig.h : 97 - is_master: Master
12:42:36.074 -> [I] I2SConfig.h : 98 - sample rate: 44100
12:42:36.074 -> [I] I2SConfig.h : 99 - bits per sample: 16
12:42:36.074 -> [I] I2SConfig.h : 100 - number of channels: 2
12:42:36.074 -> [I] I2SConfig.h : 101 - i2s_format: I2S_STD_FORMAT
12:42:36.074 -> [I] I2SConfig.h : 103 - auto_clear:1
12:42:36.074 -> [I] I2SConfig.h : 110 - buffer_count:10
12:42:36.114 -> [I] I2SConfig.h : 111 - buffer_size:512
12:42:36.114 -> [I] I2SConfig.h : 115 - pin_bck: 27
12:42:36.114 -> [I] I2SConfig.h : 116 - pin_ws: 26
12:42:36.114 -> [I] I2SConfig.h : 117 - pin_data: 25
12:42:36.114 -> [D] I2SESP32.h : 166 - i2s_driver_install
12:42:36.114 -> [D] I2SESP32.h : 183 - i2s_set_pin
12:42:36.114 -> [D] I2SESP32.h : 194 - i2s_zero_dma_buffer
12:42:36.114 -> [D] I2SESP32.h : 198 - begin - started
12:42:36.114 -> [I] AudioEncoded.h : 250 - virtual void audio_tools::EncodedAudioStream::setNotifyAudioChange(audio_tools::AudioBaseInfoDependent&)
12:42:36.154 -> [D] AudioEncoded.h : 291 - void audio_tools::EncodedAudioStream::begin()
12:42:36.154 -> [D] CodecWAV.h : 280 - virtual void audio_tools::WAVDecoder::begin()
12:42:36.154 -> [I] AudioCopy.h : 54 - buffer_size=1024
12:42:36.154 -> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
12:42:36.154 -> loop stop

pschatzmann, 2023-03-10:
This is just a big mess and impossible to understand. Try to clean up your logic first!
And try to void local variables - they might be the source of your troubles...

aqildad-create, 2023-03-10:
below is new code , speaker is not connected so cannot say about audio right now . . . 

#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
 
#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18
const uint16_t sample_rate = 44100;
 
bool recording = false;
unsigned long runtime;

I2SStream i2s; // Access I2S as a stream
AnalogAudioStream in; 
File audioFile; //empty audio file object

// Arduino Setup
void setup(void) {  
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode

  Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
       
   if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
  
    Serial.println("SPI FAILED");
  }

  // configure I2S stream
  Serial.println("starting I2S...");
 

EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream
 
// RX automatically uses port 0 with pin GPIO34
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.sample_rate = sample_rate;
  in.begin(cfgRx);
 
  // Configure out stream
  auto cfg_out = out.defaultConfig();
  cfg_out.bits_per_sample = 32;
  cfg_out.channels = 2;
  cfg_out.sample_rate = 16000;
  out.begin(cfg_out);
  //copier.setCheckAvailableForWrite(false);

 
while(runtime<30000) // loop end after 30 sec  
{

runtime = millis();

  if (!recording) {
     SD.remove("/AudioFile01.wav");
     Serial.println("FILE open");
     File   file = SD.open("/AudioFile01.wav" , FILE_READ);
     size_t fileSize = file.size();
     Serial.println("FILE SZE:"+String(fileSize));
     file.close();
    
    
    audioFile = SD.open("/AudioFile01.wav", FILE_WRITE);  // Open a new file on the SD card for writing to
[... 320 lines cut ...]
17:08:37.123 -> [D] I2SESP32.h : 87 - i2s_write 1024 -> 1024 bytes
17:08:37.123 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
17:08:37.123 -> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
17:08:37.123 -> [D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
17:08:37.123 -> [D] CodecWAV.h : 303 - virtual size_t audio_tools::WAVDecoder::write(const void*, size_t)
17:08:37.169 -> [D] I2SStream.h : 125 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
17:08:37.169 -> [D] I2SESP32.h : 80 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
17:08:37.169 -> [D] I2SESP32.h : 87 - i2s_write 1024 -> 1024 bytes
17:08:37.169 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
17:08:37.169 -> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
17:08:37.169 -> [D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
17:08:37.216 -> [D] CodecWAV.h : 303 - virtual size_t audio_tools::WAVDecoder::write(const void*, size_t)
17:08:37.216 -> [D] I2SStream.h : 125 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
17:08:37.216 -> [D] I2SESP32.h : 80 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
17:08:37.216 -> [D] I2SESP32.h : 87 - i2s_write 1024 -> 1024 bytes
17:08:37.216 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
17:08:37.216 -> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
17:08:37.264 -> [D] AudioEncoded.h : 327 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 44
17:08:37.264 -> [D] CodecWAV.h : 303 - virtual size_t audio_tools::WAVDecoder::write(const void*, size_t)
17:08:37.264 -> [D] I2SStream.h : 125 - virtual size_t audio_tools::I2SStream::write(const uint8_t*, size_t)
17:08:37.264 -> [D] I2SESP32.h : 80 - size_t audio_tools::I2SBase::writeBytes(const void*, size_t)
17:08:37.264 -> [D] I2SESP32.h : 87 - i2s_write 44 -> 44 bytes
17:08:37.264 -> [I] AudioCopy.h : 124 - StreamCopy::copy 44 -> 44 -> 44 bytes - in 1 hops
17:08:37.311 -> [D] AudioCopy.h : 91 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
17:08:37.311 -> loop stop

pschatzmann, 2023-03-10:
I think your checks for the runtime are completly wrong: you should use some timeout calculation using the start time: So the end time would be millis()+2000;

aqildad-create, 2023-03-10:
yes , i know 10 additional seconds is just delay in which i read file size after each 5 seconds .  

THS IS ADDTONAL 10 SECONDS . RECORDNG IS ONLY FOR 20 SECONDS . 

17:07:55.750 -> [I] AudioCopy.h : 124 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
17:07:55.795 -> FILE CLOSED
17:07:55.795 -> FILE SZE:485420
17:07:55.795 -> FILE END
17:08:00.781 -> FILE CLOSED
17:08:00.828 -> FILE SZE:485420
17:08:00.828 -> FILE END
17:08:05.815 -> FILE CLOSED
17:08:05.815 -> FILE SZE:485420
17:08:05.815 -> FILE END
17:08:10.856 -> [D] CodecWAV.h : 240 - audio_tools::WAVDecoder::WAVDecoder()
17:08:10.856 -> [D] AudioEncoded.h : 146 - audio_tools::EncodedAudioStream::EncodedAudioStream(audio_tools::AudioStream*, audio_tools::AudioDecoder*)

pschatzmann, 2023-03-10:
You can simplify your logic if you calcualte how many copy operations you want to do. 5 seconds of 16000 with 2 channels gives 
320000 bytes. If you use the default copy buffer size of 1024 bytes this gives about 312 copies: so you can just call copier.copyN(312);

aqildad-create, 2023-03-10:
Yes the file size I get  after 312 attempts   is 319532.

for( int xyz=0;xyz<312;xyz++)
copier.copy();

as far file size is  think its ok 1024 x 312 = 319,488  with difference of 44 bytes . ( I don't know why this difference ) 

Also I don't have speaker connected so can't say about audio out right now .

pschatzmann, 2023-03-10:
The WAV header is 44 bytes

aqildad-create, 2023-03-10:
I was getting error while using copier.copyN(312);  ( copyN not found ) .

Also  I Play recorded file in computer were audio duration is only 2second   and its very distorted , why ?

pschatzmann, 2023-03-10:
You must be using a very old version: here is the actual documentation of the class: https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_stream_copy.html

aqildad-create, 2023-03-10:
I download latest version , now copier.copyN(312);  is working but it almost take 500ms of copying single attempt . 

see the logs , 
further  I am testing other parameter of it .

0:41:59.485 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:41:59.996 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:00.511 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:00.977 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:01.490 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:02.005 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:02.473 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:02.982 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:03.495 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:04.008 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:04.476 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:04.993 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:05.507 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:06.020 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:06.485 -> [D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
20:42:06.

aqildad-create, 2023-03-10:
now i go back to 
for( int xyz=0;xyz<312;xyz++)
copier.copy();
with latest version . file is copy and size is correct  but when  play it in computer its only 2 second file , why ?

aqildad-create, 2023-03-10:
In audacity if open it as raw data then file duration is 5sec and if open as wav then duration is about 2seconds

closed 2023-03-10 by pschatzmann

aqildad-create, 2023-03-12:
sir   , i attach the function generator with adc pin and record file and open the file in audacity . my question is why the file amplitude is not same or very low as compare to adc pin ? 

how to increase the amplitude in file ?

#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
 
#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18
const uint16_t sample_rate = 44100;
 
bool recording = false;
unsigned long runtime;

I2SStream i2s; // Access I2S as a stream
AnalogAudioStream in; 
File audioFile; //empty audio file object
//ConverterAutoCenter<int16_t> converter;

ConverterScaler<int16_t> scaler(1.0, -26427, 32700 );

// Arduino Setup
void setup(void) {  
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode

  Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
       
   if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
  
    Serial.println("SPI FAILED");
  }

  // configure I2S stream
  Serial.println("starting I2S...");
 

EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream
 
// RX automatically uses port 0 with pin GPIO34
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.bits_per_sample = 16;
  cfgRx.sample_rate = sample_rate;
  
  in.begin(cfgRx);
 
  // Configure out stream
  auto cfg_out = out.defaultConfig();
  cfg_out.bits_per_sample = 16;//32;
  cfg_out.channels = 2;
  cfg_out.sample_rate = sample_rate;
  out.begin(cfg_out);

  
   SD.remove("/AudioFile01.wav");
     Serial.println("FILE open");
     File   file = SD.open("/AudioFile01.wav" , FILE_READ);
     size_t fileSize = file.size();
     Serial.println("FILE SZE:"+String(fileSize));
     file.close();
    
    
    audioFile = SD.open("/AudioFile01.wav", FILE_WRITE);  // Open a new file on the SD card for writing to
    copier.begin(out, in);  // Configure copier input and output
    Serial.println("RECORDING");
  
  
  
  
  
  
  
  
  copier.setCheckAvailableForWrite(false);
 copier.begin(out, in);
 for(byte xyz=0;xyz<=250;xyz++)
 copier.copy(scaler);
 for(byte xyz=0;xyz<=250;xyz++)
 copier.copy(scaler);
 
 audioFile.close();
  Serial.println("FILE open");
     File   file2 = SD.open("/AudioFile01.wav" , FILE_READ);
     size_t fileSize2 = file2.size();
     Serial.println("FILE SZE:"+String(fileSize2));
     file2.close();

delay(10000);
 
}
void loop()
{
  
  
[image]

}

pschatzmann, 2023-03-12:
You can make it any value range with the ConverterScaler!