pschatzmann/arduino-audio-tools #679: streams-i2s-webserver wav.ino

shri360, 2023-03-12:
Hi,

Can I connect KY-037 microphone module instead of INMP441?

if possible please share connection details for KY-037 module.

Thanks.

pschatzmann, 2023-03-12:
Just look at the examples that start with streams-adc...

closed 2023-03-12 by pschatzmann

shri360, 2023-03-12:
yes, but i want stream in web browser same as stream-i2c.

pschatzmann, 2023-03-12:
Yes, so you just change the source !
But first you need to confirm that your audio is working with expected...

shri360, 2023-03-12:
yes, audio working as expected, i can see the response in serial plotter.

shri360, 2023-03-12:
can you please help me with changing source of input?

pschatzmann, 2023-03-12:
Just replace the I2SStream class to AnalogAudioStream and change the channels to 1.

shri360, 2023-03-12:
`/**
 * @file streams-i2s-webserver_wav.ino
 *
 *  This sketch reads sound data from I2S. The result is provided as WAV stream which can be listened to in a Web Browser
 *
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("ssid","password"); // the same a above

I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 22050;
  config.channels = 2;
  config.bits_per_sample = 16;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();  
}
`

Please tell me where exactly changes to be done, to change audio input source to analougeAudioStream.

shri360, 2023-03-12:
"replace the I2SStream class with AnalogAudioStream and change the channels to 1."

i tried to change as your suggestion but i'm getting compile error.

shri360, 2023-03-13:
@pschatzmann yey!! working perfect modified source to analog Audio. appreciated for your help.