pschatzmann/arduino-audio-tools #680: AudioTools.h incorrectly defines

LenStruttmann, 2023-03-14:
In AudioTools.h, the symbol LED_BUILTIN is defined in lines 220-223 and 282-285.  This causes an issue when compiling for the board "lolin_s2_mini" using platformIO.

Is it really appropriate for AudioTools.h to even be defining that symbol?  Should not it rely on the default definition provided in pins_arduino.h?

Thanks,

Len

pschatzmann, 2023-03-14:
This definition is only used when LED_BUILTIN is not defined!  
So I don't understand why this would be an issue: please explain.

LenStruttmann, 2023-03-14:
1. When I #include "AudioTools.h", as is, LED_BUILTIN gets incorrectly defined as 13. VisualStudio identifies the definition as coming from AudioConfig.h.

2. If I modify AudioConfig.h to remove the #define LED_BUILTIN lines, LED_BUILTIN is correctly defined to 15 by pins_arduino.h.

This is because pins_arduino.h does NOT define the symbol LED_BUILTIN but, instead allocates an int for it:

`static const uint8_t LED_BUILTIN = 15;`

Therefore, in AudioConfig.h, the test:

`#ifndef LED_BUILTIN` is true, since there is no LED_BUILTIN symbol defined.

pschatzmann, 2023-03-14:
Oh good, why would anybody do something like this and define LED_BUILTIN as const!
I commented them out in my code...

closed 2023-03-14 by pschatzmann

LenStruttmann, 2023-03-14:
Thanks for making that change.

As for why, for most boards supported by Platformio, it appears that it is defined thus:

`heltec_wifi_kit_32_v3/pins_arduino.h:static const uint8_t LED_BUILTIN = 35;
heltec_wifi_kit_32_v3/pins_arduino.h:#define BUILTIN_LED  LED_BUILTIN // backward compatibility
heltec_wifi_kit_32_v3/pins_arduino.h:#define LED_BUILTIN LED_BUILTIN`

...which works with your code.

There are some boards, though, that omit that second define, which doesn't play well with your code.

BTW, AudioTools is great! I'm using it to develop a theater prop "amulet" that has lights and sound, for a local community theater production of "Peter and the Starcatcher".