pschatzmann/arduino-audio-tools #683: @file streams-i2s-i2s-2.ino .. esp32 reset .. change i2s pins

aqildad-create, 2023-03-15:
I only change i2s pins in this example and  esp32 reset continuously .. 

these 2 test working successfully . 
1:i2s webserver is working.. 
2:wav from sd is also working.

code
/**
 * @file streams-i2s-i2s-2.ino
 * @brief Copy audio from I2S to I2S: We use 2 different i2s ports!
 * @author Phil Schatzmann
 * @copyright GPLv3
 */

#include "AudioTools.h"
 
uint16_t sample_rate=44100;
uint16_t channels = 2;
uint16_t bits_per_sample = 16; // or try with 24 or 32

I2SStream in;
I2SStream out; 
StreamCopy copier(out, in); // copies sound into i2s

// Arduino Setup
void setup(void) {  
  // Open Serial 
  Serial.begin(115200);
  // change to Warning to improve the quality
  AudioLogger::instance().begin(Serial, AudioLogger::Info); 

  // start I2S in
  Serial.println("starting I2S...");
  auto config_in = in.defaultConfig(RX_MODE);
  config_in.sample_rate = sample_rate; 
  config_in.bits_per_sample = bits_per_sample; 
  config_in.i2s_format = I2S_STD_FORMAT;
  config_in.is_master = true;
  config_in.port_no = 0;
  config_in.pin_bck = 11;//14;
  config_in.pin_ws = 15;
  config_in.pin_data = 32;//22;
  // config_in.fixed_mclk = sample_rate * 256
  // config_in.pin_mck = 2
  in.begin(config_in);

  // start I2S out
  auto config_out = out.defaultConfig(TX_MODE);
  config_out.sample_rate = sample_rate; 
  config_out.bits_per_sample = bits_per_sample;
  config_out.i2s_format = I2S_STD_FORMAT;
  config_out.is_master = true;
  config_out.port_no = 1;
  config_out.pin_ws = 26;
  config_out.pin_bck = 27;
  config_out.pin_data = 25;
  
  //config_out.pin_bck = 18;
  //config_out.pin_ws = 19;
  //config_out.pin_data = 23;
  
  
  
  out.begin(config_out);

  Serial.println("I2S started...");
}

// Arduino loop - copy sound to out 
void loop() {
  copier.copy();
}

/////////////

3:56:39.914 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:56:39.914 -> configsip: 0, SPIWP:0xee
13:56:39.914 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:56:39.914 -> mode:DIO, clock div:1
13:56:39.914 -> load:0x3fff0030,len:1344
13:56:39.914 -> load:0x40078000,len:13864
13:56:39.914 -> load:0x40080400,len:3608
13:56:39.914 -> entry 0x400805f0
13:56:40.054 -> starting I2S...
13:56:40.054 -> [I] I2SConfig.h : 104 - rx/tx mode: RX_MODE
13:56:40.054 -> [I] I2SConfig.h : 105 - port_no: 0
13:56:40.054 -> [I] I2SConfig.h : 106 - is_master: Master
13:56:40.054 -> [I] I2SConfig.h : 107 - sample rate: 44100
13:56:40.054 -> [I] I2SConfig.h : 108 - bits per sample: 16
13:56:40.054 -> [I] I2SConfig.h : 109 - number of channels: 2
13:56:40.054 -> [I] I2SConfig.h : 110 - i2s_format: I2S_STD_FORMAT
13:56:40.054 -> [I] I2SConfig.h : 112 - auto_clear:0
13:56:40.100 -> [I] I2SConfig.h : 119 - buffer_count:10
13:56:40.100 -> [I] I2SConfig.h : 120 - buffer_size:512
13:56:40.100 -> [I] I2SConfig.h : 124 - pin_bck: 11
13:56:40.100 -> [I] I2SConfig.h : 126 - pin_ws: 15
13:56:40.100 -> [I] I2SConfig.h : 128 - pin_data: 32
13:56:40.985 -> ets Jul 29 2019 12:21:46
13:56:40.985 -> 
13:56:40.985 -> rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
13:56:40.985 -> configsip: 0, SPIWP:0xee
13:56:40.985 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
13:56:40.985 -> mode:DIO, clock div:1
13:56:40.985 -> load:0x3fff0030,len:1344
13:56:41.032 -> load:0x40078000,len:13864
13:56:41.032 -> load:0x40080400,len:3608
13:56:41.032 -> entry 0x400805f0
13:56:41.124 -> starting I2S...
13:56:41.170 -> [I] I2SConfig.h : 104 - rx/tx mode: RX_MODE
13:56:41.170 -> [I] I2SConfig.h : 105 - port_no: 0
13:56:41.170 -> [I] I2SConfig.h : 106 - is_master: Master
13:56:41.170 -> [I] I2SConfig.h : 107 - sample rate: 44100
13:56:41.170 -> [I] I2SConfig.h : 108 - bits per sample: 16
13:56:41.170 -> [I] I2SConfig.h : 109 - number of channels: 2
13:56:41.170 -> [I] I2SConfig.h : 110 - i2s_format: I2S_STD_FORMAT
13:56:41.170 -> [I] I2SConfig.h : 112 - auto_clear:0

aqildad-create, 2023-03-15:
issue with the power supply(same power supply for i2s web server and wav player both work . so no issue with power supply ). 
using wrong pins  ( Microphone is Working WITH WEB SERVER AND I2S DAC IS WORNG WTH WAV PLAYER . SO NO ISSUE WTH PINS OUT .

Did you try with the original pins ?
nO  I don't because microphone and i2s dac is already soldered .

pschatzmann, 2023-03-15:
Did you try to power the external DAC and external ADC from an external source ?