pschatzmann/arduino-audio-tools #685: adc-webserver

aqildad-create, 2023-03-16:
i just change i2sstream to analog and set channel to one . . . compile success but no audio out ..is this because slow wifi or what ?  

#include "AudioTools.h"

const uint16_t sample_rate = 44100;
AnalogAudioStream in; 
I2SStream out;                        
StreamCopy copier(out, in); // copy in to out
ConverterAutoCenter<int16_t> converter;

// Arduino Setup
void setup(void) {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // RX automatically uses port 0 with pin GPIO34
  auto cfgRx = in.defaultConfig(RX_MODE);
  cfgRx.sample_rate = sample_rate;
  in.begin(cfgRx);
 
  // TX on I2S_NUM_1 
  auto cfgTx = out.defaultConfig(TX_MODE);
  cfgTx.port_no = 1;
  cfgTx.sample_rate = sample_rate;
  out.begin(cfgTx);
}

// Arduino loop - copy data 
void loop() {
  copier.copy(converter);
}

//--------------------
10:37:30.117 -> IP address: 192.168.2.107
10:37:39.808 -> [I] AudioServer.h : 216 - New Client.
10:37:48.150 -> [I] AudioServer.h : 368 - sendReply - Returning encoded stream...
10:37:48.150 -> [I] CodecWAV.h : 452 - sample_rate: 22050
10:37:48.150 -> [I] CodecWAV.h : 453 - channels: 1
10:37:48.197 -> [I] CodecWAV.h : 457 - is_streamed! because length is 4294967295
10:37:48.197 -> [I] AudioCopy.h : 71 - buffer_size=1024
10:37:48.197 -> [I] CodecWAV.h : 501 - Writing Header
10:37:48.197 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:48.197 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:48.197 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:57.073 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:57.073 -> [I] AudioServer.h : 122 - stop client...
10:37:57.120 -> [I] AudioServer.h : 216 - New Client.
10:37:57.120 -> [I] AudioServer.h : 368 - sendReply - Returning encoded stream...
10:37:57.120 -> [I] CodecWAV.h : 452 - sample_rate: 22050
10:37:57.120 -> [I] CodecWAV.h : 453 - channels: 1
10:37:57.120 -> [I] CodecWAV.h : 457 - is_streamed! because length is 4294967295
10:37:57.120 -> [I] AudioCopy.h : 71 - buffer_size=1024
10:37:57.120 -> [I] CodecWAV.h : 501 - Writing Header
10:37:57.166 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:57.166 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:37:57.166 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:04.309 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:04.356 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:04.356 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:05.856 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:05.856 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:05.856 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:38:25.882 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
10:38:35.857 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:38:45.895 -> [I] AudioCopy.h : 292 - try write - 4 (open 1024 bytes) 
10:38:55.863 -> [I] AudioCopy.h : 292 - try write - 5 (open 1024 bytes) 
10:39:05.879 -> [I] AudioCopy.h : 292 - try write - 6 (open 1024 bytes) 
10:39:15.911 -> [I] AudioCopy.h : 292 - try write - 7 (open 1024 bytes) 
10:39:25.903 -> [I] AudioCopy.h : 292 - try write - 8 (open 1024 bytes) 
10:39:35.879 -> [I] AudioCopy.h : 292 - try write - 9 (open 1024 bytes) 
[... 43 lines cut ...]
10:43:00.317 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:43:03.403 -> [I] AudioCopy.h : 292 - try write - 4 (open 0 bytes) 
10:43:03.403 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 4 hops
10:43:03.403 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:03.403 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:23.421 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
10:43:33.431 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:43:33.850 -> [I] AudioCopy.h : 292 - try write - 4 (open 0 bytes) 
10:43:33.850 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 4 hops
10:43:33.850 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:33.898 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:33.944 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:33.944 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:43:53.952 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
10:44:03.947 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:44:13.953 -> [I] AudioCopy.h : 292 - try write - 4 (open 1024 bytes) 
10:44:23.959 -> [I] AudioCopy.h : 292 - try write - 5 (open 1024 bytes) 
10:44:33.958 -> [I] AudioCopy.h : 292 - try write - 6 (open 1024 bytes) 
10:44:43.957 -> [I] AudioCopy.h : 292 - try write - 7 (open 1024 bytes) 
10:44:53.958 -> [I] AudioCopy.h : 292 - try write - 8 (open 1024 bytes) 
10:45:03.960 -> [I] AudioCopy.h : 292 - try write - 9 (open 1024 bytes) 
10:45:13.966 -> [I] AudioCopy.h : 292 - try write - 10 (open 1024 bytes) 
10:45:23.964 -> [I] AudioCopy.h : 292 - try write - 11 (open 1024 bytes) 
10:45:33.959 -> [I] AudioCopy.h : 292 - try write - 12 (open 1024 bytes) 
10:45:43.977 -> [I] AudioCopy.h : 292 - try write - 13 (open 1024 bytes)

pschatzmann, 2023-03-16:
[It's not working](https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working)
by the way your issues does not make any sense, your log has nothing to do with your sketch...

closed 2023-03-16 by pschatzmann

aqildad-create, 2023-03-16:
adc to serial is working and get proper wave . . so what else should  need to trouble shoot ? 

some time i listen only wrestle in browser ? 

10:56:01.689 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:01.969 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:01.969 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:01.969 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.634 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.634 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.681 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.681 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.728 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.728 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.774 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.820 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.820 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.867 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.914 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:09.914 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:10.009 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:10.009 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:10.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:10.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:10.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:56:30.099 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
10:56:40.100 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:56:50.098 -> [I] AudioCopy.h : 292 - try write - 4 (open 1024 bytes) 
10:57:00.138 -> [I] AudioCopy.h : 292 - try write - 5 (open 1024 bytes) 
10:57:10.107 -> [I] AudioCopy.h : 292 - try write - 6 (open 1024 bytes) 
10:57:20.153 -> [I] AudioCopy.h : 292 - try write - 7 (open 1024 bytes) 
10:57:30.142 -> [I] AudioCopy.h : 292 - try write - 8 (open 1024 bytes) 
10:57:40.158 -> [I] AudioCopy.h : 292 - try write - 9 (open 1024 bytes) 
10:57:50.124 -> [I] AudioCopy.h : 292 - try write - 10 (open 1024 bytes) 
10:58:00.134 -> [I] AudioCopy.h : 292 - try write - 11 (open 1024 bytes) 
10:58:10.130 -> [I] AudioCopy.h : 292 - try write - 12 (open 1024 bytes) 
10:58:20.162 -> [I] AudioCopy.h : 292 - try write - 13 (open 1024 bytes) 
10:58:30.161 -> [I] AudioCopy.h : 292 - try write - 14 (open 1024 bytes) 
10:58:35.723 -> [I] AudioCopy.h : 292 - try write - 15 (open 0 bytes) 
10:58:35.723 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 15 hops
10:58:35.723 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:58:36.332 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
10:58:56.366 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
10:59:06.369 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
10:59:16.374 -> [I] AudioCopy.h : 292 - try write - 4 (open 1024 bytes) 
10:59:26.364 -> [I] AudioCopy.h : 292 - try write - 5 (open 1024 bytes) 
10:59:36.397 -> [I] AudioCopy.h : 292 - try write - 6 (open 1024 bytes) 
10:59:46.390 -> [I] AudioCopy.h : 292 - try write - 7 (open 1024 bytes) 
10:59:56.396 -> [I] AudioCopy.h : 292 - try write - 8 (open 1024 bytes)

pschatzmann, 2023-03-16:
Did you try to decrease the sample rate ?
If you used a converter when you tested with the CsvStream, did you use it in the Webserver as well ?

aqildad-create, 2023-03-16:
apology : below was my code  above  I posted wrong code . below is the code  i compiled for adc webserver  
#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("Boss net (67) 03062689075","36879250"); // the same a above
AnalogAudioStream i2sStream;//in; 
//I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
//  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 22050;
  config.channels = 1;//2;
  config.bits_per_sample = 16;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();

pschatzmann, 2023-03-16:
It seems that you use different converters: Test again and make sure that the signal is centered around zero!
Then use the same converter in the webserver!
Decrease the sample rate even more until you can consistently write w/o problems...

aqildad-create, 2023-03-16:
yes I decrease the sampling rate to 8000 and use ConverterAutoCenter , now Its writing success but  i only listen distorted wrestle  in browser . 

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("Boss net (67) 03062689075","36879250"); // the same a above
AnalogAudioStream i2sStream;//in; 
//I2SStream i2sStream;    // Access I2S as stream
ConverterAutoCenter<int16_t> converter; 

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
//  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 8000;//22050;
  config.channels = 1;//2;
  config.bits_per_sample = 16;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &converter);// server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();  
}

11:23:39.789 -> starting I2S...
11:23:39.789 -> [I] AnalogAudioBase.h : 54 - I2S_MODE_ADC_BUILT_IN
11:23:39.789 -> [I] AnalogAudio.h : 57 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
11:23:39.789 -> [I] AnalogAudioESP32.h : 48 - virtual bool audio_tools::AnalogDriverESP32::begin(audio_tools::AnalogConfig)
11:23:39.789 -> [I] AudioTypes.h : 63 - sample_rate: 8000
11:23:39.836 -> [I] AudioTypes.h : 64 - channels: 1
11:23:39.836 -> [I] AudioTypes.h : 65 - bits_per_sample: 16
11:23:39.836 -> [I] AnalogAudioBase.h : 75 - input pin1: 34
11:23:39.836 -> [I] AnalogAudioESP32.h : 90 - RX_MODE
11:23:39.836 -> I2S started
11:23:39.836 -> [I] CodecWAV.h : 452 - sample_rate: 8000
11:23:39.836 -> [I] CodecWAV.h : 453 - channels: 1
11:23:39.836 -> [I] CodecWAV.h : 457 - is_streamed! because length is 2147418112
11:23:39.945 -> .....
11:23:42.461 -> IP address: 192.168.2.107
11:24:14.740 -> [I] AudioServer.h : 216 - New Client.
11:24:14.786 -> [I] AudioServer.h : 368 - sendReply - Returning encoded stream...
11:24:14.786 -> [I] CodecWAV.h : 452 - sample_rate: 8000
11:24:14.786 -> [I] CodecWAV.h : 453 - channels: 1
11:24:14.786 -> [I] CodecWAV.h : 457 - is_streamed! because length is 4294967295
11:24:14.833 -> [I] AudioCopy.h : 71 - buffer_size=1024
11:24:14.833 -> [I] CodecWAV.h : 501 - Writing Header
11:24:14.833 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.833 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.833 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.833 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.833 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.880 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.880 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.880 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.880 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.880 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:24:14.927 -> [I] AudioCopy.h : 292 - try write - 2 (open 1024 bytes) 
11:24:14.927 -> [I] AudioCopy.h : 292 - try write - 3 (open 1024 bytes) 
11:24:14.927 -> [I] AudioCopy.h : 292 - try write - 4 (open 1024 bytes) 
11:24:14.927 -> [I] AudioCopy.h : 292 - try write - 5 (open 1024 bytes) 
[... 189 lines cut ...]
11:25:46.959 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:46.959 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:46.959 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:46.959 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:46.959 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.006 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.006 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.006 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.006 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.053 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.053 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.053 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.053 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.100 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[... cut ...]
11:25:47.193 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.193 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.193 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.239 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.239 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
11:25:47.239 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

pschatzmann, 2023-03-16:
Don't forget to set the logger back to warning after you made sure that you have a proper signal by verifying it with a CsvStream....

pschatzmann, 2023-03-16:
But this does not look like a sound wave at all!

aqildad-create, 2023-03-16:
I use MAX9814 auto gain control .  I was very close to it that's why it look like square wave , may be .

pschatzmann, 2023-03-16:
For my tests I was using a MCP6022 Microphone Sensor. 
I never tried this one, but I would be surprised if this one wouldn't work as well...
How did you connect it ?

aqildad-create, 2023-03-16:
connection 
MAX9814 -ESP32
VDD         -3V
GND         -GND
OUT          -GPIO34

aqildad-create, 2023-03-27:
MAX9814  can you give me idea how to handle this microphone ?

aqildad-create, 2023-03-27:
AFTER 0.01 below is plotter result 
ConverterScaler<int16_t> scaler(0.01, -26427, 32700 );

[image]

aqildad-create, 2023-03-27:
AFTER 0.001 below is plotter result ..which is not good
ConverterScaler<int16_t> scaler(0.001, -26427, 32700 );

[image]

pschatzmann, 2023-03-27:
Try to record a sine tone: e.g. from https://onlinetonegenerator.com/
Then you should also see a sine on the plotter.
As a result the goal should be that it is oscillating around 0 with an amplitude  of 20000 to 30000 if you want to hear the singnal properly