pschatzmann/arduino-audio-tools #688: 2s recorder

aqildad-create, 2023-03-17:
Mic - INMP441 .. file is recorded success , but when  I play the recorded file in audacity it is very distorted . 

Below test successfully work .
I2s -plotter 
I2s web server 

Attached here is recorded file  and its image from audacity . 
.

#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
 
#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18
const uint16_t sample_rate = 44100;
 
bool recording = false;
unsigned long runtime;

I2SStream in; // Access I2S as a stream
//AnalogAudioStream in; 
File audioFile; //empty audio file object

// Arduino Setup
void setup(void) {  
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode

  Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
       
   if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
  
    Serial.println("SPI FAILED");
  }

  // configure I2S stream
  Serial.println("starting I2S...");
 

EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream
 
// RX automatically uses port 0 with pin GPIO34
 // auto cfgRx = in.defaultConfig(RX_MODE);
 // cfgRx.sample_rate = sample_rate;
 // in.begin(cfgRx);

Serial.println("starting I2S...");
  auto config = in.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 44100;
  config.channels = 1;
  config.bits_per_sample = 32;
  in.begin(config);
  Serial.println("I2S started");

 
  // Configure out stream
  auto cfg_out = out.defaultConfig();
  cfg_out.bits_per_sample = 32;
  cfg_out.channels = 1;
  cfg_out.sample_rate = 44100;
  out.begin(cfg_out);
  //copier.setCheckAvailableForWrite(false);
[... 202 lines cut ...]

13:19:16.587 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.587 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.587 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
13:19:16.634 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.634 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.634 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
13:19:16.634 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.682 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.682 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
13:19:16.682 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.682 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.682 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
13:19:16.682 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.729 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.729 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
13:19:16.729 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
13:19:16.729 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
13:19:16.776 -> FILE CLOSED
[AudioFile01.zip](https://github.com/pschatzmann/arduino-audio-tools/files/10999375/AudioFile01.zip)

13:19:16.776 -> FILE SZE:1060908
13:19:16.776 -> FILE END

[image]

pschatzmann, 2023-03-17:
Did you set the logger to warning ?

aqildad-create, 2023-03-17:
No , but if you see the recorded wav the distortion is not at that time when silent , the time when  i speak that is very distorted  ,
https://user-images.githubusercontent.com/65456998/225852589-dc5345e3-f51d-4e2a-837d-09f2463f1845.png

pschatzmann, 2023-03-17:
You talk nonsense: silence can't be distorted!

closed 2023-03-17 by aqildad-create

aqildad-create, 2023-03-17:
For Mp3 recordng  change 
EncodedAudioStream out(&audioFile, new WAVEncoder());
to 
EncodedAudioStream out(&audioFile, new MP3DecoderHelix());
and add #include "AudioCodecs/CodecMP3Helix.h"

but file size is not recorded and i got file size 0 ..

#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
#include "AudioCodecs/CodecMP3Helix.h"
 
#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK  18
const uint16_t sample_rate = 44100;
 
bool recording = false;
unsigned long runtime;

I2SStream in; // Access I2S as a stream
//AnalogAudioStream in; 
File audioFile; //empty audio file object

// Arduino Setup
void setup(void) {  
  Serial.begin(115200);

  AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode
//AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
       
   if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {
  
    Serial.println("SPI FAILED");
  }

  // configure I2S stream
  Serial.println("starting I2S...");
 

//EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
EncodedAudioStream out(&audioFile, new MP3DecoderHelix()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream
 
// RX automatically uses port 0 with pin GPIO34
 // auto cfgRx = in.defaultConfig(RX_MODE);
 // cfgRx.sample_rate = sample_rate;
 // in.begin(cfgRx);

Serial.println("starting I2S...");
  auto config = in.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = 44100;
  config.channels = 1;
  config.bits_per_sample = 32;
  in.begin(config);
  Serial.println("I2S started");

 
  // Configure out stream
  auto cfg_out = out.defaultConfig();
  cfg_out.bits_per_sample = 32;
  cfg_out.channels = 1;
  cfg_out.sample_rate = 44100;
[... 158 lines cut ...]
22:18:03.979 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.166 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.166 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.215 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.215 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.351 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.351 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.351 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.397 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.865 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.865 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.865 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.865 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:05.695 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:05.695 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
22:18:05.695 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:05.695 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:06.577 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:06.577 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
22:18:06.577 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:06.577 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:07.411 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:07.411 -> FILE CLOSED
22:18:07.411 -> FILE SZE:0
22:18:07.411 -> FILE END

ksaye, 2023-04-11:
@aqildad-create or @pschatzmann , do you have a working sample code?

I tried the following:

    #include "Arduino.h"
    #include "AudioTools.h"
    #include "SD.h"
    
    #define MicPIN_WS GPIO_NUM_32
    #define MicPIN_SCK GPIO_NUM_33
    #define MicPIN_SD GPIO_NUM_35
    
    bool recording = false;
    unsigned long runtime;
    
    I2SStream in; // Access I2S as a stream
    File audioFile; //empty audio file object
    
    // Arduino Setup
    void setup(void) {
      Serial.begin(115200);
    
      //AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode
    
      Serial.println("SERIAL STARTED");
    
      if (!SD.begin()) {
    
      Serial.println("SPI FAILED");
      }
    
      // configure I2S stream
      Serial.println("starting I2S...");
    
      EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
      StreamCopy copier(out, in); // copies sound from i2s to audio file stream
       
      Serial.println("starting I2S...");
      auto config = in.defaultConfig(RX_MODE);
      config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT #23
      config.sample_rate = 16000;
      config.channels = 2;
      config.bits_per_sample = 16;
      config.pin_ws = MicPIN_WS;       //WS
      config.pin_bck = MicPIN_SCK;     //SCK
      config.pin_data = MicPIN_SD;     //SD
      in.begin(config);
      Serial.println("I2S started");
    
      // Configure out stream
      auto cfg_out = out.defaultConfig();
      cfg_out.bits_per_sample = 16;
      cfg_out.channels = 2;
      cfg_out.sample_rate = 16000;
      out.begin(cfg_out);
      copier.setCheckAvailableForWrite(false);
    
      SD.remove("/AudioFile01.wav");
      audioFile = SD.open("/AudioFile01.wav", FILE_WRITE); // Open a new file on the SD card for writing to
      copier.begin(out, in); // Configure copier input and output
      Serial.println("RECORDING");
      
      while(runtime<30000) // loop end after 30 sec
        {
          runtime = millis();
          copier.copy(); // Copy sound from I2S to the audio file
        }
        
      audioFile.close(); // Closes the file after 5 seconds
      Serial.println("FILE CLOSED");
    
      Serial.println("FILE END");
      delay(30000);
      Serial.println("End of delay");
    }
    
    void loop()
    {
      delay(1000);
    }

Which produced the following wav file, which is clearly not 30 seconds:

[AudioFile01 (1).zip](https://github.com/pschatzmann/arduino-audio-tools/files/11203999/AudioFile01.1.zip)

The properties of the file are:
[image]

I am using an ESP32, a SD Card reader and a simple INMP441 Mems Microphone.

pschatzmann, 2023-04-11:
Your logic around the runtime is completely wrong:  you will need to record the start time and add the timeout to it to determine the end time!