pschatzmann/arduino-audio-tools #691: I2s recording mp3 .

aqildad-create, 2023-03-18:
i2s to wave recording is working now for For Mp3 recording i change
EncodedAudioStream out(&audioFile, new WAVEncoder());
to
EncodedAudioStream out(&audioFile, new MP3DecoderHelix());
and add #include "AudioCodecs/CodecMP3Helix.h"

but file size is not recorded and i got file size 0 .. I don't know why is this ..

#include <SPI.h>
#include "Arduino.h"
#include "AudioTools.h"
#include "SD.h"
#include "AudioCodecs/CodecMP3Helix.h"

#define PIN_AUDIO_KIT_SD_CARD_CS 5
#define PIN_AUDIO_KIT_SD_CARD_MISO 19
#define PIN_AUDIO_KIT_SD_CARD_MOSI 23
#define PIN_AUDIO_KIT_SD_CARD_CLK 18
const uint16_t sample_rate = 44100;

bool recording = false;
unsigned long runtime;

I2SStream in; // Access I2S as a stream
//AnalogAudioStream in;
File audioFile; //empty audio file object

// Arduino Setup
void setup(void) {
Serial.begin(115200);

AudioLogger::instance().begin(Serial, AudioLogger::Debug); // AudioLogger set to Debug mode
//AudioLogger::instance().begin(Serial, AudioLogger::Warning);

Serial.println("SERIAL STARTED");
SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);

if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS)) {

Serial.println("SPI FAILED");
}

// configure I2S stream
Serial.println("starting I2S...");

//EncodedAudioStream out(&audioFile, new WAVEncoder()); // Access audioFile as a stream
EncodedAudioStream out(&audioFile, new MP3DecoderHelix()); // Access audioFile as a stream
StreamCopy copier(out, in); // copies sound from i2s to audio file stream

// RX automatically uses port 0 with pin GPIO34
// auto cfgRx = in.defaultConfig(RX_MODE);
// cfgRx.sample_rate = sample_rate;
// in.begin(cfgRx);

Serial.println("starting I2S...");
auto config = in.defaultConfig(RX_MODE);
config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
config.sample_rate = 44100;
config.channels = 1;
config.bits_per_sample = 32;
in.begin(config);
Serial.println("I2S started");

// Configure out stream
auto cfg_out = out.defaultConfig();
cfg_out.bits_per_sample = 32;
cfg_out.channels = 1;
cfg_out.sample_rate = 44100;
out.begin(cfg_out);
copier.setCheckAvailableForWrite(false);
[... 149 lines cut ...]
22:18:03.979 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.166 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.166 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.215 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.215 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.351 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.351 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.351 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.397 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:04.865 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:04.865 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT::copy() [with T = unsigned char; size_t = unsigned int]
22:18:04.865 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:04.865 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:05.695 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:05.695 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT::copy() [with T = unsigned char; size_t = unsigned int]
22:18:05.695 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:05.695 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:06.577 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:06.577 -> [D] AudioCopy.h : 92 - size_t audio_tools::StreamCopyT::copy() [with T = unsigned char; size_t = unsigned int]
22:18:06.577 -> [D] AudioEncoded.h : 336 - virtual size_t audio_tools::EncodedAudioPrint::write(const uint8_t*, size_t): 1024
22:18:06.577 -> [D] CodecMP3Helix.h : 108 - virtual size_t audio_tools::MP3DecoderHelix::write(const void*, size_t): 1024
22:18:07.411 -> [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
22:18:07.411 -> FILE CLOSED
22:18:07.411 -> FILE SZE:0
22:18:07.411 -> FILE END

aqildad-create, 2023-03-20:
Sir your library is very supportive and i am learning allot from it .  . 
Now , I try to replace and add multiple headers but un able to succeed for mp3 recording , 

I try to replace EncodedAudioStream out(&audioFile, new WAVEncoder());
wth 
EncodedAudioStream out(&audioFile, new MP3EncoderHelix()); or EncodedAudioStream out(&audioFile, new MP3EncoderLame());  But no success . 

Do you have relevant example and documentation of recording i2s as MP3 ?

pschatzmann, 2023-03-20:
Just read the documentation: There is no MP3EncoderHelix

aqildad-create, 2023-03-20:
i incude #include "AudioCodecs/CodecMP3LAME.h"
and use EncodedAudioStream out(&audioFile, new MP3EncoderLAME()); , then program compile successfully but throw below error in logs

19:03:56.365 -> [D] I2SStream.h : 34 - audio_tools::I2SStream::I2SStream(int)
19:03:56.365 -> [D] CodecMP3LAME.h : 42 - audio_tools::MP3EncoderLAME::MP3EncoderLAME()
19:03:56.365 -> [D] AudioEncoded.h : 229 - audio_tools::EncodedAudioPrint::EncodedAudioPrint(Print*, audio_tools::AudioEncoder*)
19:03:56.365 -> [D] CodecMP3LAME.h : 57 - virtual void audio_tools::MP3EncoderLAME::setOutputStream(Print&)
19:03:56.365 -> [D] AudioCopy.h : 26 - audio_tools::StreamCopyT<T>::StreamCopyT(Print&, audio_tools::AudioStream&, int) [with T = unsigned char]
19:03:56.415 -> [I] AudioCopy.h : 79 - buffer_size=1073618896
19:03:56.415 -> [D] AudioCopy.h : 333 - audio_tools::StreamCopy::StreamCopy(Print&, audio_tools::AudioStream&, int)
19:03:56.415 -> starting I2S...
19:03:56.415 -> [D] I2SStream.h : 53 - bool audio_tools::I2SStream::begin(audio_tools::I2SConfig)
19:03:56.415 -> [D] I2SESP32.h : 44 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfig)
19:03:56.415 -> [D] I2SESP32.h : 140 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfig, int, int)
19:03:56.415 -> [I] I2SConfig.h : 104 - rx/tx mode: RX_MODE
19:03:56.415 -> [I] I2SConfig.h : 105 - port_no: 0
19:03:56.460 -> [I] I2SConfig.h : 106 - is_master: Master
19:03:56.460 -> [I] I2SConfig.h : 107 - sample rate: 16000
19:03:56.460 -> [I] I2SConfig.h : 108 - bits per sample: 32
19:03:56.460 -> [I] I2SConfig.h : 109 - number of channels: 1
19:03:56.460 -> [I] I2SConfig.h : 110 - i2s_format: I2S_STD_FORMAT
19:03:56.460 -> [I] I2SConfig.h : 112 - auto_clear:0
19:03:56.460 -> [I] I2SConfig.h : 119 - buffer_count:10
19:03:56.460 -> [I] I2SConfig.h : 120 - buffer_size:512
19:03:56.460 -> [I] I2SConfig.h : 124 - pin_bck: 14
19:03:56.460 -> [I] I2SConfig.h : 126 - pin_ws: 15
19:03:56.460 -> [I] I2SConfig.h : 128 - pin_data: 32
19:03:56.460 -> [D] I2SESP32.h : 172 - i2s_driver_install
19:03:56.460 -> [D] I2SESP32.h : 189 - i2s_set_pin
19:03:56.507 -> [D] I2SESP32.h : 200 - i2s_zero_dma_buffer
19:03:56.507 -> [D] I2SESP32.h : 204 - begin - started
19:03:56.507 -> I2S started
19:03:56.507 -> [D] AudioEncoded.h : 310 - bool audio_tools::EncodedAudioPrint::begin(audio_tools::AudioBaseInfo)
19:03:56.507 -> [D] CodecMP3LAME.h : 66 - virtual void audio_tools::MP3EncoderLAME::setAudioInfo(audio_tools::AudioBaseInfo)
19:03:56.507 -> [D] CodecMP3LAME.h : 57 - virtual void audio_tools::MP3EncoderLAME::setOutputStream(Print&)
19:03:56.507 -> [I] CodecMP3LAME.h : 166 - LibLAME channels: 1
19:03:56.507 -> [I] CodecMP3LAME.h : 167 - LibLAME sample_rate: 16000
19:03:56.507 -> [I] CodecMP3LAME.h : 168 - LibLAME bits_per_sample: 32
19:03:56.554 -> [I] CodecMP3LAME.h : 169 - LibLAME quality: 7
19:03:56.554 -> [Error] C:\Users\hp\Documents\Arduino\libraries\arduino-liblame-main\src\liblame\lame.c : 2772 - calloc(1,85840) -> 0x0 
19:03:56.554 -> available MALLOC_CAP_8BIT: 47092 / MALLOC_CAP_32BIT: 47092  / MALLOC_CAP_SPIRAM: 0

pschatzmann, 2023-03-20:
Please read the README of the liblame project!

aqildad-create, 2023-03-20:
Sr , as per my understanding  i change the config.h with below mage but still  am getting below log error , 

 

[image]

21:50:13.004 -> [D] I2SStream.h : 34 - audio_tools::I2SStream::I2SStream(int)
21:50:13.004 -> [D] CodecMP3LAME.h : 42 - audio_tools::MP3EncoderLAME::MP3EncoderLAME()
21:50:13.004 -> [D] AudioEncoded.h : 229 - audio_tools::EncodedAudioPrint::EncodedAudioPrint(Print*, audio_tools::AudioEncoder*)
21:50:13.004 -> [D] CodecMP3LAME.h : 57 - virtual void audio_tools::MP3EncoderLAME::setOutputStream(Print&)
21:50:13.004 -> [D] AudioCopy.h : 26 - audio_tools::StreamCopyT<T>::StreamCopyT(Print&, audio_tools::AudioStream&, int) [with T = unsigned char]
21:50:13.051 -> [I] AudioCopy.h : 79 - buffer_size=-1287651329
21:50:13.051 -> [D] AudioCopy.h : 333 - audio_tools::StreamCopy::StreamCopy(Print&, audio_tools::AudioStream&, int)
21:50:13.051 -> starting I2S...
21:50:13.051 -> [D] I2SStream.h : 53 - bool audio_tools::I2SStream::begin(audio_tools::I2SConfig)
21:50:13.051 -> [D] I2SESP32.h : 44 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfig)
21:50:13.051 -> [D] I2SESP32.h : 140 - bool audio_tools::I2SDriverESP32::begin(audio_tools::I2SConfig, int, int)
21:50:13.051 -> [I] I2SConfig.h : 104 - rx/tx mode: RX_MODE
21:50:13.051 -> [I] I2SConfig.h : 105 - port_no: 0
21:50:13.098 -> [I] I2SConfig.h : 106 - is_master: Master
21:50:13.098 -> [I] I2SConfig.h : 107 - sample rate: 44100
21:50:13.098 -> [I] I2SConfig.h : 108 - bits per sample: 32
21:50:13.098 -> [I] I2SConfig.h : 109 - number of channels: 1
21:50:13.098 -> [I] I2SConfig.h : 110 - i2s_format: I2S_STD_FORMAT
21:50:13.098 -> [I] I2SConfig.h : 112 - auto_clear:0
21:50:13.098 -> [I] I2SConfig.h : 119 - buffer_count:10
21:50:13.098 -> [I] I2SConfig.h : 120 - buffer_size:512
21:50:13.098 -> [I] I2SConfig.h : 124 - pin_bck: 14
21:50:13.098 -> [I] I2SConfig.h : 126 - pin_ws: 15
21:50:13.098 -> [I] I2SConfig.h : 128 - pin_data: 32
21:50:13.098 -> [D] I2SESP32.h : 172 - i2s_driver_install
21:50:13.098 -> [D] I2SESP32.h : 189 - i2s_set_pin
21:50:13.145 -> [D] I2SESP32.h : 200 - i2s_zero_dma_buffer
21:50:13.145 -> [D] I2SESP32.h : 204 - begin - started
21:50:13.145 -> I2S started
21:50:13.145 -> [D] AudioEncoded.h : 310 - bool audio_tools::EncodedAudioPrint::begin(audio_tools::AudioBaseInfo)
21:50:13.145 -> [D] CodecMP3LAME.h : 66 - virtual void audio_tools::MP3EncoderLAME::setAudioInfo(audio_tools::AudioBaseInfo)
21:50:13.145 -> [D] CodecMP3LAME.h : 57 - virtual void audio_tools::MP3EncoderLAME::setOutputStream(Print&)
21:50:13.145 -> [I] CodecMP3LAME.h : 166 - LibLAME channels: 1
21:50:13.145 -> [I] CodecMP3LAME.h : 167 - LibLAME sample_rate: 44100
21:50:13.145 -> [I] CodecMP3LAME.h : 168 - LibLAME bits_per_sample: 32
21:50:13.193 -> [I] CodecMP3LAME.h : 169 - LibLAME quality: 7
21:50:13.193 -> [Error] C:\Users\hp\Documents\Arduino\libraries\arduino-liblame-main\src\liblame\lame.c : 2772 - calloc(1,85840) -> 0x0 
21:50:13.193 -> available MALLOC_CAP_8BIT: 47092 / MALLOC_CAP_32BIT: 47092  / MALLOC_CAP_SPIRAM: 0

pschatzmann, 2023-03-20:
A regular ESP32 has not enough RAM you need one with PSRAM