pschatzmann/arduino-audio-tools #692: Trying to create the instance player.begin() in the loop, then the program crashes when player.copy() is called

urrudelu30, 2023-03-18:
Hi,

I am developing a simple application where I sent the wifi and pass over Bluetooth, then I want to connect to player. I put here my code in the loop():

     String wifiname, wifipassword;
      NVS.getString("wifi", wifiname);
      NVS.getString("pass", wifipassword);
      wifi = wifiname.c_str();
      password = wifipassword.c_str();
      Serial.printf("El ssid es %s\n",wifiname); 
      URLStream urlStream (wifi, password);// Usaremos el wifi para movernos por internet
      AudioSourceURL source(urlStream, urls, "audio/mp3"); // Cogemos archivos mp3 de internet
      AudioPlayer  player (source, i2s, decoder);
      player.begin();
      delay(500);
      player.copy();

When player.copy() is called these error appears:

[W] HttpRequest.h : 153 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 91 - size: 0
[I] AudioCopy.h : 69 - buffer_size=1024
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
CORRUPT HEAP: Bad head at 0x3ffdd8f8. Expected 0xabba1234 got 0x3ffec76c

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace: 0x400838b1:0x3ffec1a0 0x40093501:0x3ffec1c0 0x40099439:0x3ffec1e0 0x400990ab:0x3ffec310 0x40083d75:0x3ffec330 0x40099469:0x3ffec350 0x401ef7e1:0x3ffec370 0x400d7d41:0x3ffec390 0x400dd867:0x3ffec3b0 0x400ddda5:0x3ffec3d0 0x400dddbd:0x3ffec3f0 0x400def42:0x3ffec410 0x400dfdba:0x3ffec430 0x400dfff6:0x3ffecad0

I decoded with Arduino but still dont have the solution:

[image]

pschatzmann, 2023-03-18:
But this does not make any sense: The loop is executed a couple of 1000 times a second. Why would you try to create a new url connection and a new player in the loop just to destroy it again a couple of milliseconds later!

urrudelu30, 2023-03-18:
I can put only the first condition in the loop when the credentials of the wifi arrive at the esp32, but if I do that as I defined the player in this scope I obtain the error in the rest of the code as: "error: 'player' was not declared in this scope". Any Idea of how I can work with a player with an SSID and password that are not preloaded in the code and arrive in the loop when the phone is connected?

pschatzmann, 2023-03-18:
Just put the player and URL connection into global memory and do not provide any userid/password.
Then you can still manage your login and further processing in the loop..

closed 2023-03-18 by pschatzmann