pschatzmann/arduino-audio-tools #695: Choppy audio from stream http://streamingv2.shoutcast.com/rtl-1025

FatherMarco1971, 2023-03-19:
ANybody can help me to understand why the stream:

http://streamingv2.shoutcast.com/rtl-1025

is full of interruction?
maybe the bitrate is too high? seems that I2S_BUFFER_SIZE have some influence but too using higher value possible it doesn't work.
I'm tryin the player-url_icy-i2s example.

pschatzmann, 2023-03-19:
I guess you will need to increase the mp3 buffer sizes!

See 
- https://pschatzmann.github.io/arduino-audio-tools/classaudio__tools_1_1_m_p3_decoder_helix.html
- https://github.com/pschatzmann/arduino-libhelix

FatherMarco1971, 2023-03-19:
First of all Thanks!
You are right.
THings goes better but I've to increase sizes a lot:
```
#define MP3_MAX_OUTPUT_SIZE 2048*5//1024 * 5
#define MP3_MAX_FRAME_SIZE 65536 //1600
```
Not perfect yet there are some noise but things goes better better

FatherMarco1971, 2023-03-20:
Playing more with setting i solved the issues i don't know if can be of interest for somebody:

#define MP3_MAX_OUTPUT_SIZE 2048*5//1024 * 5
#define MP3_MAX_FRAME_SIZE 65536 //1600

#define DEFAULT_BUFFER_SIZE 4096 //1024
#define I2S_BUFFER_SIZE 1024 //512

Thanks again for your big work.

closed 2023-03-21 by pschatzmann