pschatzmann/arduino-audio-tools #706: Sometimes after my machineState comes to Player state this error message appear

urrudelu30, 2023-03-23:
Appears that player.copy() is working fine due to StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops appears but the other traces([E] Fade.h : 271 - setAudioInfo not called) make me confuse.

Initialize Begin
[I] AudioPlayer.h : 327 - setVolume(1.000000)
[I] VolumeStream.h : 183 - setVolume: 1.000000
[I] VolumeStream.h : 183 - setVolume: 1.000000
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSource.h : 175 - selectStream: 0/4 -> http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] URLStream.h : 76 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] Url.h : 74 - Url::parse
[I] Url.h : 110 - url->http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] Url.h : 111 - host->streaming.swisstxt.ch
[I] Url.h : 112 - protocol->http
[I] Url.h : 113 - path->/m/drsvirus/mp3_128
[I] Url.h : 114 - port->80
[I] URLStream.h : 254 - WiFiClient
[I] HttpRequest.h : 193 - process connecting to host streaming.swisstxt.ch port 80
[I] HttpRequest.h : 181 - connected 1 timeout 60000
[I] HttpRequest.h : 204 - Free heap: 93388
[I] HttpHeader.h : 358 - HttpRequestHeader::setValues - path: /m/drsvirus/mp3_128
[I] HttpHeader.h : 255 - HttpHeader::write
[I] HttpHeader.h : 367 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 382 - -> GET /m/drsvirus/mp3_128 HTTP/1.1
[I] HttpHeader.h : 192 -  -> Host: streaming.swisstxt.ch 
[I] HttpHeader.h : 192 -  -> Connection: close
[I] HttpHeader.h : 192 -  -> Accept: audio/mp3
[I] HttpHeader.h : 298 -  -> <CR LF>
[I] HttpRequest.h : 227 - Request written ... waiting for reply
[I] HttpHeader.h : 225 - HttpHeader::read
[W] HttpHeader.h : 232 - Waiting for data...
[I] HttpHeader.h : 162 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Content-Type: audio/mpeg
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Date: Thu, 23 Mar 2023 12:33:17 GMT
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-br:128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ice-audio-info: bitrate=128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-br:128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-description:SRF-Virus_MP3@128Kbps
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-genre:Entertainment
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-mimetype:audio/mpeg
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-name:/SRF-Virus_Room2
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-pub:0
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-url:www.srgssr.ch
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Server: Icecast 2.4.0-kh10
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Cache-Control: no-cache, no-store
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Methods: GET, OPTIONS, HEAD
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Connection: Close
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Expires: Mon, 26 Jul 1997 05:00:00 GMT
[I] HttpHeader.h : 162 - HttpHeader::readLine ->
[W] HttpRequest.h : 153 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 91 - size: 0
[I] AudioCopy.h : 69 - buffer_size=1024
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[E] Fade.h : 271 - setAudioInfo not called
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

pschatzmann, 2023-03-23:
This is happening when the AudioPlayer could not determine how many channels there are. 
How did you manage to get this ?

urrudelu30, 2023-03-23:
I am working with AudioPlayer in the State 1 of my machineState as:

URLStream urlStream (wifi, password);// Usaremos el wifi para movernos por internet
AudioSourceURL source(urlStream, urls, "audio/mp3"); // Cogemos archivos mp3 de internet
AudioPlayer  player (source, i2s, decoder);

Second State is working with BluethootSerial and Wifi, powering off the Ble when need to works with Wifi.

Third State is working with A2DP Bluethoot from your library BluetoothA2DPSink.h as:

i2s_pin_config_t my_pin_config = {
           .bck_io_num = 26,
           .ws_io_num = 25,
           .data_out_num = 22,
           .data_in_num = I2S_PIN_NO_CHANGE
          };
          a2dp_sink.set_pin_config(my_pin_config);
          a2dp_sink.set_avrc_metadata_attribute_mask(ESP_AVRC_MD_ATTR_TITLE | ESP_AVRC_MD_ATTR_ARTIST);
          a2dp_sink.set_avrc_metadata_callback(avrc_metadata_callback);
          a2dp_sink.start("A2DPAcousticMobileDock");

I think I have some problems whit the i2s, but not know how to resolve it, any idea?

urrudelu30, 2023-03-23:
I have applied your last changes without setting this new line and some critical error is happening.

the erros is this: E (709) I2S: i2s_driver_uninstall(2047): I2S port 0 has not installed
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

[image]

pschatzmann, 2023-03-23:
Yes, this did not work. When testing, I was running into the same problem. I needed to move the initial setup of the Fade to the begin method. However this assumes that the output has some valid values defined....

urrudelu30, 2023-03-23:
I have seen your last commit, but unfortunately I have a similar issue as before:

[I] I2SStream.h : 74 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 63 - sample_rate: 0
[I] AudioTypes.h : 64 - channels: 0
[I] AudioTypes.h : 65 - bits_per_sample: 16
[I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 107 - port_no: 0
[I] I2SConfig.h : 108 - is_master: Master
[I] I2SConfig.h : 109 - sample rate: 0
[I] I2SConfig.h : 110 - bits per sample: 16
[I] I2SConfig.h : 111 - number of channels: 0
[I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 114 - auto_clear: true
[I] I2SConfig.h : 116 - use_apll: true
[I] I2SConfig.h : 121 - buffer_count:6
[I] I2SConfig.h : 122 - buffer_size:512
[I] I2SConfig.h : 126 - pin_bck: 14
[I] I2SConfig.h : 128 - pin_ws: 15
[I] I2SConfig.h : 130 - pin_data: 22
[I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 107 - port_no: 0
[I] I2SConfig.h : 108 - is_master: Master
[I] I2SConfig.h : 109 - sample rate: 0
[I] I2SConfig.h : 110 - bits per sample: 16
[I] I2SConfig.h : 111 - number of channels: 0
[I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 114 - auto_clear: true
[I] I2SConfig.h : 116 - use_apll: true
[I] I2SConfig.h : 121 - buffer_count:6
[I] I2SConfig.h : 122 - buffer_size:512
[I] I2SConfig.h : 126 - pin_bck: 14
[I] I2SConfig.h : 128 - pin_ws: 15
[I] I2SConfig.h : 130 - pin_data: 22
Guru Meditation Error: Core  0 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  0 register dump:
PC      : 0x400ee750  PS      : 0x00060030  A0      : 0x800ef450  A1      : 0x3ffd54e0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffbc19c  A5      : 0x00000000
A6      : 0x3ffbc19c  A7      : 0x3ffc670c  A8      : 0x00000100  A9      : 0x00000002  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffc670c
A14     : 0xffffffbf  A15     : 0x00000008  SAR     : 0x00000019  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4009129c  LEND    : 0x400912a7  LCOUNT  : 0x00000000

pschatzmann, 2023-03-23:
So you need to figure out why you have:
[I] AudioTypes.h : 63 - sample_rate: 0
[I] AudioTypes.h : 64 - channels: 0

pschatzmann, 2023-03-23:
The bug must be somewhere in your code...

urrudelu30, 2023-03-23:
I have put an ESP_Restar when I pass from the 3er State to the first state which is AudioPlayer: Here you can see the initialization in the setup is correct (2 number of channels), once the player.begin is called the problem appears, there is channels 0 when is called

Wifi
Password
Mode 0
[I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 107 - port_no: 1
[I] I2SConfig.h : 108 - is_master: Master
[I] I2SConfig.h : 109 - sample rate: 22050
[I] I2SConfig.h : 110 - bits per sample: 16
[I] I2SConfig.h : 111 - number of channels: 2
[I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 114 - auto_clear: true
[I] I2SConfig.h : 116 - use_apll: true
[I] I2SConfig.h : 121 - buffer_count:6
[I] I2SConfig.h : 122 - buffer_size:512
[I] I2SConfig.h : 126 - pin_bck: 14
[I] I2SConfig.h : 128 - pin_ws: 15
[I] I2SConfig.h : 130 - pin_data: 22
WiFi no conectadoEl ssid es 5484
.................................................................................................................192.168.1.131
Inicializo Begin[I] AudioPlayer.h : 352 - setVolume(1.000000)
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] AudioTypes.h : 63 - sample_rate: 0
[I] AudioTypes.h : 64 - channels: 0
[I] AudioTypes.h : 65 - bits_per_sample: 16
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSource.h : 175 - selectStream: 0/4 -> http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] URLStream.h : 89 - virtual bool audio_tools::URLStream::begin(const char*, const char*, MethodID, const char*, const char*): http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] Url.h : 74 - Url::parse
[I] Url.h : 110 - url->http://streaming.swisstxt.ch/m/drsvirus/mp3_128
[I] Url.h : 111 - host->streaming.swisstxt.ch
[I] Url.h : 112 - protocol->http
[I] Url.h : 113 - path->/m/drsvirus/mp3_128
[I] Url.h : 114 - port->80
[I] URLStream.h : 314 - void audio_tools::URLStream::login()
[I] URLStream.h : 290 - WiFiClient
[I] HttpRequest.h : 215 - process connecting to host streaming.swisstxt.ch port 80
[I] HttpRequest.h : 202 - connected 1 timeout 60000
[I] HttpRequest.h : 226 - Free heap: 101232
[I] HttpHeader.h : 360 - HttpRequestHeader::setValues - path: /m/drsvirus/mp3_128
[I] HttpHeader.h : 126 - Content-Length 0
[I] HttpHeader.h : 257 - HttpHeader::write
[I] HttpHeader.h : 369 - HttpRequestHeader::write1stLine
[I] HttpHeader.h : 384 - -> GET /m/drsvirus/mp3_128 HTTP/1.1
[I] HttpHeader.h : 192 -  -> Content-Length: 0
[I] HttpHeader.h : 192 -  -> Host: streaming.swisstxt.ch
[I] HttpHeader.h : 192 -  -> Connection: close 
[I] HttpHeader.h : 192 -  -> Accept: audio/mp3
[I] HttpHeader.h : 300 -  -> <CR LF>
[I] HttpRequest.h : 263 - Request written ... waiting for reply
[I] HttpHeader.h : 225 - HttpHeader::read
[W] HttpHeader.h : 232 - Waiting for data...
[I] HttpHeader.h : 162 - HttpHeader::readLine -> HTTP/1.1 200 OK
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Content-Type: audio/mpeg
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Date: Thu, 23 Mar 2023 14:58:30 GMT
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-br:128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> ice-audio-info: bitrate=128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-br:128
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-description:SRF-Virus_MP3@128Kbps
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-genre:Entertainment
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-mimetype:audio/mpeg
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-name:/SRF-Virus_Room1
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-pub:0
[I] HttpHeader.h : 162 - HttpHeader::readLine -> icy-url:www.srgssr.ch
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Server: Icecast 2.4.0-kh10
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Cache-Control: no-cache, no-store
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Origin: *
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Access-Control-Allow-Methods: GET, OPTIONS, HEAD
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Connection: Close
[I] HttpHeader.h : 162 - HttpHeader::readLine -> Expires: Mon, 26 Jul 1997 05:00:00 GMT
[I] HttpHeader.h : 162 - HttpHeader::readLine -> 
[W] HttpRequest.h : 167 - no CONTENT_LENGTH found in reply
[I] URLStream.h : 110 - size: 0
[I] URLStream.h : 331 - virtual bool audio_tools::URLStream::waitForData()
[I] AudioCopy.h : 71 - buffer_size=1024
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 136 - channels=0
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 136 - channels=0
[E] Fade.h : 136 - channels=0
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 136 - channels=0
[E] Fade.h : 136 - channels=0
[E] Fade.h : 136 - channels=0
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[E] Fade.h : 136 - channels=0

urrudelu30, 2023-03-23:
I think I am able to solve it: putting the following line in the 1er state.

 auto config = i2s.defaultConfig(TX_MODE);
 player.setAudioInfo(config);

When the code is power up it will reset these variables as in the setup was configured (not problem I hope), when the machine state is working and pass from the third to first state these workarounds the problem:

] URLStream.h : 331 - virtual bool audio_tools::URLStream::waitForData()
[I] AudioCopy.h : 71 - buffer_size=1024
[I] AudioPlayer.h : 243 - sample_rate: 44100
[I] AudioPlayer.h : 244 - bits_per_sample: 16
[I] AudioPlayer.h : 245 - channels: 2
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] VolumeStream.h : 191 - setVolume: 1.000000
[I] AudioTypes.h : 63 - sample_rate: 44100
[I] AudioTypes.h : 64 - channels: 2
[I] AudioTypes.h : 65 - bits_per_sample: 16
[I] I2SStream.h : 74 - virtual void audio_tools::I2SStream::setAudioInfo(audio_tools::AudioBaseInfo)
[I] AudioTypes.h : 63 - sample_rate: 44100
[I] AudioTypes.h : 64 - channels: 2
[I] AudioTypes.h : 65 - bits_per_sample: 16
[I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 107 - port_no: 1
[I] I2SConfig.h : 108 - is_master: Master
[I] I2SConfig.h : 109 - sample rate: 44100
[I] I2SConfig.h : 110 - bits per sample: 16
[I] I2SConfig.h : 111 - number of channels: 2
[I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 114 - auto_clear: true
[I] I2SConfig.h : 116 - use_apll: true
[I] I2SConfig.h : 121 - buffer_count:6
[I] I2SConfig.h : 122 - buffer_size:512
[I] I2SConfig.h : 126 - pin_bck: 14
[I] I2SConfig.h : 128 - pin_ws: 15
[I] I2SConfig.h : 130 - pin_data: 22
[I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 107 - port_no: 1
[I] I2SConfig.h : 108 - is_master: Master
[I] I2SConfig.h : 109 - sample rate: 44100
[I] I2SConfig.h : 110 - bits per sample: 16
[I] I2SConfig.h : 111 - number of channels: 2
[I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 114 - auto_clear: true
[I] I2SConfig.h : 116 - use_apll: true
[I] I2SConfig.h : 121 - buffer_count:6
[I] I2SConfig.h : 122 - buffer_size:512
[I] I2SConfig.h : 126 - pin_bck: 14
[I] I2SConfig.h : 128 - pin_ws: 15
[I] I2SConfig.h : 130 - pin_data: 22
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

closed 2023-03-24 by pschatzmann