pschatzmann/arduino-audio-tools #708: Crackling with generating single audio-impulse

proleaveprod, 2023-03-24:
Hi! First of all, **thank you for your library a lot**!

I have a task where I need to play sound impulse from array with 2048 numbers.
I have used your init function for **ES8388**, after that i just send my array by function **i2s_write**. 
I hear a few clips after reproduction and i don’t know what is it.

[image]

I tried to use **i2s_start()** before and **i2s_stop** after i2s_write function, but it doesn't work well. sometimes i see clips on the graph.

[image]

Please, explain me, **how to properly generate a single impulse**?
I think there is good way for solve my problem, but i couldn't find it.

P.s.
Currently, i have 512 16-bit values ( 512 * 2 and *2     (because of 2 channels) = 2048)
[image]

pschatzmann, 2023-03-24:
I see plenty of problems with your approach:
- I suggest not to mix different API  and if you decide to use the AudioKit it would be kit.write() instead of i2s_write
- However I would recommend to use this (Arduino Audio Tools) library instead the AudioKit api: So use the AudioKitStream class instead to setup and output audio.
- I would represent the data as array of int16_t with one channel. In the AudioKitStream configuration you can indicate that channels=1;  
- Instead of waiting, starting and stopping you can just write silence (with writeSilence()), this avoids noise from the codec not getting any data in time

pschatzmann, 2023-03-24:
I thinks this should have rather been a discussion. I don't see any problem with my functionality, so I am closing this issue.