pschatzmann/arduino-audio-tools #711: No output from player_url_icy_audiokit sample

FatherMarco1971, 2023-03-24:
Last commit broken something. I had to revert to releses:
arduino-audio-tools-0.9.3
arduino-audiokit-0.6.2
to make it work again.
The sampl output work as expected.
In the player_url_icy_audiokit seems everithing ok (he url is open the stream decode start...) but no sound come out from headphone....
I'm not expert enough to suggest a solution but if you need information to debug happy to help.

pschatzmann, 2023-03-25:
I was working on an option to build the audiotools with the audiokit in IDF. 
I finally committed all outstanding changes of the audiokit and audiotools project to make sure that both projects are up to date.

It is working perfectly on my side, so I suggest that you double check with the latest updates.

FatherMarco1971, 2023-03-25:
Downloaded all again with github.
Now there is a compiler error:
c:\Users\marco\Documents\Arduino\libraries\arduino-audiokit\src/audio_hal/audio_gpio.h:16:28: fatal error: hal/gpio_types.h: No such file or directory
There is a dependency i don't understand maybe?

pschatzmann, 2023-03-25:
Please go to -> Tools -> Boards Manager and check the version of your installed ESP32.
The latest version is 2.0.7

FatherMarco1971, 2023-03-25:
Sorry. You are right need to update board! Work like a charm!

Thank you and sorry...i've mutch to learn.

pschatzmann, 2023-03-25:
Nevertheless, this was helpful, 
I committed a correction, so it will work with the old version again...

closed 2023-03-25 by pschatzmann