pschatzmann/arduino-audio-tools #712: Communcaton ESPNPW send-comuncaton espnow rsv codec Sbc

aqildad-create, 2023-03-25:
I RUN ABOVE EXAMPLE AND ADD MAC ADDRESSES , BUT NO COMMUNCATON SUCCESS . 

/**
 * @file example-serial-send.ino
 * @author Phil Schatzmann
 * @brief Sending encoded audio over ESPNow
 * @version 0.1
 * @date 2022-03-09
 *
 * @copyright Copyright (c) 2022
 */

#include "AudioTools.h"
#include "AudioLibs/Communication.h"
#include "AudioCodecs/CodecSBC.h"

uint16_t sample_rate = 32000;
uint8_t channels = 2;  // The stream will have 2 channels
SineWaveGenerator<int16_t> sineWave( 32000);  // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<int16_t> sound( sineWave); // Stream generated from sine wave
ESPNowStream now;
EncodedAudioStream encoder(&now, new SBCEncoder()); // encode and write to ESP-now
StreamCopy copier(encoder, sound);  // copies sound into i2s
const char *peers[] = {"0C:B8:15:C3:40:68"};

Sender code
void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  auto cfg = now.defaultConfig();
  cfg.mac_address = "94:E6:86:04:89:44";
  now.begin(cfg);
  now.addPeers(peers);

  // Setup sine wave
  auto cfgs = sineWave.defaultConfig();
  cfgs.sample_rate = sample_rate;
  cfgs.channels = channels;
  cfgs.bits_per_sample = 16;
  sineWave.begin(cfgs, N_B4);

  // start encoder
  encoder.begin(cfgs);
  
  Serial.println("Sender started...");
}

void loop() { 
  copier.copy();
}

//--------receiver code
/**
 * @file example-serial-receive.ino
 * @author Phil Schatzmann
 * @brief Receiving audio via ESPNow and decoding data to I2S 
 * @version 0.1
 * @date 2022-03-09
 * 
 * @copyright Copyright (c) 2022
 */

#include "AudioTools.h"
#include "AudioLibs/Communication.h"
#include "AudioCodecs/CodecSBC.h"
// #include "AudioLibs/AudioKit.h"

ESPNowStream now;
I2SStream out;  // or AudioKitStream
EncodedAudioStream decoder(&out, new SBCDecoder(256)); // decode and write to I2S - ESP Now is limited to 256 bytes
StreamCopy copier(decoder, now);     
const char *peers[] = {"94:E6:86:04:89:44"};

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Warning);

  // setup esp-now
  auto cfg = now.defaultConfig();
  cfg.mac_address = "0C:B8:15:C3:40:68";
  now.begin(cfg);
  now.addPeers(peers);

  // start I2S
  Serial.println("starting I2S...");
  auto config = out.defaultConfig(TX_MODE);
  config.pin_ws = 2;
         config.pin_bck = 15;
         config.pin_data = 3;
  
  out.begin(config);

  // start decoder
  decoder.begin();

  Serial.println("Receiver started...");
}

void loop() { 
  copier.copy();
}

//--------sender log
17:38:20.493 -> 
17:38:20.493 -> mac: 94:E6:86:04:89:44
17:38:20.493 -> Sender started...

//receiver log 

17:38:04.336 -> mac: 0C:B8:15:C3:40:68
17:38:04.336 -> starting I2S...
17:38:04.336 -> Receiver started...

aqildad-create, 2023-03-25:
Sr , your library is working like  a charm , it was my mistake in receiver of i2s pin assignment , now  I can clearly listen the sine tone . 

But sir now  i change sine generator  to microphone input (INMP441) , at receiver their is huge noise .

below is my sender code 
 
#include "AudioTools.h"
#include "AudioLibs/Communication.h"
#include "AudioCodecs/CodecSBC.h"

I2SStream i2sStream;    // Access I2S as stream
//ConverterFillLeftAndRight<int32_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

uint16_t sample_rate = 32000;
uint8_t channels = 2;  // The stream will have 2 channels
 
ESPNowStream now;
EncodedAudioStream encoder(&now, new SBCEncoder()); // encode and write to ESP-now
StreamCopy copier(encoder, i2sStream);  // copies sound into i2s
const char *peers[] = {"0C:B8:15:C3:40:68"};

void setup() {
  Serial.begin(115200);
 // AudioLogger::instance().begin(Serial, AudioLogger::Warning);
 AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial, AudioLogger::Info);  
  auto cfg = now.defaultConfig();
  cfg.mac_address = "94:E6:86:04:89:44";
  now.begin(cfg);
  now.addPeers(peers);

  
   // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.sample_rate = sample_rate;//22050;
  config.channels = 2;
  config.bits_per_sample =32;
  i2sStream.begin(config);
  Serial.println("I2S started");

  
 

  // start encoder
  encoder.begin(config);
  
  Serial.println("Sender started...");
}

void loop() { 
  copier.copy();
}

aqildad-create, 2023-03-26:
is above possible with i2s microphone (INMP441) ? 

I2s-> SBC encoder -> espnow -> SBC decoder -> i2S DAC

pschatzmann, 2023-03-26:
I don't see any reason why this should not work...

aqildad-create, 2023-03-26:
apology if I am missing document. 
from were i will get SBC supported  audio values ? I google for it but cant find related document , 

And which i2s microphone you use for your testing , i have INMP441 , does this make any difference ?

pschatzmann, 2023-03-26:
The INMP441 creates noise on one channel if you use it with 16 bits. So you would need to filter that out or use 32 bits and convert them to 16 bits

aqildad-create, 2023-03-26:
yes , you are correct , when  was I testing i2s webserver i  get the same nose when  was using 16bts , but when  I change it to 32 bit then i get clear audio on webserver , Now  i am using same 32 bit but still its very distorted in receiver after ESPnow , 

THS IS SENDER CODE and receiver code is same as communication SBC decoder . 

#include "AudioTools.h"
#include "AudioLibs/Communication.h"
#include "AudioCodecs/CodecSBC.h"

I2SStream i2sStream; // Access I2S as stream
 
uint16_t sample_rate = 32000;
uint8_t channels = 2; // The stream will have 2 channels

ESPNowStream now;
EncodedAudioStream encoder(&now, new SBCEncoder()); // encode and write to ESP-now
StreamCopy copier(encoder, i2sStream); // copies sound into i2s
const char *peers[] = {"0C:B8:15:C3:40:68"};

void setup() {
Serial.begin(115200);
// AudioLogger::instance().begin(Serial, AudioLogger::Warning);
AudioLogger::instance().begin(Serial, AudioLogger::Warning);
//AudioLogger::instance().begin(Serial, AudioLogger::Info);
auto cfg = now.defaultConfig();
cfg.mac_address = "94:E6:86:04:89:44";
now.begin(cfg);
now.addPeers(peers);

// start i2s input with default configuration
Serial.println("starting I2S...");
auto config = i2sStream.defaultConfig(RX_MODE);
config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
config.sample_rate = sample_rate;//22050;
config.channels = 2;
config.bits_per_sample =32;
i2sStream.begin(config);
Serial.println("I2S started");

// start encoder
encoder.begin(config);

Serial.println("Sender started...");
}

void loop() {
copier.copy();
}

pschatzmann, 2023-03-26:
You can't use 32bit values with the codec!
You can use a FormatConverterStream Class to convert 32 to 16 bits

aqildad-create, 2023-03-26:
FormatConverterStream  . .  give me error 
[image]

pschatzmann, 2023-03-26:
Just read the error message. This class must not have any template!

aqildad-create, 2023-03-26:
Sir , please 
Is their any example were you used FormatConverterStream ?

aqildad-create, 2023-03-26:
try to use ChannelFormatConverterStreamT . It compile but i am getting same nose in receiver 

ChannelFormatConverterStreamT<int16_t> conv(i2sStream);
StreamCopy copier(encoder, conv); // copies sound into i2s