pschatzmann/arduino-audio-tools #715: Support for HLS streams

technoblogy, 2023-03-26:
Is there support for HLS streams in AudioTools? The BBC radio stations are only available as 320kb/s HLS streams and I’d really like to be able to play them. For example BBC Radio 3 is on:

https://a.files.bbci.co.uk/media/live/manifesto/audio/simulcast/hls/uk/sbr_high/ak/bbc_radio_three.m3u8

It seems that the esp-adf framework claims to support them:

https://github.com/espressif/esp-adf

pschatzmann, 2023-03-26:
No, but feel free to extend the framework.
Here is a MP3 link for BBC 3 https://stream.live.vc.bbcmedia.co.uk/bbc_radio_three
And finally a link for all [GB radios](https://de1.api.radio-browser.info/xml/stations/bycountrycodeexact/gb)

closed 2023-03-26 by pschatzmann

technoblogy, 2023-03-26:
Thank you! I read somewhere that the BBC no longer provide MP3 streams, so that link will solve the problem.