pschatzmann/arduino-audio-tools #716: Getting the examples to work on a LyraT 4.3 board

technoblogy, 2023-03-26:
I have both an AI Thinker AudioKit board and an ESP32 LyraT 4.3 board.

I’ve successfully got the AI Thinker AudioKit board playing Internet radio streams through the on-board amplifiers to speakers, using your example file:

    player-url_icy-audiokit

I’ve now tried the LyraT board. I wasn’t sure what example file to use, so tried:

    player-url_icy-i2s

It connects to the stream correctly and I get as far as a series of lines:

    [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
    [I] AudioCopy.h : 136 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops

but no sound. Is there something else I need to do to enable the amplifier, and the touch buttons?

Sorry if this is already in the documentation somewhere, and thank you for an amazing resource!

pschatzmann, 2023-03-26:
Check the Readme of the AudioKit project! 
You need to select the proper board

technoblogy, 2023-03-26:
OK, thank you - I'll try that.