pschatzmann/arduino-audio-tools #749: Example "streams-i2s-webserver_wav.ino" not working

spawn451, 2023-04-04:
I have been trying for several days to make the stream-i2s-webserver_wav_ino work but without success.
I went through all the issues raised here but none helped me. My configuration is as follows: EPS32-WROOM-32 + INMP411 Microphone. I tested the streams-generator-webserver_wav which works correctly but the serial plotter does not seems to work either. When I open the browser I only hear a crackling noise.
If anyone could help me that would be really nice.

Here is my sketch:

#include "AudioTools.h"

//AudioEncodedServer server(new WAVEncoder(),"ssid","password");  
AudioWAVServer server("ssid","password"); // the same a above

I2SStream i2sStream;    // Access I2S as stream
ConverterFillLeftAndRight<int16_t> filler(LeftIsEmpty); // fill both channels - or change to RightIsEmpty

void setup(){
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);
  //AudioLogger::instance().begin(Serial, AudioLogger::Warning);
  //AudioLogger::instance().begin(Serial, AudioLogger::Debug);

  // start i2s input with default configuration
  Serial.println("starting I2S...");
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.pin_ws = 15;
  config.pin_bck = 14;
  config.pin_data = 32;
  config.sample_rate = 22050;
  config.channels = 2;
  config.bits_per_sample = 16;
  i2sStream.begin(config);
  Serial.println("I2S started");

  // start data sink
  server.begin(i2sStream, config, &filler);
}

// Arduino loop  
void loop() {
  // Handle new connections
  server.copy();  
}

pschatzmann, 2023-04-04:
The idea was that you use the csv output w/o any converter, so that you see on which channel you get the noise. The you would apply the ConverterFillLeftAndRight with the parameter that applies for your case.

Instead you can also skip this (comment out the ConverterFillLeftAndRight and remove it from the server.begin()) and just change the bits_per_sample to 32

closed 2023-04-04 by pschatzmann

spawn451, 2023-04-05:
It was a soldering issue ^^ It works now :) Thanks

spawn451, 2023-04-14:
I don't think I'm in the right place to ask this question  but I would like to know if it was possible to reduce latency/delay of audio transmission when using the stream on the webserver. I performed tests playing with the parameters (DEFAULT_BUFFER_SIZE,I2S_BUFFER_SIZE,I2S_BUFFER_COUNT) but without much improvement.
Can using multicore processing help or is this a hardware limitation?

pschatzmann, 2023-04-14:
The buffering is done by the receiver e.g. the browser in your case.

spawn451, 2023-04-15:
In combination with go2rtc(webrtc) I get a great result with almost 0 latency :-) Thanks again for your guidance.

jrbecart, 2025-02-09:
@spawn451 did you use rtsp or webrtc? Do you have an example?