pschatzmann/arduino-audio-tools #756: INMP441 Microphone & RTSP

spawn451, 2023-04-06:
I'm trying to use RTSP to stream  from INMP441 Microphone but it is not working.
No sound at all.
I think the code below is totally wrong but I really don't know how to do it.
I am a novice in this field :(

#include "AudioTools.h"
//#include "AudioLibs/AudioKit.h"
#include "AudioLibs/RTSPStream.h"
#include "AudioStreamer.h"
#include "RTSPServer.h"

int port = 554;
//AudioKitStream kit;  // Audio source
//RTSPSourceFromAudioStream source(kit); // IAudioSource for RTSP
I2SStream i2sStream;
RTSPSourceFromAudioStream source(i2sStream);
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP
RTSPServer rtsp = RTSPServer(&streamer, port);

const char* wifi = "wifi";
const char* password = "password";

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // start i2s input with default configuration
  auto config = i2sStream.defaultConfig(RX_MODE);
  config.i2s_format = I2S_STD_FORMAT; // if quality is bad change to I2S_LSB_FORMAT https://github.com/pschatzmann/arduino-audio-tools/issues/23
  config.pin_ws = 15;
  config.pin_bck = 14;
  config.pin_data = 32;
  config.sample_rate = 22050;
  config.channels = 1;
  config.bits_per_sample = 32;
  i2sStream.begin(config);

  // Start Wifi
  rtsp.begin(wifi, password);

}

void loop() { delay(1000); }

pschatzmann, 2023-04-08:
RTP does not support 32 bits, so you will need to convert it to 16 bits. I also did not manage to have a sample rate 22050 working, when I tried last only 8000 was giving a stable output.

closed 2023-04-08 by pschatzmann

spaceX-creater, 2023-04-10:
i change bits to 16 and change sample rate to 8000 than my esp32 continuous restarting.

spawn451, 2023-04-11:
Unfortunately  still  no sound when changing sample_rate  to 8000 and bits_per_sample to 16.
No errors but no sound while playing the stream through VLC.

spawn451, 2023-04-11:
Unfortunately I changed the sample_rate to 8000 and bits_per_sample to 16 but still no sound.

pschatzmann, 2023-04-11:
But the INMP441 does not produce any meaningful sound with 16 bits!
I am not sure if this is your only problem but you will certainly need to convert the data...

spawn451, 2023-04-11:
Ok, that probably explains my problem. Could you give me a track to perform the data conversion, I'm a bit lost :(

pschatzmann, 2023-04-11:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Converting-the-Data-Format

spawn451, 2023-04-11:
I tried to adapt my sketch with the data conversion but cannot make it work :(
I gete [E] AudioStreamsConverter.h : 592 - begin failed
I think I forgot the "from" part but I'm stuck

int port = 554;

I2SStream i2sStream; // Audio source
RTSPSourceFromAudioStream source(i2sStream); // IAudioSource for RTSP
AudioStreamer streamer = AudioStreamer(&source); // Stream audio via RTSP
RTSPServer rtsp = RTSPServer(&streamer, port);

AudioInfo from(22050,1,32);
AudioInfo to(8000,1,16);
FormatConverterStream converter(i2sStream); 
StreamCopy copier(i2sStream, converter);

const char* wifi = "wifi";
const char* password = "password";

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // Define Converter
  converter.begin(from, to);

  auto cfg = i2sStream.defaultConfig(RX_MODE);

  cfg.i2s_format = I2S_STD_FORMAT;
  cfg.copyFrom(to);
  cfg.pin_ws = 2;
  cfg.pin_bck = 14;
  cfg.pin_data = 15;
  cfg.port_no = 1;
  i2sStream.begin(cfg);
  
  // Start Wifi
  rtsp.begin(wifi, password);

}

void loop() { delay(1000); }

spawn451, 2023-04-12:
Thank  a lot for your help, I succeeded :) It works