pschatzmann/arduino-audio-tools #759: Compilation ERRORS

elshnkhll, 2023-04-08:
libraries/audio-tools/src/AudioCodecs/CodecBase64.h:216:16: error: 'class Print' has no member named 'flush'
       p_print->flush();
                ^

libraries/audio-tools/src/AudioCodecs/CodecBase64.h:294:14: error: 'class Print' has no member named 'flush'
     p_print->flush();
              ^

libraries/audio-tools/src/AudioTools/ResampleStream.h:367:24: error: conversion from 'double' to 'audio_tools::int24_t' is ambiguous
     return round(result);
                        ^

libraries/audio-tools/src/AudioBasic/Int24.h:42:3: note: candidate: audio_tools::int24_t::int24_t(const float&)
   int24_t(const float &in) {
   ^
libraries/audio-tools/src/AudioBasic/Int24.h:34:3: note: candidate: audio_tools::int24_t::int24_t(const int64_t&)
   int24_t(const int64_t &in) {
   ^
libraries/audio-tools/src/AudioBasic/Int24.h:30:3: note: candidate: audio_tools::int24_t::int24_t(const int32_t&)
   int24_t(const int32_t &in) {
   ^
libraries/audio-tools/src/AudioBasic/Int24.h:26:3: note: candidate: audio_tools::int24_t::int24_t(const int16_t&)
   int24_t(const int16_t &in) {
   ^

exit status 1

Compilation error: exit status 1

pschatzmann, 2023-04-08:
I am wondering which Arduino implementation does not support Print flush()!
What processor implementation did you select ?

pschatzmann, 2023-04-09:
I am closing this issue because it has nothing to do with my library

closed 2023-04-09 by pschatzmann

pschatzmann, 2023-04-12:
I am not supporting any old ESP32 releases any more: The current version is 2.0.7!
Please upgrade to an actual version!