pschatzmann/arduino-audio-tools #766: Latest code downloaded can't compile in Arduino IDE

spawn451, 2023-04-11:
I have just download your latest code and my sketch  based on "streams-i2s-webserver_wav.ino" doesn't compile anymore (works with previous realease).

In file included from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/AudioCodecs.h:13:0,
                 from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:71,
                 from C:\Users\DJE\Documents\Arduino\cam\cam.ino:28:
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecBase64.h: In member function 'virtual void audio_tools::EncoderBase64::begin()':
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecBase64.h:216:16: error: 'class Print' has no member named 'flush'
       p_print->flush();
                ^
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecBase64.h: In member function 'void audio_tools::EncoderBase64::encodeLine(uint8_t*, size_t)':
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioCodecs/CodecBase64.h:294:14: error: 'class Print' has no member named 'flush'
     p_print->flush();
              ^
In file included from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioStreamsConverter.h:3:0,
                 from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:65,
                 from C:\Users\DJE\Documents\Arduino\cam\cam.ino:28:
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h: In instantiation of 'T audio_tools::ResampleStream::getValue(T*, float, int) [with T = audio_tools::int24_t]':
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:340:31:   required from 'size_t audio_tools::ResampleStream::write(Print*, const uint8_t*, size_t, size_t&) [with T = audio_tools::int24_t; size_t = unsigned int; uint8_t = unsigned char]'
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:285:62:   required from here
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/ResampleStream.h:367:24: error: conversion from 'double' to 'audio_tools::int24_t' is ambiguous
     return round(result);
                        ^
In file included from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools/AudioTypes.h:8:0,
                 from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimerBase.h:4,
                 from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTimer/AudioTimer.h:10,
                 from c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioTools.h:52,
                 from C:\Users\DJE\Documents\Arduino\cam\cam.ino:28:
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Int24.h:42:3: note: candidate: audio_tools::int24_t::int24_t(const float&)
   int24_t(const float &in) {
   ^
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Int24.h:34:3: note: candidate: audio_tools::int24_t::int24_t(const int64_t&)
   int24_t(const int64_t &in) {
   ^
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Int24.h:30:3: note: candidate: audio_tools::int24_t::int24_t(const int32_t&)
   int24_t(const int32_t &in) {
   ^
c:\Users\DJE\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioBasic/Int24.h:26:3: note: candidate: audio_tools::int24_t::int24_t(const int16_t&)
   int24_t(const int16_t &in) {
   ^

exit status 1

Compilation error: exit status 1

closed 2023-04-11 by pschatzmann

spawn451, 2023-04-11:
Really weird that  compilation works with previous release. Nothing has change in my setup, I will try to found out the reason why I got this error. Thanks :)

pschatzmann, 2023-04-12:
If you check in the board manager: what version are you using ?

spawn451, 2023-04-12:
[image]

spawn451, 2023-04-12:
This the version I am using (the latest in the list)
[image]

pschatzmann, 2023-04-12:
This is crazy! Why are you still using this old, obsolete release?
The actual release is 2.0.7!

spawn451, 2023-04-12:
I thought the version was really old. I will adapt this. Thanks a lot again for your availability.

pschatzmann, 2023-04-12:
I committed a correction that makes it compile with this old version again, but I still recommend to upgrade!