pschatzmann/arduino-audio-tools #770: soc/dac_channel.h is missing on ESP32-C3

alexflint, 2023-04-13:
I am trying to use this library with ESP32-C3. I get the following error (I have defined USE_I2S_ANALOG):

```
fatal error: soc/dac_channel.h: No such file or directory
```

I guess this is because ESP32-C3 doesn't have a digital to analog converter. However, I really just want to read data from an analog microphone, so I technically only need the ADC, not the DAC, and ESP32-C3 does have the ADC. Is there any way I can get the one without the other?

pschatzmann, 2023-04-13:
Unfortunately the C3 does not support reading analog using the DMA via the I2S api.
I recommend to use an I2S microphone. The quality will also be much better ...

alexflint, 2023-04-14:
Thank you for the pointer I might try working with that!

closed 2023-04-14 by alexflint

pschatzmann, 2023-04-14:
Others might also be interested in this: Feel free to submit a pull request if you have something working