pschatzmann/arduino-audio-tools #771: Missing Example

SebasKam, 2023-04-14:
Hey there.

I was doing a project with the BluetoothA2DPSource library, but I need to play an audio after playing the first one and using the a2dp_source.start  function again gives an error. So I looked up the arduino audio tools class and I saw in the examples that there is an example called "A Simple SdFat Audio Player with Output to Bluetooth Speaker" that might be useful for my project, the thing is that it seems that the example is not available anymore, can you help me with that? 

 

[image]

pschatzmann, 2023-04-14:
All examples are logically named!
Here are the links to all examples that should work 

- https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-basic-api/base-player-a2dp
- https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-player/player-sdfat-a2dp

Of cause you can use any [documented sd API](https://github.com/pschatzmann/arduino-audio-tools/wiki/The-Audio-Player-Class)

SebasKam, 2023-04-14:
Sorry for bother you again. I was trying to run the examples you gave me and there are some errors. I have installed all the libraries and these errors appeared. 

In file included from C:\Users\kamse\OneDrive\Escritorio\Tesis\mp3\mp3.ino:14:0:
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:23:1: error: 'SynchronizedBufferRTOS' does not name a type
 SynchronizedBufferRTOS<uint8_t>a2dp_buffer{A2DP_BUFFER_SIZE * A2DP_BUFFER_COUNT, A2DP_BUFFER_SIZE, portMAX_DELAY, portMAX_DELAY};
 ^
In file included from C:\Users\kamse\OneDrive\Escritorio\Tesis\mp3\mp3.ino:14:0:
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In member function 'bool audio_tools::A2DPStream::begin(audio_tools::A2DPConfig)':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:123:13: error: 'a2dp_buffer' was not declared in this scope
             a2dp_buffer.resize(cfg.bufferSize);
             ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In member function 'virtual size_t audio_tools::A2DPStream::write(const uint8_t*, size_t)':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:194:29: error: 'a2dp_buffer' was not declared in this scope
                 while(len > a2dp_buffer.availableForWrite()){
                             ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:204:29: error: 'a2dp_buffer' was not declared in this scope
             size_t result = a2dp_buffer.writeArray(data, len);
                             ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In member function 'virtual size_t audio_tools::A2DPStream::readBytes(uint8_t*, size_t)':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:220:29: error: 'a2dp_buffer' was not declared in this scope
             size_t result = a2dp_buffer.readArray(data, len);
                             ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In member function 'virtual int audio_tools::A2DPStream::available()':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:228:20: error: 'a2dp_buffer' was not declared in this scope
             return a2dp_buffer.available();
                    ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In member function 'virtual int audio_tools::A2DPStream::availableForWrite()':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:235:20: error: 'a2dp_buffer' was not declared in this scope
             return a2dp_buffer.availableForWrite();
                    ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In static member function 'static int32_t audio_tools::A2DPStream::a2dp_stream_source_sound_data(uint8_t*, int32_t)':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:284:30: error: 'a2dp_buffer' was not declared in this scope
                 result_len = a2dp_buffer.readArray((uint8_t*)data, len);
                              ^
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h: In static member function 'static void audio_tools::A2DPStream::a2dp_stream_sink_sound_data(const uint8_t*, uint32_t)':
C:\Users\kamse\OneDrive\Documents\Arduino\libraries\arduino-audio-tools-main\src/AudioLibs/AudioA2DP.h:312:39: error: 'a2dp_buffer' was not declared in this scope
                 uint32_t result_len = a2dp_buffer.writeArray(data, len);

closed 2023-04-15 by pschatzmann