pschatzmann/arduino-audio-tools #773: Please add some information to the pcm5102 breakout board description

haecksenwerk, 2023-04-15:
First of all, thanks for this great library. I had a lot of fun making a little internet radio with it :) 

The [pcm5102 DAC module](https://github.com/pschatzmann/arduino-audio-tools/wiki/External-DAC#hiletgo-pcm5102-i2s-iis-lossless-digital-audio-dac-decoder) did not work for me right away. I had to add solder bridges on the bottom side. The bridges H1-H4 are connected to the not assembled pin header. The mapping is as follows:

H1L (FLT) = L
H2L (DMP) = L
H3L (XMT) = H
H4L (FMT) = L  (most important, sets the audio format to I2S)

[image]

pschatzmann, 2023-04-15:
Did you try to change the i2s format in the configuration ?

haecksenwerk, 2023-04-15:
@pschatzmann No, I wasn't aware of that option

pschatzmann, 2023-04-15:
I updated the documentation. It seems that the DACs from Adafruit are also not using standard I2S...

closed 2023-04-15 by pschatzmann