pschatzmann/arduino-audio-tools #775: Deleting EncodedAudioStream object

Bleach665, 2023-04-16:
First many thanks for amazing library.

Board:Esp32-wroom-32U
Other settings from platform.io:
````
platform = https://github.com/platformio/platform-espressif32.git
board = esp32dev
framework = arduino
lib_deps = 
	etherkit/Etherkit Si5351
	etherkit/Etherkit JTEncode
	;Hash
	links2004/WebSockets
	blynkkk/Blynk@^1.0.1
	gyverlibs/GyverNTP
	https://github.com/pschatzmann/arduino-libhelix.git
    https://github.com/pschatzmann/arduino-audio-tools.git
````

After executing StartStream (code below) radio stream hear perfectly on GPIO 26. Unfortunately on deleting EncodedAudioStream I receive exception (pasted below).

Code to reproduce:
<details>

````
URLStream * pUrlStream = NULL;
AnalogAudioStream *pAnalogAS = NULL;
EncodedAudioStream *pDecAS = NULL;
StreamCopy *pStreamCopier = NULL;
MP3DecoderHelix * pMP3DecoderHelix = NULL;

void StopStream()
{
  if(!pUrlStream)
    return ;
  
  pStreamCopier->end();
  pUrlStream->end();
  pDecAS->end();
  pDecAS->setSt
  //pDecAS->setNotifyAudioChange(NULL);
  pAnalogAS->end();

  delete pStreamCopier;
  pStreamCopier = NULL;
  delete pUrlStream;
  pUrlStream = NULL;
  // delete pAnalogAS;
  // pAnalogAS = NULL;
  delete pDecAS; <---------------------------------------------------------- Error occurred here
  pDecAS = NULL;
  

  Serial.println("Audio stopped.");
  webSocket.broadcastTXT("Audio stopped.");
}

void StartStream()
{
  //StopStream();

  pUrlStream = new URLStream;
  pAnalogAS = new AnalogAudioStream; // final output of decoded stream
  pMP3DecoderHelix = new MP3DecoderHelix();
  pDecAS = new EncodedAudioStream(pAnalogAS, pMP3DecoderHelix); // Decoding stream
  pStreamCopier = new StreamCopy(*pDecAS, *pUrlStream); // copy url to decoder

  // setup analog
  auto config = pAnalogAS->defaultConfig(TX_MODE);
  pAnalogAS->begin(config);

  // setup I2S based on sampling rate provided by decoder
  //pDecAS->setNotifyAudioChange(*pAnalogAS);
  pDecAS->begin();

  pUrlStream->begin(pRadioRoksURL);

  Serial.println("Audio started.");
  webSocket.broadcastTXT("Audio started.");
}

````
</details>

Error message:
<details>

````
CORRUPT HEAP: Bad head at 0x3ffd0980. Expected 0xabba1234 got 0x3ffd3184

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace: 0x40083865:0x3ffb1ba0 0x4008c171:0x3ffb1bc0 0x40091ef9:0x3ffb1be0 0x40091b6b:0x3ffb1d10 0x40083d29:0x3ffb1d30 0x40091f29:0x3ffb1d50 0x4016a491:0x3ffb1d70 0x400d761d:0x3ffb1d90 0x400d7677:0x3ffb1db0 0x400d7738:0x3ffb1dd0 0x400d7751:0x3ffb1df0 0x400d77c7:0x3ffb1e10 0x400d9cba:0x3ffb1e30 0x4018431a:0x3ffb1ea0 0x401849f6:0x3ffb1ec0 0x401849b1:0x3ffb1ef0 0x400e3cec:0x3ffb1f10 0x40184ab1:0x3ffb1f40 0x400e3f1a:0x3ffb1f60 0x400e3fa9:0x3ffb1f90 0x400e4206:0x3ffb1fb0 0x400e426a:0x3ffb1ff0 0x400e3f1a:0x3ffb2010 0x400e3fa9:0x3ffb2040 0x400e403d:0x3ffb2060 0x400e41b1:0x3ffb20a0 0x400e426a:0x3ffb20e0 0x400e3f1a:0x3ffb2100 0x400e3fa9:0x3ffb2130 0x400e403d:0x3ffb2150 0x400e4096:0x3ffb2190 0x400e424f:0x3ffb21d0 0x400e376d:0x3ffb21f0 0x400e37b2:0x3ffb2230 0x400e37ce:0x3ffb2250 0x400dab13:0x3ffb2270 0x400ec0d5:0x3ffb2290

ELF file SHA256: 8670cf09f1092fc2

Rebooting...
````
</details>

pschatzmann, 2023-04-16:
I committed a correction: However this is not a recommended use case and it has definitely not been tested!
The recommended way is to define global variables and when you are done you can call end() on them.

If you use a solution off the beaten track, you should be able to analyse the stack trace and submit corrections yourself....

closed 2023-04-16 by pschatzmann

Bleach665, 2023-04-16:
@pschatzmann Unfortunately now I receive another error:
````
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ee73a  PS      : 0x00060530  A0      : 0x800d4217  A1      : 0x3ffb1d50  
A2      : 0x00000000  A3      : 0x3ffc582c  A4      : 0x3ffc582c  A5      : 0x00000000
A6      : 0x00000000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1d50  
A10     : 0x00000005  A11     : 0x00000003  A12     : 0x0000000a  A13     : 0x00000000
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4008a0b5  LEND    : 0x4008a0c5  LCOUNT  : 0xfffffffe

Backtrace: 0x400ee737:0x3ffb1d50 0x400d4214:0x3ffb1d90 0x400d429c:0x3ffb1db0 0x400d7673:0x3ffb1dd0 0x400d7689:0x3ffb1df0 0x400d7783:0x3ffb1e10 0x400d9cce:0x3ffb1e30 0x4018432e:0x3ffb1ea0 0x40184a0a:0x3ffb1ec0 0x401849c5:0x3ffb1ef0 0x400e3d00:0x3ffb1f10 0x40184ac5:0x3ffb1f40 0x400e3f2e:0x3ffb1f60 0x400e3fbd:0x3ffb1f90 0x400e421a:0x3ffb1fb0 0x400e427e:0x3ffb1ff0 0x400e3f2e:0x3ffb2010 0x400e3fbd:0x3ffb2040 0x400e4051:0x3ffb2060 0x400e41c5:0x3ffb20a0 0x400e427e:0x3ffb20e0 0x400e3f2e:0x3ffb2100 0x400e3fbd:0x3ffb2130 0x400e4051:0x3ffb2150 0x400e40aa:0x3ffb2190 0x400e4263:0x3ffb21d0 0x400e3781:0x3ffb21f0 0x400e37c6:0x3ffb2230 0x400e37e2:0x3ffb2250 0x400dab27:0x3ffb2270 0x400ec0e9:0x3ffb2290

ELF file SHA256: 4421ac3d7e732ac3

Rebooting...
```` 

> The recommended way is to define global variables and when you are done you can call end() on them.

I look examples and understand that. The reason why I do this that project beside decode audio stream(and transmit it), will perform other memory-consuming tasks. So some time I free memory by deleting audio-tools objects.

pschatzmann, 2023-04-16:
As I have written above: you need to analyze the stack-trace.
Feel free to submit your corrections with some pull requests...