pschatzmann/arduino-audio-tools #777: Playing mp3 from SD stutters

fehlfarbe, 2023-04-16:
Hi,

thanks for your great work! I'm working on an ESP32 radio that can play FM, webstreams and mp3s from an SD card. FM (via I2S -> I2S) and the urlstream works perfect but playing mp3 files from SD card stutters and makes noises. I created a minimal example that simulates the behavior of my AudioManager class to make sure it's not an interference with my web control or so:

```
#include <Arduino.h>
#include <AudioTools.h>
#include <AudioCodecs/CodecMP3Helix.h>
#include <FS.h>
#include <SD.h>

// Digital I/O used
#define I2S_DOUT 25
#define I2S_BCLK 4
#define I2S_WS 27
#define I2S_DIN 35

I2SStream i2s;
EncodedAudioStream decoder; // Decoding stream
VolumeStream volumeStream;
StreamCopy copier;   // copy url to decoder
MP3DecoderHelix mp3decoder;
MeasuringStream out(50, &Serial); // final output of decoded stream
fs::File file;

void setup() {
    Serial.begin(115200);
    AudioLogger::instance().begin(Serial, AudioLogger::Warning);

    // Initialize SD
    while (!SD.begin())
    {
        Serial.println("Card Mount Failed");
        delay(1000);
        while (true)
            ;
    }

    uint8_t cardType = SD.cardType();
    if (cardType == CARD_NONE)
    {
        Serial.println("No SD card attached");
        while (true)
            ;
    }

    // audio stream stuff
    decoder.setOutput(&volumeStream);
    decoder.setDecoder(&mp3decoder);

    // setup volumeStream
    VolumeStreamConfig cfg;
    cfg.channels = 2;
    cfg.bits_per_sample = 16;
    cfg.sample_rate = 44100;
    cfg.volume = 1.0f;
    volumeStream.begin(cfg);
    volumeStream.setTarget(out);
    Serial.printf("Channels %d\n", volumeStream.audioInfo().channels);

    // setup I2S
    auto config = i2s.defaultConfig(RXTX_MODE);
    config.pin_bck = I2S_BCLK;
    config.pin_data = I2S_DOUT;
    config.pin_ws = I2S_WS;
    config.pin_data_rx = I2S_DIN;
    config.auto_clear = true;
    i2s.begin(config);

    // reset / stop copier and decoder before opening a new stream/fm/file
    copier.end();
    decoder.end();

    // activate decoder
    decoder.setNotifyAudioChange(out);
    decoder.begin();
    // open file
    file = SD.open("/songs/URSULA - URSULA - 11 Auf der anderen Seite.mp3");
    copier.begin(decoder, file);
}

void loop() {
    copier.copy();
}
```

The MeasuringStream output looks good and I don't get warnings from the AudioLogger:
```
==> Bytes per second: 108000
==> Bytes per second: 315000
==> Bytes per second: 224000
==> Bytes per second: 201000
==> Bytes per second: 232000
==> Bytes per second: 192000
==> Bytes per second: 178000
==> Bytes per second: 194000
==> Bytes per second: 217000
==> Bytes per second: 198000
==> Bytes per second: 139000
==> Bytes per second: 190000
...
```

I tried different mp3 files and removing the VolumeStream also makes no difference.

Am I missing something?

pschatzmann, 2023-04-16:
I guess you tried to use some mp3 with some "crazy" bitrates (in Audacity wording).
Try to increase the mp3 buffer sizes, that might help...

fehlfarbe, 2023-04-17:
I tried several files with 128 kbit/s but all did stutter. Maybe I should've also read the documentation of your Helix decoder. Increasing the mp3 frame size from 1600 to 3200 did the job :smile: Thanks a lot!

There was a Github Repo about the RDA5807FP radio module but it seems not available anymore: https://github.com/ieee1076/I2S_OUTPUT_FM_TUNER

I added some code to enable and configure the I2S mode to this repo: https://github.com/mathertel/Radio
There is an example at: https://github.com/mathertel/Radio/blob/master/examples/TestRDA5807FP/TestRDA5807FP.ino

Unfortunately I didn't find a breakout board but you don't need many components to make it work: 
[image]

You can find my project at https://github.com/fehlfarbe/ESP32-Alarm-clock/tree/rda5807fp/ The Readme still shows my old version with RDA5807M that doesn't support I2S. I also created a small PCB with KiCAD that has an ESP32 and UDA1334A breakout board and the RDA5807FP radio chip: https://github.com/fehlfarbe/ESP32-Alarm-clock/tree/rda5807fp/hardware/pcb/mp3alarm

closed 2023-04-17 by fehlfarbe