pschatzmann/arduino-audio-tools #784: streams-i2s-webserver_wav.ino" and volume control

spawn451, 2023-04-17:
I created the sketch below which allows you to stream the audio and increase the volume of the microphone. The latter works very well however I call the "audio_server.begin" function twice because my parameters are not identical. Is this the right way to proceed or is it possible to make the call only once ? (I have already thought of using the AudioConfigLocal file so as not to see certain parameters in the sketch but without success). Thank you in advance for your help.

AudioWAVServer audio_server("ssid","password");

I2SStream i2sStream; // Access I2S as stream

VolumeStream volume(i2sStream);

// start i2s input with default configuration
Serial.println("starting I2S...");

auto cfg = i2sStream.defaultConfig(RX_MODE);

cfg.i2s_format = I2S_STD_FORMAT;
cfg.pin_ws = 2;
cfg.pin_bck = 14;
cfg.pin_data = 15;
cfg.port_no = 1;
cfg.sample_rate = 22050;
cfg.channels = 1;
cfg.bits_per_sample = 32;

i2sStream.begin(cfg);
Serial.println("I2S started");

// start data sink
audio_server.begin(i2sStream, cfg);

auto config = volume.defaultConfig();
config.allow_boost = true;  // uses linear control allowing values > 1.0. 
config.sample_rate = 22050;
config.channels = 1;
config.bits_per_sample = 32;
volume.begin(config);
volume.setVolume(5); // Too big values lead to clipping!

// start volume
audio_server.begin(volume, config);

pschatzmann, 2023-04-17:
For questions like this you should rather use a discussion.
Issues should be used to report bugs!

closed 2023-04-17 by pschatzmann

spawn451, 2023-04-17:
OK thank you, sorry I will use the discussion next time, my bad.