pschatzmann/arduino-audio-tools #813: Decoder write is not blocking write

aqildad-create, 2023-04-29:
I store data in ring buffer and then write to decoder ,i observe writing 45000 bytes to decoder take only 25 milliseconds  . which is very fast writing . 

PRE-TEST : decoder is working and encoded data is correct because when  i write directly from esponow i get proper audio . 

so i concluded that  decoder.write is not blocking write , so how to use decoder from BUFFERstream like ring buffer in my case ? 

below is minimal code : 

RingBufferStream MyBuffer(50000);
SBCDecoder dec;
EncodedAudioStream decStream(i2s, dec);
.
loop ----->>>>>>>>>>>>>>>>>>>>>>>>>for 45000 bytes of encoded data take only 25 millisconds to write ..  
    copied = MyBuffer.readBytes((uint8_t *)buffrr , 76 );   decStream.write((uint8_t *)buffrr , copied );

pschatzmann, 2023-04-29:
When you write you need to check the result to know how much bytes have been written!
If this is on the receiving side of ESP-now, why don't you just use the built in buffer. You can increase it or decrease if you are not happy with the default.

By the way this is not a bug!

closed 2023-04-29 by pschatzmann

aqildad-create, 2023-04-29:
**When you write you need to check the result to know how much bytes have been written!**
 copied = MyBuffer.readBytes((uint8_t *)buffrr , 76 );   
 int wrtee=decStream.write((uint8_t *)buffrr , copied );-->> and it  correct   76 in my case

**If this is on the receiving side of ESP-now, why don't you just use the built in buffer. You can increase it or decrease if you are not happy with the default.**
for now I am using basic espnow  ,i  am not using your esp now stream , due to some reason .  so which built in buffer espnow have . pls guide i dont now . I search from internet but cant find much detail of this buffer 

**If you want to limit the sending speed you can use a Throttle.**
I try to limit like this . Is this proper way ? and how much Millis should  i Throttle .  Is it need some calculation ?
  copied = MyBuffer.readBytes((uint8_t *) buffrr , 76 );   
    int wrtee=decStream.write((uint8_t *)buffrr , copied ); 
      smalldelay=millis();
    while(millis()- smalldelay < 4 );

**By the way this is not a bug! So why do you create an issue for this?**
 though may be write decoder is blocking write . Its my mistake

pschatzmann, 2023-04-29:
The reason not to use my stream implementation has just gone away yesterday!

If you use the basic espnow api there is no buffer: You can look at my implementation how to manage the confirmations. If I remember right I am waiting for the buffer to have enough space before sending the confirmation. Be careful, I think you need to use a mutex prevent concurrency issues.

For the use of the Throttle I suggest to have a look at the UDP example.