pschatzmann/arduino-audio-tools #829: Allocation of OutputMixer object and OutputMixer.setOutputStream()

leelx78, 2023-05-08:
Hi @pschatzmann just a minor thing. I needed to allocate an OutputMixer object inside another class but I noticed there is no constructor without arguments, nor a way to set the output stream so I added it as below: 

```
 class OutputMixer : public Print {
 
  public:
  OutputMixer(){

  };
  void setOutputStream(Print &finalOutput){
    p_final_output = &finalOutput;
  }

  <the rest is unchanged>
  ``` 
  
  Can this be added to the code? Might give more flexibility since we can configure and reconfigure the mixer later after creation.
  
  Thanks!

pschatzmann, 2023-05-08:
Sur, just submit a pull request with your change

pschatzmann, 2023-05-18:
What's holding you up ? Can I still expect your contribution ?

leelx78, 2023-05-20:
Sorry @pschatzmann I'm not familiar with pull requests so I left it there for the time being. 

Are you pasting it from there? I'm fine with both setOutputStream() and setOutput() names.

Thanks

pschatzmann, 2023-05-20:
I just committed some new functionality!
Pull requests are not too difficult, you would need to fork this project first and then you can make your changes in your fork.

closed 2023-05-20 by pschatzmann