pschatzmann/arduino-audio-tools #830: strange slow playback and stutter with "player-url_icy-i2s"-example

art51go, 2023-05-09:
I compiled the "player-url_icy-i2s"-example in Arduino 1.8.19 for a DOIT ESP32 DevKitV1 with pcm5102A DAC. Playing a radio-station there are strange glitches (or stuttering) - sample:

[sample.zip](https://github.com/pschatzmann/arduino-audio-tools/files/11429082/sample.zip)

To make it more clear, I recorded my voice just counting and played it through a web-server. 

[player-url_icy-i2s---link-1-10.zip](https://github.com/pschatzmann/arduino-audio-tools/files/11428952/player-url_icy-i2s---link-1-10.zip)

In the first part my voice is too deep and it stutters. The second part is, how it should sound. All samples where recorded with a microphone from the loudspeaker. Compiling the example with MSCode and PlatformIO had the the same result.

pschatzmann, 2023-05-09:
I think you just ignored the helix documentation!

closed 2023-05-09 by pschatzmann

art51go, 2023-05-09:
It would be very kind, if you explain this a little, what you mean with
this. Is this the normal behavior, if it is run with a 128 KBit/s
mp3-stream?

pschatzmann, 2023-05-09:
These effects you would usually get with high bitrates and then you would need to increase the buffers.
On my mp3 files I never had any issues myself...

art51go, 2023-05-09:
So we will have to roll back to VS1053-solutions to have reliable and good
quality sound?