pschatzmann/arduino-audio-tools #843: ResampleStream.h exception handling

leelx78, 2023-05-15:
Hi @pschatzmann I noticed that if for any reason the ResampleStream class cannot write to the output (e.g. `p_out->availableForWrite() < sizeof(T)`) it will get stuck and not recover.

Shouldn't there be an exit condition when such exception happens, e.g.:
```
// process all samples
    while (idx < frames) {
      T frame[info.channels];
      for (int ch = 0; ch < info.channels; ch++) {
        T result = getValue<T>(data, idx, ch);
        if (p_out->availableForWrite() < sizeof(T)) {
          LOGE("Could not write");
          return 0; //Returning 0 and dropping the packet? Or should it be "return written;"? (This line is new)
        }
        frame[ch] = result;
      }
```

or should we force a `p_out->flush()`? Seems a bit extreme though. 

Any thoughts on the added `return 0;` or `return written;` statement?

leelx78, 2023-05-15:
I have a lot of encoded voice data coming in from UDP (mono, low bitrate) and then it's converted to stereo (48kHz/16 bit) which means that each packet is expanded a lot once received. My RX buffer is about 20000 bytes which is enough for this but sometimes the network might accumulate many UDP packets which are then received all at once and would not fit on the buffer.

Although this is not ResampleStream 's fault, I think there should be some exception handling that makes this class fail gracefully in this case. In the original code there is no return statement after LOGE() and that tends to freeze the system. Adding a return seems to improve things.

Any drawbacks?

pschatzmann, 2023-05-15:
I think the easiest would be to make a check before the loop and return 0 if there is not enough space.
The other option is to add this check in the loop and return the processed bytes.
Returning 0 in the loop will just lead that the processed bytes will be reprocessed again.

leelx78, 2023-05-15:
ok so how about
```
        if (p_out->availableForWrite() < sizeof(T)) {
          LOGE("Could not write - DROPPING PACKET");
          return written;
        }
        
 ```

I will also do a check beforehand but still think it's important it fails as gracefully as possible

pschatzmann, 2023-05-15:
Hmm, difficult to foresee all implications.  
You would need to do some testing both on the input and output side

uknod, 2023-05-18:
Appologies if this is the wrong place to put this, but I am a novice at posting.
I have tried to use your MP3 sd player to bluetooth speaker, and it works, except the playback is at about 10x speed, any idea why thi sis happening?

pschatzmann, 2023-05-18:
Double check in the log that decoding is giving 2 channels with 44100 samples per second. 
If this is not the case you need to change  the format before you send it to A2DP

uknod, 2023-05-18:
Hi

 

Where can I find the log?

 

Geoff.

pschatzmann, 2023-05-18:
Reading the documentation usually helps!

uknod, 2023-05-18:
Lol, I thought I had, I’ve looked everywhere for it. I have literally spent 5 hours trying to get it working, unfortunately, not everyone is an expert at this sort of thing.

 

I am trying my best.

uknod, 2023-05-19:
OK I have read every document I can on this, and no where can I find anything about a log file???????? Please can you not just tell me where I can find it so I find the problem with my x10 speed files

pschatzmann, 2023-05-19:
The logging is described in the README and every example starts by configuring the logger

uknod, 2023-05-19:
Hi.
OK I now know how arduino logging works, I thought it was just the monitor, didn't realise there were many levels of logging, and I was looking for a file. Now the logging is working I can see the problem and fixed it, it was as you said. It was a mono file instead of sterio that was causing the problem.
So now its working, I assume the player.copy is a routine that you have to run often to keep the player scanning. 
To be honest I think it is working very speradically, I had sound once, the log file has all kinds of incombatability issues, I think I have spent long enough on this and might just have to revert to using wired speakers. This may be a project for another day. Very clever though, you have put a lot of work into this I can see.

closed 2023-08-06 by pschatzmann