pschatzmann/arduino-audio-tools #845: communication not working with binary container

aqildad-create, 2023-05-16:
Without container communication work but with container getting log error .

below s sender and receiver logic with log error at transmitter . 

Sender--------------------------------------------------------
I2SStream i2sIn; 
ESPNowStream now;
BinaryContainerEncoder  BinaryCon( new SBCEncoder() );
EncodedAudioStream encoder( &now,  &BinaryCon   );
StreamCopy copier(encoder, i2sIn); 
.
uint8_t myBufferbuffer[30];
BinaryCon.writeMeta(myBufferbuffer, 10) ;

Receiver :------------------------------------------------------
MultiOutput Multioutt;
MetaDataOutput outMeta; 
ESPNowStream now;
I2SStream out;
EncodedAudioStream decoder(&Multioutt, new BinaryContainerDecoder(new SBCDecoder()));
StreamCopy copier(decoder, now);
.
Multioutt.add(outMeta);
Multioutt.add(out);
outMeta.setCallback(printMetaData);
outMeta.begin();
.
.
.
12:40:50.080 -> [W] Communication.h : 230 - Write failed - retrying again
12:40:52.076 -> [W] Communication.h : 230 - Write failed - retrying again
12:40:54.078 -> [W] Communication.h : 230 - Write failed - retrying again

pschatzmann, 2023-05-18:
I committed a [correction and an example](https://github.com/pschatzmann/arduino-audio-tools/commit/28e44abacc0ffef273e7aca557470b8b9c2c82d9)

closed 2023-05-18 by pschatzmann

aqildad-create, 2023-05-18:
i try above example wth espnow .. i am receiving correct meta successfully

 But for stream data  it showing log as write fail but at receiver end espnow data is coming but showinglog as [W] ContainerBinary.h : 404 - Unsupported tye
and some time showing no log but no audio out at receiver 

.. why it is like this ? 

Sender--------------------------------------------------------
I2SStream i2sIn; // coming from the mic
ESPNowStream now;
BinaryContainerEncoder cont_enc(new SBCEncoder());
EncodedAudioStream encoder(&now, &cont_enc);
 StreamCopy copierr(encoder, i2sIn); 
Sender log:---------------------------------------------------
 
9:47:01.215 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:03.206 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:05.244 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:07.224 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:09.234 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:11.274 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:13.283 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:15.272 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:17.276 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:19.302 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:21.303 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:23.299 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:25.324 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:27.356 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:29.364 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:31.366 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:33.359 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:35.384 -> [W] Communication.h : 230 - Write failed - retrying again
19:47:37.384 -> [W] Communication.h : 230 - Write failed - retrying again
 

Receiver :------------------------------------------------------
I2SStream out;
BinaryContainerDecoder cont_dec(new SBCDecoder());
EncodedAudioStream decoder(&out, &cont_dec);
StreamCopy copier(decoder, now);
Receiver log:--------------------
20:11:17.574 -> Receiver started...
20:11:18.887 -> ====> opus: volume 0.50
.
.
19:47:54.609 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.609 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.649 -> [W] ContainerBinary.h : 404 - Unsupported tye
19:47:54.684 -> [W] ContainerBinary.h : 404 - Unsupported tye

aqildad-create, 2023-05-18:
after log to info at receiver i am getting this at receiver but no audio out 
.
.
20:25:30.326 -> [I] Communication.h : 109 - setting mac 94:E6:86:04:16:EC
20:25:30.326 -> [I] Communication.h : 133 - Setting ESP-NEW rate
20:25:30.372 -> 
20:25:30.372 -> mac: 94:E6:86:04:16:EC
20:25:30.372 -> [I] Communication.h : 315 - esp_now_init: W
20:25:30.372 -> [I] Communication.h : 161 - addPeer: 94:e6:86:03:45:ac
20:25:30.372 -> starting I2S...
20:25:30.372 -> [I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
20:25:30.372 -> [I] I2SConfig.h : 107 - port_no: 0
20:25:30.372 -> [I] I2SConfig.h : 108 - is_master: Master
20:25:30.372 -> [I] I2SConfig.h : 109 - sample rate: 32000
20:25:30.372 -> [I] I2SConfig.h : 110 - bits per sample: 16
20:25:30.372 -> [I] I2SConfig.h : 111 - number of channels: 1
20:25:30.372 -> [I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
20:25:30.372 -> [I] I2SConfig.h : 114 - auto_clear: true
20:25:30.418 -> [I] I2SConfig.h : 116 - use_apll: true
20:25:30.418 -> [I] I2SConfig.h : 121 - buffer_count:6
20:25:30.418 -> [I] I2SConfig.h : 122 - buffer_size:512
20:25:30.418 -> [I] I2SConfig.h : 126 - pin_bck: 27
20:25:30.418 -> [I] I2SConfig.h : 128 - pin_ws: 26
20:25:30.418 -> [I] I2SConfig.h : 130 - pin_data: 25
20:25:30.418 -> [I] CodecSBC.h : 44 - virtual void audio_tools::SBCDecoder::begin()
20:25:30.418 -> Receiver started...
20:25:32.216 -> [I] Buffers.h : 295 - resize: 100000
20:25:32.216 -> ====> opus: volume 0.50
20:25:32.216 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:25:42.847 -> [I] AudioTypes.h : 85 - sample_rate: 1
20:25:42.847 -> [I] AudioTypes.h : 86 - channels: 16
20:25:42.847 -> [I] AudioTypes.h : 87 - bits_per_sample: -1386842112
20:25:42.847 -> [I] StreamCopy.h : 139 - StreamCopy::copy 540 -> 540 -> 540 bytes - in 1 hops
20:25:42.847 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:25:42.847 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:25:42.847 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:25:42.847 -> [I] StreamCopy.h : 139 - StreamCopy::copy 258 -> 258 -> 258 bytes - in 1 hops
20:25:42.893 -> [I] StreamCopy.h : 139 - StreamCopy::copy 782 -> 782 -> 782 bytes - in 1 hops

aqildad-create, 2023-05-18:
above log at receiver decoder was begin empty but I correct it same as encoder begin . 

AudioInfo to  (32000, 1, 16);
decoder.begin(to);

so with this below are the logs 

20:36:44.557 -> [I] Communication.h : 109 - setting mac 94:E6:86:04:16:EC
20:36:44.557 -> [I] Communication.h : 133 - Setting ESP-NEW rate
20:36:44.557 -> 
20:36:44.557 -> mac: 94:E6:86:04:16:EC
20:36:44.557 -> [I] Communication.h : 315 - esp_now_init: W
20:36:44.557 -> [I] Communication.h : 161 - addPeer: 94:e6:86:03:45:ac
20:36:44.557 -> starting I2S...
20:36:44.557 -> [I] I2SConfig.h : 106 - rx/tx mode: TX_MODE
20:36:44.557 -> [I] I2SConfig.h : 107 - port_no: 0
20:36:44.557 -> [I] I2SConfig.h : 108 - is_master: Master
20:36:44.557 -> [I] I2SConfig.h : 109 - sample rate: 32000
20:36:44.557 -> [I] I2SConfig.h : 110 - bits per sample: 16
20:36:44.604 -> [I] I2SConfig.h : 111 - number of channels: 1
20:36:44.604 -> [I] I2SConfig.h : 112 - i2s_format: I2S_STD_FORMAT
20:36:44.604 -> [I] I2SConfig.h : 114 - auto_clear: true
20:36:44.604 -> [I] I2SConfig.h : 116 - use_apll: true
20:36:44.604 -> [I] I2SConfig.h : 121 - buffer_count:6
20:36:44.604 -> [I] I2SConfig.h : 122 - buffer_size:512
20:36:44.604 -> [I] I2SConfig.h : 126 - pin_bck: 27
20:36:44.604 -> [I] I2SConfig.h : 128 - pin_ws: 26
20:36:44.604 -> [I] I2SConfig.h : 130 - pin_data: 25
20:36:44.604 -> [I] CodecSBC.h : 44 - virtual void audio_tools::SBCDecoder::begin()
20:36:44.604 -> Receiver started...
20:36:45.531 -> [I] Buffers.h : 295 - resize: 100000
20:36:45.531 -> ====> opus: volume 0.50
20:36:45.531 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:53.873 -> [I] AudioTypes.h : 85 - sample_rate: 1
20:36:53.873 -> [I] AudioTypes.h : 86 - channels: 16
20:36:53.873 -> [I] AudioTypes.h : 87 - bits_per_sample: -1386842112
20:36:53.873 -> [I] StreamCopy.h : 139 - StreamCopy::copy 536 -> 536 -> 536 bytes - in 1 hops
20:36:53.873 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:53.873 -> [I] StreamCopy.h : 139 - StreamCopy::copy 12 -> 12 -> 12 bytes - in 1 hops
20:36:53.873 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1016 -> 1016 -> 1016 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1016 -> 1016 -> 1016 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 766 -> 766 -> 766 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 262 -> 262 -> 262 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 857 -> 857 -> 857 bytes - in 1 hops
20:36:53.920 -> [I] StreamCopy.h : 139 - StreamCopy::copy 175 -> 175 -> 175 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1020 -> 1020 -> 1020 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 766 -> 766 -> 766 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 262 -> 262 -> 262 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 770 -> 770 -> 770 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 262 -> 262 -> 262 bytes - in 1 hops
20:36:53.967 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1020 -> 1020 -> 1020 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 12 -> 12 -> 12 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1020 -> 1020 -> 1020 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.013 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 801 -> 801 -> 801 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 227 -> 227 -> 227 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1020 -> 1020 -> 1020 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:54.060 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.106 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.153 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1016 -> 1016 -> 1016 bytes - in 1 hops
20:36:54.153 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
[... cut ...]
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1016 -> 1016 -> 1016 bytes - in 1 hops
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 16 -> 16 -> 16 bytes - in 1 hops
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 8 -> 8 -> 8 bytes - in 1 hops
20:36:54.200 -> [I] StreamCopy.h : 139 - StreamCopy::copy 766 -> 766 -> 766 bytes - in 1 hops

aqildad-create, 2023-05-18:
after further testing with out Binary Container  i  am getting proper audio  so i can say now their is some issue with binary container .