pschatzmann/arduino-audio-tools #849: i2s to webserver example

khan46000, 2023-05-17:
Hope you are having productive day!
I am using i2s to webserver example with esp-eye.Even after configuring i2s pins i am not getting any sound on the webpage.Instead the audio player loads and "stop client " is what i get on serial monitor.

pschatzmann, 2023-05-18:
https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working

You can also check if you get a valid wav file by downloading the data eg. with wget...

closed 2023-05-18 by pschatzmann

oloftangrot, 2023-05-24:
I also have an issue with the base-i2s-a2dp example. And have tried to eliminate any hardware issue by checking with a simpler source package. My conclusion when reading this "closed" issue is that arduino-audio-tools has a more general problem with using the EPS-IDF i2s driver that should be fixed.

oloftangrot, 2023-05-24:
The URL to the simpler package can be found in the issue 88, https://github.com/pschatzmann/arduino-audiokit/issues/88. I accidentally opened it in the work repo though. You do need to change the pins used to match the pins used in arduino-audio-tools (as reported in the terminal when the sketch starts).

pschatzmann, 2023-05-24:
> I also have an issue with the base-i2s-a2dp example. And have tried to eliminate any hardware issue by checking with a simpler source package. My conclusion when reading this "closed" issue is that [...]

I don't think that it is an issue with the i2s driver, but with your microphone not supporting 16 bits. Did you try to use 32 bits as suggested in your original issue?

khan46000, 2023-05-25:
i have tried ffmpeg using url but still no sound.
Did you find any solution or what other sketch can i use to listen sound captured with i2s library via wifi.
Have a nice day!

pschatzmann, 2023-05-25:
Can you try to download the data into a file e.g. with wget or curl and share the result ?
What is your input device ?
Did you consider the information provided in the Wiki ? https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC

khan46000, 2023-05-25:
i am getting ffmpeg using url but the sound is very slow and secondly i am not getting audio on web browser.
I am using esp-eye built in microphone.

khan46000, 2023-05-25:
I mean i can hear sound only when i am very close to the mic.I dont get any lag.
auto config = i2sStream.defaultConfig(RX_MODE);
    config.pin_bck = 26;
    config.pin_ws = 32;
    config.pin_data = 33;
    config.i2s_format = I2S_STD_FORMAT;
    config.sample_rate = 16000;
    config.channels = 2;
    config.bits_per_sample = 32;
  i2sStream.begin(config);
This is the configuration

pschatzmann, 2023-05-25:
https://github.com/pschatzmann/arduino-audio-tools/wiki/Volume-Control

khan46000, 2023-06-08:
I can listen to the audio on the client using ffmpeg like VLC media player but not on web browser like chrome.
[image]