pschatzmann/arduino-audio-tools #853: copier.copy() is blocking write

aqildad-create, 2023-05-18:
I2SStream i2sIn;
StreamCopy copier(&now, i2sIn);
 .
copier.copy()

I cant stop the communication or perform other task if communication write fail . 

 i need retry unlimited because every data is important  but need non blocking espnow so that  can perform small amount of other task in loop even if espnow fall to write .

pschatzmann, 2023-05-18:
If you need to do multiple things in parallel you can consider to use some tasks.
If you really need the I2S to be non blocking, feel free to extend the framework: this should be very easy to implement!
If you have it working don't forget to share your solution with a pull request.

closed 2023-05-18 by pschatzmann

aqildad-create, 2023-05-18:
I need espnow send to be non blocking . . .

aqildad-create, 2023-05-18:
actually if esp-now write fail then i cant stop the communication   copier.copy()  --> StreamCopy copier(&now, i2sIn);

I have start stop/button  if button press **communication start** and if in case the receiver power off or any reason writes fail then i cant use stop button because API continuously trying to write .

pschatzmann, 2023-05-18:
Then, why don't you just set a limit to the write_retry_count ?

aqildad-create, 2023-05-18:
because every data is important to me  . I am sending some other files also . 

what if write_retry_count complete ? API(copier will fetch new fresh data and send again) then in this case pervious fetch data will lost . 

that's why i want the retry count to -1 ( unlimited ) but via button press i can stop the sending at any time .