pschatzmann/arduino-audio-tools #865: Webserver streaming issue

johnpiker, 2023-05-25:
I m getting this on serial monitor,
 [E] StreamCopy.h : 289 - write to target has failed! (1024 bytes)
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
[I] AudioServer.h : 122 - stop client...
I am able to open the ip address but no audio is being generated
[image]

johnpiker, 2023-05-25:
[image]

johnpiker, 2023-05-25:
After this  I uploaded my program after erasing the flash. I got no error but still i was not able to get any audio on the server
[New Text Document.txt](https://github.com/pschatzmann/arduino-audio-tools/files/11563909/New.Text.Document.txt)

pschatzmann, 2023-05-25:
Before you can file any issue you need to do your homework and test your microphone or your input device and consider the documentation that is provided:
- check closed issues and discussion
- https://github.com/pschatzmann/arduino-audio-tools/wiki/It's-not-working
- https://github.com/pschatzmann/arduino-audio-tools/wiki/External-ADC

closed 2023-05-25 by pschatzmann

pitlochry, 2023-07-18:
I have the same issue with [streams-i2s-webserver_wav.ino](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-webserver/streams-i2s-webserver_wav/streams-i2s-webserver_wav.ino)

No audio can be listened to in the webserver, altough ip can be opened. Similar/Identical to @johnpiker the error message is either

> 13:39:08.798 -> [E] StreamCopy.h : 289 - write to target has failed! (1024 bytes)
13:39:08.798 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 0 bytes - in 22 hops
13:39:08.798 -> [I] AudioServer.h : 122 - stop client...

or

> 23:14:31.034 -> IP address: 192.168.178.200
23:14:37.502 -> [I] AudioServer.h : 216 - New Client.
23:14:37.568 -> [I] AudioServer.h : 368 - sendReply - Returning encoded stream...
23:14:37.568 -> [I] StreamCopy.h : 74 - buffer_size=1024
23:14:37.568 -> [I] CodecWAV.h : 496 - Writing Header
23:14:37.568 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.602 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.602 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.602 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.602 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.635 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.635 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.635 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.668 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.668 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.668 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.668 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.701 -> [I] AudioServer.h : 122 - stop client...
23:14:37.890 -> [I] AudioServer.h : 216 - New Client.
23:14:37.890 -> [I] AudioServer.h : 368 - sendReply - Returning encoded stream...
23:14:37.890 -> [I] StreamCopy.h : 74 - buffer_size=1024
23:14:37.890 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
23:14:37.890 -> [I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
... 1033 identical messages ...
23:14:53.703 -> [I] AudioServer.h : 122 - stop client...

Infos:
- the microphone INMP441 was tested and is functional with the same config in other inos
- I use the latest release 0.9.6 (I also tried to clone the github repo, which runs into compiling errors)

@pschatzmann Which homework should be done next?
@johnpiker Have you been able to solve your problem?

pschatzmann, 2023-07-19:
https://github.com/pschatzmann/arduino-audio-tools/issues/893