pschatzmann/arduino-audio-tools #888: Issue with playing mp3 from SD card with spdif output?

pwilkowski, 2023-06-05:
Hello

I am having some troubles trying to play mp3 with spdif out, no sound is being produced

Its is on ESP32
ESP-WROOM-32

My setup:

```cpp
const char *startFilePath="/";
const char* ext="mp3";
AudioSourceSDFAT* source;
SPDIFOutput* spdif;
MP3DecoderHelix decoder; 
AudioPlayer* player;

void setup() {
	AudioLogger::instance().begin(Serial, AudioLogger::Info);
	spdif = new SPDIFOutput();
	auto cfg = spdif->defaultConfig();
	cfg.pin_data = 27;
	cfg.sample_rate = 44100;
	cfg.channels = 2;
	cfg.bits_per_sample = 16;
	spdif->begin(cfg);

	source = new AudioSourceSDFAT(startFilePath, ext, 5);
	player = new AudioPlayer(*source, *spdif, decoder);
	player->begin();
}
void loop() {
	player->copy();
}
```

log

```
[I] I2SConfig.h : 113 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 114 - port_no: 0
[I] I2SConfig.h : 115 - is_master: Master
[I] I2SConfig.h : 116 - sample rate: 88200
[I] I2SConfig.h : 117 - bits per sample: 32
[I] I2SConfig.h : 118 - number of channels: 2
[I] I2SConfig.h : 119 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 121 - auto_clear: true
[I] I2SConfig.h : 123 - use_apll: true
[I] I2SConfig.h : 126 - fixed_mclk: 22579200
[I] I2SConfig.h : 128 - buffer_count:6
[I] I2SConfig.h : 129 - buffer_size:512
[I] I2SConfig.h : 143 - pin_data: 27
[I] AudioSourceSDFAT.h : 53 - SD chipSelect: 5
[I] AudioSourceSDFAT.h : 54 - SD speedMHz: 2
[I] AudioSourceSDFAT.h : 55 - ext: mp3
[I] AudioPlayer.h : 348 - setVolume(1.000000)
[I] VolumeStream.h : 186 - setVolume: 1.000000
[I] VolumeStream.h : 186 - setVolume: 1.000000
[I] AudioTypes.h : 85 - sample_rate: 0
[I] AudioTypes.h : 86 - channels: 0
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] MetaDataID3.h : 564 - virtual void audio_tools::MetaDataID3::begin()
[I] AudioSourceSDFAT.h : 101 - selectStream SDFAT: 0
[I] AudioSourceSDFAT.h : 120 - -> selectStream: /testold.mp3
[I] StreamCopy.h : 74 - buffer_size=1024
Starting up app[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] AudioPlayer.h : 239 - sample_rate: 44100
[I] AudioPlayer.h : 240 - bits_per_sample: 16
[I] AudioPlayer.h : 241 - channels: 2
[I] VolumeStream.h : 186 - setVolume: 1.000000
[I] VolumeStream.h : 186 - setVolume: 1.000000
[I] AudioTypes.h : 85 - sample_rate: 44100
[I] AudioTypes.h : 86 - channels: 2
[I] AudioTypes.h : 87 - bits_per_sample: 16
[I] I2SConfig.h : 113 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 114 - port_no: 0
[I] I2SConfig.h : 115 - is_master: Master
[I] I2SConfig.h : 116 - sample rate: 88200
[I] I2SConfig.h : 117 - bits per sample: 32
[I] I2SConfig.h : 118 - number of channels: 2
[I] I2SConfig.h : 119 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 121 - auto_clear: true
[I] I2SConfig.h : 123 - use_apll: true
[I] I2SConfig.h : 126 - fixed_mclk: 22579200
[I] I2SConfig.h : 128 - buffer_count:6
[I] I2SConfig.h : 129 - buffer_size:512
[I] I2SConfig.h : 143 - pin_data: 23
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
```

any idea what could be wrong? I checked with oscilloscope and pin out is 0V flat

pwilkowski, 2023-06-05:
Update:
I think there may be something wrong with spdif class as it seems to write couple of bytes and then dies?

```
[I] I2SConfig.h : 113 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 114 - port_no: 0
[I] I2SConfig.h : 115 - is_master: Master
[I] I2SConfig.h : 116 - sample rate: 88200
[I] I2SConfig.h : 117 - bits per sample: 32
[I] I2SConfig.h : 118 - number of channels: 2
[I] I2SConfig.h : 119 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 121 - auto_clear: true
[I] I2SConfig.h : 123 - use_apll: true
[I] I2SConfig.h : 126 - fixed_mclk: 22579200
[I] I2SConfig.h : 128 - buffer_count:6
[I] I2SConfig.h : 129 - buffer_size:512
[I] I2SConfig.h : 143 - pin_data: 27
idx.txt	99 bytes
idx-def.txt	7 bytes
testold.mp3	6110818 bytes
test.mp3x	118814943 bytes
	7344836 bytes
	7434681 bytes
file size 6110818
file is av 6110818
file is av ptr 1073489712
[I] StreamCopy.h : 74 - buffer_size=1594848797
[I] StreamCopy.h : 60 - buffer_size=1024
Starting up apptrying to write
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
trying to write
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
trying to write
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
trying to write
[I] I2SConfig.h : 113 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 114 - port_no: 0
[I] I2SConfig.h : 115 - is_master: Master
[I] I2SConfig.h : 116 - sample rate: 88200
[I] I2SConfig.h : 117 - bits per sample: 32
[I] I2SConfig.h : 118 - number of channels: 2
[I] I2SConfig.h : 119 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 121 - auto_clear: true
[I] I2SConfig.h : 123 - use_apll: true
[I] I2SConfig.h : 126 - fixed_mclk: 22579200
[I] I2SConfig.h : 128 - buffer_count:6
[I] I2SConfig.h : 129 - buffer_size:512
[I] I2SConfig.h : 143 - pin_data: 23
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
trying to write
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 0 -> 0 bytes - in 0 hops
failed to write
```

or maybe there needs to be delay somewhere?

pschatzmann, 2023-06-06:
I just tried the [orginal test case](https://github.com/pschatzmann/arduino-audio-tools/tree/main/examples/examples-stream/streams-generator-spdif) which does not show this issue. 

So it must come from your file: I suggest that you try to create some test cases with reading and decoding mp3 files...

pwilkowski, 2023-06-06:
I spend good couple of hours trying various test cases and this example works for me:

https://github.com/pschatzmann/arduino-audio-tools/blob/9d259461b7ae2f0ce0299f32350e9f79500dda02/examples/examples-basic-api/basic-a2dp-spdif/basic-a2dp-spdif.ino#L3

This is where i started from, unfortunately your example with sine wave spdif did not work. I just hear cracking from speakers.

I have tested this in both arduino and platformio. 

my pio config:

```ini
[env:upesy_wroom]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
board_build.partitions = min_spiffs.csv
monitor_filters = esp32_exception_decoder
lib_deps =
	greiman/SdFat@^2.2.2
	https://github.com/pschatzmann/arduino-libhelix
	https://github.com/pschatzmann/arduino-libmad.git
	https://github.com/pschatzmann/arduino-audiokit
	https://github.com/pschatzmann/arduino-audio-tools.git
build_flags =
	-Ofast
	-DI2S_BUFFER_COUNT=8
	-DI2S_BUFFER_SIZE=256
	-DDEFAULT_BUFFER_SIZE=2048
```

At the same time i tested ESP8266 with another library and the same mp3 files:
https://github.com/earlephilhower/ESP8266Audio

And it works without issues, reason why i want to use esp32 is that other hardware does not seem to work well with 8266 like lcd display and im starting to lack pins. Thus i found your library.

Here is the repository that works (ESP8266 and ESP8266Audio lib):
https://github.com/kaminaris/Mp3Player8266

Here is the repository that doesn't work (previous attempts are in main.txt):
https://github.com/kaminaris/Mp3Player32

So idk what kind of mp3 files are suppose to be there to work?

pschatzmann, 2023-06-06:
I don't really have any way to test the SPDIF so I am afraid that I can't really help: I just know that some users confirmed that it was working for them.

Any mp3 with a __standard bitrate__ should work w/o any changes. If you use some mp3 with some extreme or insane bitrate you will need to adjust the buffer size as explained in the Readme of the libhelix library. (I was using the bitrate classification from Audicity)

If Earle Phil Howers library is working for you, why don't you just use it with the ESP32 ?

pwilkowski, 2023-06-06:
i reencoded all my mp3 files to 128k because esp8266 was not able to keep up with higher bitrate ones

as for the its not working page, it doesn't explain why bluetooth audio example works. My bet goes to data not being processed correctly for mp3 -> spdif and maybe bluetooth already preprocesses pcm samples in some way to make it compatible?

pwilkowski, 2023-06-06:
I had exact same config on my test case as A2DP example (set to pin 27)

Which is why it was really weird for me than code from above did not work because when reading mp3 it stated exactly that.
44100, 2 channels, 16 bit

pschatzmann, 2023-06-06:
Maybe you should play around with the i2s buffer settings: If you check the A2DP example 
a buffer_count=30 and buffer_size = 384 is used. This is optimized for the output size (384) produced by A2DP

pwilkowski, 2023-06-06:
Yes i did try with couple settings as you can see in build flags:

```
build_flags =
	-Ofast
	-DI2S_BUFFER_COUNT=8
	-DI2S_BUFFER_SIZE=256
	-DDEFAULT_BUFFER_SIZE=2048
```

first thing ive checked was 30 and 384

pschatzmann, 2023-06-06:
I don't think 384 is a good value for mp3

pwilkowski, 2023-06-06:
Ive rebuild the project with 
```
build_flags =
	-Ofast
	-DI2S_BUFFER_COUNT=8
	-DI2S_BUFFER_SIZE=11776```
and now it mcu resets

tried also with I2S_BUFFER_SIZE=512

and cannot hear anything anymore

pschatzmann, 2023-06-12:
I committed a correction that should address the issue with the pin

pwilkowski, 2023-06-12:
Thanks, Any idea why only one speaker plays? When playing with PC it works on both but here, left speaker is silent