pschatzmann/arduino-audio-tools #893: I2S to webserver not working

khan46000, 2023-06-08:
i can listen to the audio on the client side using ffmpeg like VLC media player but not on web browser like chrome.It only display a player without any audio on web browser.
[image]

pschatzmann, 2023-06-08:
I noticed the same thing and was wondering if the browsers have changed and are much more restrictive now

khan46000, 2023-06-08:
The esp32 examples like camera web server are running fine.
Did you get the audio on any browser with success.

pschatzmann, 2023-06-08:
It used to work w/o any problems in the past. Currently I have no idea what could cause this issue

pschatzmann, 2023-06-08:
I keep this open as a reminder that this needs some extensive investigation.

Lukmicsch, 2023-06-08:
I had the same problem and wrote a rudimentary Python client as a sanity check. I could post it here if you think that might help for the time being.

yu2zz, 2023-06-08:
First, thank You for Your hard work on these libraries. Regarding this streaming issue I am having the same problem. Browsers show the player graphic but no audio is actually being passed through. The board is AI Audio Kit v2.2 A247. I have tried both .aac and .wav examples, as well as multiple sample rates starting from 8000. No luck.

a-1an, 2023-07-16:
hey where do i get the address of webpage to see the audio? @yu2zz  @khan46000 @pschatzmann 

@Lukmicsch could you post the python script please?

pschatzmann, 2023-07-16:
You can also just use a wget to download the audio into a file

Lukmicsch, 2023-07-16:
@a-1an 

```python
import requests
import pyaudio

# Replace with your device
url = "http://"

# Set up PyAudio
p = pyaudio.PyAudio()
stream = p.open(format=pyaudio.paInt16, channels=2, rate=22050, output=True)

response = requests.get(url, stream=True)

if response.status_code == 200:
    try:
        # Read and play the audio stream in chunks
        for chunk in response.iter_content(chunk_size=1024):
            if chunk:
                stream.write(chunk)
    except KeyboardInterrupt:
        # User pressed ctrl+c, stop playing
        stream.stop_stream()
        stream.close()
        p.terminate()
else:
    print("Could not connect to the server, received status code:", response.status_code)

tomas-javurek, 2023-08-04:
It would be CORS, resp. insecure media source

hovercraft-github, 2023-11-01:
Looks like I found the root cause of this issue.
Appears that browser initially makes two concurrent connections to our stream server, but the second one then remains. The problem is that the second connection doesn't receive a coding format header, e.g a WAV file header.
My workaround is to restart encoder in the `AudioEncoderServer` class for every new client connection, just placing this
```
        		encoder->end();
        		encoder->begin();

```
in the begin of `AudioEncoderServer::sendReplyContent` method should work.
Tested with Chromium Version 118.0.5993.117 so far.

pschatzmann, 2023-11-01:
Oh, cool: that explains why playing via the command line has been working

closed 2023-11-01 by pschatzmann