pschatzmann/arduino-audio-tools #898: Initializing a second I2S device causes stream copy errors

that-one-arab, 2023-06-13:
## Description
I'm trying to go for the following flow:
- Save a microphone recording to an SD card
- Play the recording from the SD card to a speaker

Both the speaker and the microphone are I2S devices

When starting `copier.copy()` from the microphone to the encoder that will save to SD card, the below error occurs:
```
E (183) I2S: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1352 (i2s_read):rx NULL
[E] I2SESP32.h : 103 - size_t audio_tools::I2SDriverESP32::readBytes(void*, size_t)
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 0 -> 0 bytes - in 0 hops
```

## Code
Below is the minimum code to reproduce the issue
```cpp
#include "SD.h"
#include <SPI.h>
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

I2SStream speaker;                                            
EncodedAudioStream decoder(&speaker, new MP3DecoderHelix());  
StreamCopy speakerCopier;

I2SStream mic;                                         
File audioFile;                                        
EncodedAudioStream encoder(&audioFile, new WAVEncoder());  
StreamCopy copier(encoder, mic);                       

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  if (!SD.begin(5)) {
    Serial.println("Card Mount Failed");
    while (1)
      ;
  }

  ////
  // When this block of code is initialized the errors occur
  auto speakerConfig = speaker.defaultConfig(TX_MODE);
  speakerConfig.channels = 2;
  speakerConfig.sample_rate = 16000;
  speakerConfig.pin_ws = 15;
  speakerConfig.pin_bck = 14;
  speakerConfig.pin_data = 22;
  speaker.begin(speakerConfig);
  decoder.setNotifyAudioChange(speaker);
  decoder.begin();
  ////

  auto micConfig = mic.defaultConfig(RX_MODE);
  micConfig.channels = 2;
  micConfig.sample_rate = 16000;
  micConfig.pin_ws = 21;
  micConfig.pin_bck = 12;
  micConfig.pin_data = 32;
  mic.begin(micConfig);

  auto cfg_out = encoder.defaultConfig();
  cfg_out.channels = 2;
  cfg_out.sample_rate = 16000;
  encoder.begin(cfg_out);
  copier.setCheckAvailableForWrite(false);

  if (SD.remove("/AudioFile01.wav")) {
    Serial.println("File deleted");
  }
  audioFile = SD.open("/AudioFile01.wav", FILE_WRITE);

  copier.begin(encoder, mic);
}

void loop() {
  copier.copy();
}
```

## Logs:

```
[I] I2SConfig.h : 117 - rx/tx mode: TX_MODE
[I] I2SConfig.h : 118 - port_no: 0
[I] I2SConfig.h : 119 - is_master: Master
[I] I2SConfig.h : 120 - sample rate: 16000
[I] I2SConfig.h : 121 - bits per sample: 16
[I] I2SConfig.h : 122 - number of channels: 2
[I] I2SConfig.h : 123 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 125 - auto_clear: true
[I] I2SConfig.h : 127 - use_apll: true
[I] I2SConfig.h : 132 - buffer_count:6
[I] I2SConfig.h : 133 - buffer_size:512
[I] I2SConfig.h : 143 - pin_bck: 14
[I] I2SConfig.h : 145 - pin_ws: 15
[I] I2SConfig.h : 147 - pin_data: 22
[I] AudioEncoded.h : 269 - virtual void audio_tools::EncodedAudioOutput::setNotifyAudioChange(audio_tools::AudioInfoSupport&)
[I] I2SConfig.h : 117 - rx/tx mode: RX_MODE
[I] I2SConfig.h : 118 - port_no: 0
[I] I2SConfig.h : 119 - is_master: Master
[I] I2SConfig.h : 120 - sample rate: 16000
[I] I2SConfig.h : 121 - bits per sample: 16
[I] I2SConfig.h : 122 - number of channels: 2
[I] I2SConfig.h : 123 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 125 - auto_clear: false
[I] I2SConfig.h : 127 - use_apll: true
[I] I2SConfig.h : 132 - buffer_count:6
[I] I2SConfig.h : 133 - buffer_size:512
[I] I2SConfig.h : 143 - pin_bck: 12
[I] I2SConfig.h : 145 - pin_ws: 21
[I] I2SConfig.h : 147 - pin_data: 32
[I] CodecWAV.h : 482 - sample_rate: 16000
[I] CodecWAV.h : 483 - channels: 2
[I] CodecWAV.h : 492 - is_streamed! because length is 2147418112
[I] StreamCopy.h : 82 - buffer_size=1024
E (183) I2S: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/i2s.c:1352 (i2s_read):rx NULL
[E] I2SESP32.h : 103 - size_t audio_tools::I2SDriverESP32::readBytes(void*, size_t)
[I] StreamCopy.h : 139 - StreamCopy::copy 1024 -> 0 -> 0 bytes - in 0 hops
```

pschatzmann, 2023-06-13:
Please check the working example. You forgot to define the ports!

closed 2023-06-13 by pschatzmann

that-one-arab, 2023-06-14:
Yes indeed, I needed to define the ports.

Here's how it should look like:

```cpp
#include "SD.h"
#include <SPI.h>
#include "AudioTools.h"
#include "AudioCodecs/CodecMP3Helix.h"

I2SStream speaker;                                            
EncodedAudioStream decoder(&speaker, new MP3DecoderHelix());  
StreamCopy speakerCopier;

I2SStream mic;                                         
File audioFile;                                        
EncodedAudioStream encoder(&audioFile, new WAVEncoder());  
StreamCopy copier(encoder, mic);                       

void setup() {
  Serial.begin(115200);
  AudioLogger::instance().begin(Serial, AudioLogger::Info);

  if (!SD.begin(5)) {
    Serial.println("Card Mount Failed");
    while (1)
      ;
  }

  auto speakerConfig = speaker.defaultConfig(TX_MODE);
  speakerConfig.channels = 2;
  speakerConfig.sample_rate = 16000;
  speakerConfig.pin_ws = 15;
  speakerConfig.pin_bck = 14;
  speakerConfig.pin_data = 22;
  speakerConfig.port_no = 1; // Since micConfig initializes port_no on 0, we have to change either the mic or speaker port numbers.
  speaker.begin(speakerConfig);
  decoder.setNotifyAudioChange(speaker);
  decoder.begin();

  auto micConfig = mic.defaultConfig(RX_MODE);
  micConfig.channels = 2;
  micConfig.sample_rate = 16000;
  micConfig.pin_ws = 21;
  micConfig.pin_bck = 12;
  micConfig.pin_data = 32;
  mic.begin(micConfig);

  auto cfg_out = encoder.defaultConfig();
  cfg_out.channels = 2;
  cfg_out.sample_rate = 16000;
  encoder.begin(cfg_out);
  copier.setCheckAvailableForWrite(false);

  if (SD.remove("/AudioFile01.wav")) {
    Serial.println("File deleted");
  }
  audioFile = SD.open("/AudioFile01.wav", FILE_WRITE);

  copier.begin(encoder, mic);
}

void loop() {
  copier.copy();
}
```