pschatzmann/arduino-audio-tools #900: Portenta H7 Compatability

trylaarsdam, 2023-06-14:
Hi there,

I've been trying out this library in PlatformIO for use with a Portenta H7, since I saw it was referenced in the Nicla Voice documentation from Arduino, and figured I'd give it a shot on the Portenta.

I got the library to compile by undefining Arduino's absolute value implementation and defining use int24 from int 
```cpp
#undef abs
#define USE_INT24_FROM_INT
#include "AudioTools.h"
```
However I ran into an issue where I2SStream is not defined by default. I then tried defining `STM32` to enable the STM32 mode:
```cpp
#define STM32
```
But that results in an error saying I am then missing the `stm32-i2s.h` header.

Just for fun, and not expecting it to work since the Arduino I2S library doesn't support the Portenta :/ I defined `USE_I2S`, but since the Arduino I2S library is not present it obviously does not compile. 

So, if the Portenta is supported, what am I missing, otherwise, would there be any plans to support it in the future? I recognize it's fairly new and stuck in between Arduino and ST's ecosystems so it's not the easiest board in the world.

Thanks

pschatzmann, 2023-06-14:
I tried to google the topic I2S and Portenta and I did not find a single example.
It seems that the challenge is really that there is no I2S library: this is a general problem for all STM processors.
I once created a [library for the Black Pill](https://github.com/pschatzmann/stm32f411-i2s), so something similar would be needed. 

But looking of the cost of these processors, my motivation is quite low...

trylaarsdam, 2023-06-15:
Here's what I get when I try out the basic stream-generator-analog example:
`[E] AnalogAudioBase.h : 144 - Only RX_MODE supported`

```cpp
#include "WiFi.h"
#include <Arduino.h>
#define ARDUINO_ARCH_MBED_PORTENTA
#include "AudioTools.h"

typedef int16_t sound_t;                                   // sound will be represented as int16_t (with 2 bytes)
AudioInfo info(44100, 2, 16);
SineWaveGenerator<sound_t> sineWave(32000);                // subclass of SoundGenerator with max amplitude of 32000
GeneratedSoundStream<sound_t> sound(sineWave);             // Stream generated from sine wave
AnalogAudioStream out; 
StreamCopy copier(out, sound);                             // copies sound into i2s

void setup() {
	Serial.begin(9600);
	while(!Serial);
	AudioLogger::instance().begin(Serial, AudioLogger::Info);

  // // start the analog output
  audio_tools::AnalogConfig config = out.defaultConfig(TX_MODE);
  config.copyFrom(info); 
  // out.begin(config); // causes Mbed panic

  // // Setup sine wave
  // sineWave.begin(info, N_B4);
  // Serial.println("started...");
}

void loop() {
	// copier.copy();
}
```

It seems like only the ESP32 define will enable TX mode. I will need to do RX eventually, but for now, I want to get TX working since that's also a requirement. 

In the meantime, I'm going to look at just implementing the I2S protocol on the Portenta in my own library, but let me know if there is a way to enable TX through your library - no reason to reinvent the wheel.

pschatzmann, 2023-06-15:
I don't understand your problem: [This example](https://github.com/pschatzmann/arduino-audio-tools/blob/main/examples/examples-stream/streams-generator-analog/streams-generator-analog.ino) is just compiling fine and just gives some warnings - but no errors!

pschatzmann, 2023-06-15:
I think it is more efficient, when you create a fork and do the corrections yourself right away.
If it is working you can share them with a pull request...

As I said, I do not have any of these boards, so I cant test it myself...

trylaarsdam, 2023-06-15:
I did the same thing by just bypassing it, unfortunately, later calls in the `AnalogAudioStream.out` function causes a panic on the Portenta, so I need to do some research. 

Here's what logging gets sent before the crash:
```
[I] AnalogAudio.h : 58 - bool audio_tools::AnalogAudioStream::begin(audio_tools::AnalogConfig)
[I] AnalogAudioMBED.h : 26 - virtual bool audio_tools::AnalogDriverMBED::begin(audio_tools::AnalogConfig)
[I] Buffers.h : 295 - resize: 512
```

I'll let you know if I get something working. Thanks!

trylaarsdam, 2023-06-16:
I was able to get I2S working on the Portenta by bringing in the CubeMX initialization functions and created a [very simple PIO library](https://github.com/trylaarsdam/portenta-i2s).

I haven't tried DMA as I still need to bring over some other code from Cube for that, but calling `HAL_I2S_Transmit` is working for me with a sine wave. Here's all my initialization code for I2S2 on the Portenta/STM32H747:
https://github.com/trylaarsdam/portenta-i2s/blob/main/src/i2s_init.cpp

I don't have time to work on a PR to include this into the STM32 mode of this lib, but I'd be happy to test something out if you add in the code.

Thanks for your help

pschatzmann, 2023-06-17:
Looks promising. Unfortunately I am currently very sick and will not be able to work on this.
Why don't you just try to implement the Print interface ? 
This way you could use your class directly like any other output class of my API