pschatzmann/arduino-audio-tools #901: VolumeStream  dor Wav-Files

lechk82, 2023-06-15:
How to apply a VolumeStream to wave files? I would like to control the volume for both channels separately. The playback works, but the setting of the VolumeStream has no influence on the sound level.

Here is my Code:

```
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "AudioLibs/AudioSourceSD.h"

AudioKitStream i2s;
VolumeStream volume(i2s);
EncodedAudioStream decoder(&volume, new WAVDecoder()); 
StreamCopy player;
File audioFile;

auto i2s_config = i2s.defaultConfig(TX_MODE);
auto vol_config = volume.defaultConfig();

void setup() {
  Serial.begin(115200);

  SD.begin(PIN_AUDIO_KIT_SD_CARD_CS, AUDIOKIT_SD_SPI);
  audioFile = SD.open("/rec/wav/file.wav");

  i2s.begin(i2s_config);
  volume.begin(vol_config);
  volume.setVolume(0.0,0);
  volume.setVolume(0.0,1); 
  decoder.begin();
  player.begin(decoder, audioFile);
}

void loop() {
  player.copy();
}
```

pschatzmann, 2023-06-16:
Hmm, this should work but why are you calling 
```
  volume.setVolume(0.0,0);
  volume.setVolume(0.0,1); 
```
Instead of just
```
  volume.setVolume(0.0, 1.0); 
```

Are there any informations, warnings or errors in the log ?

lechk82, 2023-06-16:
Thanks for the quick reply. I use the volumestream because eventually I want to set both channels for different levels.  The wave file contains 2 channels, I checked it again in audacity.

Here is the log output:

```
[D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[D] AudioEncoded.h : 332 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[D] CodecWAV.h : 306 - virtual size_t audio_tools::WAVDecoder::write(const void*, size_t)
[D] AudioStreams.h : 1067 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[D] AudioCopy.h : 90 - size_t audio_tools::StreamCopyT<T>::copy() [with T = unsigned char; size_t = unsigned int]
[D] AudioEncoded.h : 332 - virtual size_t audio_tools::EncodedAudioStream::write(const uint8_t*, size_t): 1024
[D] CodecWAV.h : 306 - virtual size_t audio_tools::WAVDecoder::write(const void*, size_t)
[D] AudioStreams.h : 1067 - virtual size_t audio_tools::VolumeStream::write(const uint8_t*, size_t)
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
[I] I2SConfig.h : 104 - i2s_format: I2S_STD_FORMAT
[I] I2SConfig.h : 106 - auto_clear:0
[I] I2SConfig.h : 113 - buffer_count:10
[I] I2SConfig.h : 114 - buffer_size:512
[I] I2SConfig.h : 118 - pin_bck: 14
[I] I2SConfig.h : 119 - pin_ws: 15
[I] I2SConfig.h : 120 - pin_data: 0
[W] AudioKit.h : 54 - sd_active = true
[I] AudioKit.h : 202 - Channels 2->2
[I] AudioKit.h : 599 - void audio_tools::AudioKitStream::setupActions()
[W] AudioKit.h : 611 - Mode Button ignored because of conflict: 13 
[I] AudioActions.h : 42 - ActionLogic::add pin: 39 / logic: 2
[I] AudioActions.h : 55 - pin 39 -> INPUT
[I] AudioKit.h : 349 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 18 / logic: 0
[I] AudioActions.h : 52 - pin 18 -> INPUT_PULLUP
[I] AudioKit.h : 349 - void audio_tools::AudioKitStream::addAction(int, void (*)(bool, int, void*), void*)
[I] AudioActions.h : 42 - ActionLogic::add pin: 5 / logic: 0
[I] AudioActions.h : 52 - pin 5 -> INPUT_PULLUP
[I] AudioStreams.h : 1106 - setVolume: 1.000000
[I] AudioStreams.h : 1106 - setVolume: 1.000000
[I] AudioStreams.h : 1106 - setVolume: 0.000000
[I] AudioStreams.h : 1106 - setVolume: 0.000000
[I] AudioCopy.h : 69 - buffer_size=1024
[I] CodecWAV.h : 138 - WAVHeader::write: 1024 -> 44 -> 44
[I] CodecWAV.h : 49 - WAVHeader::begin: 44
[I] CodecWAV.h : 218 - WAVHeader sound_pos: 44
[I] CodecWAV.h : 219 - WAVHeader channels: 2 
[I] CodecWAV.h : 220 - WAVHeader bits_per_sample: 16
[I] CodecWAV.h : 221 - WAVHeader sample_rate: 44100 
[I] CodecWAV.h : 222 - WAVHeader format: 1
[I] CodecWAV.h : 323 - WAV sample_rate: 44100
[I] CodecWAV.h : 324 - WAV data_length: 2147418104
[I] CodecWAV.h : 325 - WAV is_streamed: 1
[I] CodecWAV.h : 326 - WAV is_valid: true
[I] CodecWAV.h : 344 - isValid: true
[I] AudioStreams.h : 1106 - setVolume: 1.000000
[I] AudioStreams.h : 1106 - setVolume: 1.000000
[I] CodecWAV.h : 347 - WAVDecoder writing first sound data
[I] AudioCopy.h : 256 - try write - 2 (open 0 bytes) 
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 2 hops
[I] AudioCopy.h : 134 - StreamCopy::copy 1024 -> 1024 -> 1024 bytes - in 1 hops
...
```

pschatzmann, 2023-06-17:
I changed the logic, so that the begin method is not called again when the AudioInfo is changed by the decoder.
You could also resolve the issue to call setVolume in the loop...

lechk82, 2023-06-17:
Thanks. I will test it in the next days and then provide feedback here.

lechk82, 2023-06-19:
This works now as expected. However, `i2s.setVolume(1.0); ` is required, otherwise setVolume will not work for ch1 and ch1 will remain silent. I do not know if this is intentional.

```
#include "AudioTools.h"
#include "AudioLibs/AudioKit.h"
#include "SD.h"
#include "SPI.h"

AudioKitStream i2s;
VolumeStream volume(i2s);
EncodedAudioStream decoder(&volume, new WAVDecoder()); 
StreamCopy player;
File audioFile;

auto i2s_config = i2s.defaultConfig(TX_MODE);
auto volume_config = volume.defaultConfig();

void initSDCard() {
  SPI.begin(PIN_AUDIO_KIT_SD_CARD_CLK, PIN_AUDIO_KIT_SD_CARD_MISO, PIN_AUDIO_KIT_SD_CARD_MOSI, PIN_AUDIO_KIT_SD_CARD_CS);
  if (!SD.begin(PIN_AUDIO_KIT_SD_CARD_CS, AUDIOKIT_SD_SPI)) {
    Serial.println("Card Mount Failed");
    return;
  } else {
    Serial.println("Card Mount Success");
  }
}

#define MAXVOL (0.8f)
#define MINVOL (0.0f)
#define VOLSTEP (0.1f)

void setup() {
  Serial.begin(115200);
  initSDCard();

  i2s.begin(i2s_config);
  i2s.setVolume(1.0);         // <-- this is crucial, otherwise setVolume will not work for ch1 and ch1 will remain silent.
  volume.begin(volume_config);
  volume.setVolume(MAXVOL,0);
  volume.setVolume(MINVOL,1);
  decoder.begin();
  audioFile = SD.open("/rec/wav/file.wav");
  player.begin(decoder, audioFile);
}

unsigned long t1=0;
volatile float v0 = MAXVOL;
volatile float v1 = 0.0f;

void loop() {
  if((millis()-t1)>250){
      if(v0 < MINVOL) v0 = MAXVOL;
      if(v1 > MAXVOL) v1 = MINVOL;
      v0-=VOLSTEP;
      v1+=VOLSTEP;
      volume.setVolume(v0,0);
      volume.setVolume(v1,1);
      t1=millis();
  }
  player.copy();
}
```

pschatzmann, 2023-06-19:
The issue was actually that any setVolume() with 1.0 was deactivating the volume control!
I removed this logic...

closed 2023-06-19 by pschatzmann